openscad/openscad #5098: cmake aborts with no quickhull/QuickHull.cpp

zaitcev, 2024-04-22:
**Environment**
Linux, Fedora 39

**Issue**
Error looks like this:

[zaitcev@lebethron build]$ cmake .. -DEXPERIMENTAL=1 > ../cmake.out 2>&1        
......................................................

CMake Error at submodules/manifold/src/third_party/CMakeLists.txt:1 (add_library):
  Cannot find source file:

    quickhull/QuickHull.cpp

CMake Error at submodules/manifold/src/third_party/CMakeLists.txt:1 (add_library):
  No SOURCES given to target: quickhull

CMake Generate step failed.  Build files cannot be regenerated correctly.

[zaitcev@lebethron build]$ 

The problem may have something to do with quickhull being a submodule
of manifold. My submodules look like this:

[zaitcev@lebethron openscad-work]$ git submodule status
 1ea402208c3127ffb443931e9bb1681c191dacca libraries/MCAD (openscad-2019.05-78-g1ea4022)
 059a173f161a39a4ab28a32a5b89adabd4f77370 submodules/OpenCSG (opencsg-1-6-0-release-1-g059a173)
 c03a5ac3adbb002ef29646648adbbebba179000d submodules/manifold (v2.3.1a-26-gc03a5ac)
 fe61be80bf8942e764e18d469e96f8dacab44728 submodules/mimalloc (v1.7.7)
 c3dc841af4dbf44669e65b82cb68a575864326bd submodules/sanitizers-cmake (c3dc841)
[zaitcev@lebethron openscad-work]$ 

As you can see, quickhull is not present.

One weird part, this obviously precludes anything from being done at all, but no other developer complained. User error, right? But what is it? I always do the same thing that README.md says to do: checkout, mkdir build, cd build, cmake .. -DEXPERIMENTAL=1 and done.

zaitcev, 2024-04-22:
Actually never mind. This worked:

[zaitcev@lebethron openscad-work]$ git submodule update --recursive

closed 2024-04-22 by zaitcev

tnm23, 2024-07-19:
Am getting this same error as of today; git submodule update --recursive does not fix it.

kintel, 2024-07-19:
@tnm23 try this from a clean folder:
```
git clone --recursive git@github.com:openscad/openscad.git 
ls -l openscad/submodules/manifold/src/third_party/quickhull
```

You should see smth. like this:
```
total 104
-rw-r--r--  1 kintel  staff   5216 19 Jul 14:56 ConvexHull.hpp
-rw-r--r--  1 kintel  staff   2210 19 Jul 14:56 HalfEdgeMesh.hpp
-rw-r--r--  1 kintel  staff   1204 19 Jul 14:56 MathUtils.hpp
-rw-r--r--  1 kintel  staff  19262 19 Jul 14:56 QuickHull.cpp
-rw-r--r--  1 kintel  staff  10552 19 Jul 14:56 QuickHull.hpp
-rw-r--r--  1 kintel  staff   1699 19 Jul 14:56 README.md
drwxr-xr-x  8 kintel  staff    256 19 Jul 14:56 Structs
drwxr-xr-x  6 kintel  staff    192 19 Jul 14:56 Tests
```

If it's good so far, try to build and report back.