openscad/openscad #5115: Release Policy

mkohlhaas, 2024-04-28:
There are more than 1000 commits since the last release three years ago. I just wonder what the release policy is.

Kind regards!

pca006132, 2024-04-28:
I don't there is an official release policy. There are still a lot of stuff needed to make experimental features stabilize (but thankfully is about to finish? @kintel), and there are also incompatibilities with older version.

See #4825.

kintel, 2024-04-28:
The primary agreed upon policy is that features which are part of the release should be long-term supportable. This primarily means that designs created with such a release should keep working in future releases, and designs created in the past should work without having to be upgraded.

UI features are less critical.

There's still a lot of work to do, and we're a very small dev team on volunteers, so progress is not easy to anticipate.

closed 2024-04-28 by mkohlhaas