openscad/openscad #5116: Viewport-Control widget can be made too narrow; tab order broken.

haikusw, 2024-04-29:
**Describe the bug**
PR #5107 improved the Viewport-Control widget by allowing it to morph to a vertical layout when the widget became too narrow for the original horizontal layout.  

However after that change the widget could be made too narrow (issue 1) and look bad because the controls were "cut off" and only partially visible.  #5107 also lost the left-to-right top-to-bottom tab order within the widget (issue 2).  This PR fixes both of these issues.

**To Reproduce issue 1 - widget may be made too narrow**
1. Open a new document in OpenSCAD.
2. If the Viewport-Control isn't visible, select "Viewport-Control" from the Window menu
3. If the Viewport-Control isn't in the right size of the Window, move it there.
4. Resize the Viewport-Control widget to as small as you can get it.
5. Notice that its contents are no longer fully visible (screenshot below).

**Expected behavior issue 1**
The widget width should be constrained to be _at least_ the minimum width necessary to fit the contents and avoid cutting off any of the controls in the window.

**To Reproduce issue 2 - tab order is weird**
1. Open a new document in OpenSCAD.
2. If the Viewport-Control isn't visible, select "Viewport-Control" from the Window menu
3. Set the focus to the first control in the Viewport-Control widget: Aspect Ratio.
4. Hit the tab key repeatedly and watch the focus jump around in the widget in a less than useful order.

**Screenshots**
Issue 1 - Viewport-Control can be made too narrow:

[image]

Issue 2 - Viewport-Control tab order is weird:

[image]

**Environment and Version info (please complete the following information):**
 - OS: macOS 13.6.6
 - System: 13" M1 MacBook Pro
 - OpenSCAD Version:  Head of master as of 4/28/2024 (hash: 97184930d36d6d5cb126b5e0e0cd5ca3c606c05f ) that includes #5107

t-paul, 2024-04-29:
As said in #4706 this is intentional (and probably not possible to solve).

closed 2024-04-29 by t-paul

haikusw, 2024-04-29:
@t-paul Maybe checkout the branch, build it and take a look at it?  There are two issues, as noted, and they are both 100% solved as far as I can tell with a very simple fix for both.

t-paul, 2024-04-29:
Of course, see comment there.

haikusw, 2024-04-29:
Thanks.   Was just surprised at this being closed before the PR was evaluated.

t-paul, 2024-04-29:
Yeah, I thought the other one is an issue too, I did not have enough coffee yet to see the PR icon. But that's not a problem, just the PR is fine.

haikusw, 2024-04-29:
I'm on the other end of the bleary eyes - just after midnight in my timezone.  Time for some Zzzzz.  I'll check in tomorrow in case you want some changes to the PR.

commit bf01874d (2024-05-14): Merge pull request #5117 from haikusw/viewport-control-layout-minwidth

Viewport-Control widget: Limit width to at least dynamic minimum size of widget. Add tab order back in.  Fixes #5116