openscad/openscad #5118: simple helper (add one button) to reset time

petaflot, 2024-05-01:
**Is your feature request related to a problem? Please describe.**
I use the animate feature a lot when working with assemblies with moving parts, ie. gears and reductors. when I don't want to use the animation (such as rendering parts for print), often I want $t to be at zero, so I need to open the animate widget, double-click on the time input field, input `0`, hit `[enter]`, then close the animate widget.

this should be a very quick fix for anyone who knows their way around the codebase.

**Describe the solution you'd like**
in the button bar (where you can set the projection, zoom level, enable/disable axis..), add a button to reset $t=0

**Describe alternatives you've considered**
see problem above

jordanbrown0, 2024-05-01:
You don't mention what version you're using, but note that in the current development snapshots the Animate pane / window has a rewind-to-start button.  It's not at the top level where you request, but it's a couple fewer steps.

petaflot, 2024-05-02:
I have version 2024.02.18...

cool that it's getting done :-) reminds of a bunch of other things, but I need to make a thorough description cuz it's a little more complicated.

since I'm using media-gfx/openscad-9999::gentoo I upgraded the package, I am now running 2024.05.02 but I still don't see the button...

jordanbrown0, 2024-05-02:
It's been there for a while.

Here's my Animate window:
[image]

The "rewind to start" button is the left one, the one that looks sort of like `|<<`.

petaflot, 2024-05-02:
oh dude I so missed that XD

closed 2024-05-02 by petaflot

t-paul, 2024-05-02:
Just for completeness... 5 days ago that window changed a bit, see https://github.com/openscad/openscad/pull/5112 so a mini screencast. The `|<<` is still there :grinning:.

petaflot, 2024-05-02:
since I usually make looping animations I kinda fail to see the point of the `>>|`button.

for completeness, see my suggestion here: https://github.com/gsohler/openscad/issues/26

jordanbrown0, 2024-05-02:
`>>|` would let you confirm that the last frame matches the first (modulo any movement that *should* happen between them).