openscad/openscad #5120: slicer integration and parameters

petaflot, 2024-05-02:
I find myself editing the slicing parameters for basically every model (rather, part of model) that I print. This is cumbersome and a cause of stupid errors (forgetting to set this or that parameter in the slicer).

AFAIK, the best slicer now is the arachne slicer ; some UIs allow for more than one slicer, so anyhow this should not be a tough barrier to overcome.

I would really like to be able to
* set slicer options in `.scad` files comments
* use openscad to call the slicer while passing the desired options (and possibly send to printer, ie. with octoprint)

pca006132, 2024-05-04:
I think this is much easier to implement in a slicer rather than openscad. If you implement this in a slicer, this is just letting the slicer read the `.scad` file comment for options, call openscad (via subprocess) to generate the model, and import the model into the slicer. And I still have no idea how to implement this in openscad without modifying the slicer code.

petaflot, 2024-05-06:
a "slicer" (actually GUIs for slicers) reads an stl file, displays it, and calls the slicer (possibly offering some configuration widgets). finally, it displays an new object that is the sum all all toolpaths.

Openscad generates the object, exports it as an stl file.

I fail to see how it would be difficult get Openscad to call a command-line tool for slicing, such as the [Cura engine](https://github.com/Ultimaker/CuraEngine). However, I fail to see why a particular, "random" slicer frontend would implement reading numerous arbitrary and more or less obscure filetypes from a very large variety of 3D editors.

OpenSCAD already had this feature to send the STL to a print service (commercial or a local octoprint instance) : either case, for a result pf any use the slicing parameters must be provided : specifying the slicing parameters inside the source file would only fill a gap that exists since the `F8` was added. Printing with "default" (or "random") parameters makes for absolutely useless parts unless one is printing silly figures, companion cubes, you-name-it that has no actual mechanical purpose.

I understand that for this to be cross-compatible between many slicers, it would involve a joint effort for the community, and better definition of some concepts : IMHO this would only help improve the global understanding and quality of parts produced.

gsohler, 2024-05-06:
Seems that 3MF can embed Slicer information

https://lib3mf.readthedocs.io/en/master/source/Cpp/lib3mf_SliceStackIterator.html

Remaining tasks are just

* how to use this function and what it can do
* openscad to export this information
* slicer to understand this information

tcurtin, 2024-05-06:
A +1 for using the 3mf.  I export directly to 3mf and then pull into Bambu Slicer, and for this particular project have many parts (from multiple 3mf files) I need to separate to objects but also keep aligned.  Being able to indicate position, filament, etc would be an amazing improvement both in accuracy and efficiency!

gsohler, 2024-05-06:
right now i am adding the possibility to  store several "designs" together in the same 3mf file(right now 3mf v2 ).
Can you give few examples of useful slicer settings you want to attach to the part ?
(are the properties siting on the part, the file, a certain z range or very different ? )

gsohler, 2024-05-06:
This is how it could look like

[image]

file is at http://www.guenther-sohler.net/export.3mf

t-paul, 2024-05-06:
As discussed multiple times, this is not happening for security reasons. At this point OpenSCAD can easily run as UI thumbnailer exactly because it will not start bitcoin mining when loading some random file from the internet.

To be clear, the feature requested, does NOT need procedural save in the language.

Saving will be possible via the python integration once that gets official, and this is guarded by explicit user action.

gsohler, 2024-05-06:
You are right .. no File write action triggered by script.
But I feel that the script should still define the set of parts to be exported.
The file-write-action itself can of course be started by GUI

petaflot, 2024-05-07:
Parameters I would like to see, for the entire part:
* top layer count
* bottom layer count
* shells
* infill type and percentage
* top fill pattern
* bottom fill pattern
* layer height
* vase mode or not vase mode
* export file name (absolute or relative) because I very often have a file with a number of parts (modules) and at least one "display" module that shows the assembled part (often animated) and these parts must be rendered and exported separately in the correct position for printing

Being able to override these parameters

Also, ideally
* where to generate support (I guess solids that, when they intersect with a face, mark that face as needing support)
* where NOT to generate supports (similar as above) ; could include volumes supports should avoid (if slicer supports it)
* seam position
* color hint

gsohler, 2024-05-07:
so you plan to use this function to retrieve object layer properties ?

LIB3MF_DECLSPEC Lib3MFResult lib3mf_meshobject_getobjectlevelproperty(Lib3MF_MeshObject pMeshObject, Lib3MF_uint32 * pUniqueResourceID, Lib3MF_uint32 * pPropertyID, bool * pHasObjectLevelProperty);

do you already have example code ?

rather than specifying top layer  counts and bottom layer count , and perimeter shells i would rather have per-face shells.

jordanbrown0, 2025-03-08:
This probably ties into proposals to allow adding arbitrary tags to CSG nodes, that an exporter could then consume to add to the exported data.

jordanbrown0, 2025-03-08:
See #5089 and #4977.

closed 2026-05-07 by jordanbrown0