openscad/openscad #5121: OpenScad as library

rafal-tarnow, 2024-05-03:
I think it would be an interesting idea to transform OpenScad into a library and separate the GUI from the application logic. Currently, I'm working on a 3D application where I would need a library that would allow me to easily build 3D models on the fly. If OpenScad could be compiled into a library, it would be a great solution for me. I could then use my own application interface, load scripts from it, compile them using OpenScad, retrieve geometries, and also provide error messages to the user. Essentially, I think I could achieve this by running OpenScad as a process and executing these tasks using CLI. However, this doesn't seem like a very elegant solution to me, and I'm not sure about handling compilation errors, or how useful the --info option would be.

rafal-tarnow, 2024-05-03:
I started browsing the CMake project, I'm not sure if the option option(BUILD_SHARED_LIBS "Build shared library" OFF) might be what I'm looking for?

t-paul, 2024-05-03:
Duplicate of #193, also not likely to happen unless someone can really dedicate a lot of time to that topic. A minimal very high level version would be very useful for WASM too.

closed 2024-05-03 by t-paul

t-paul, 2024-05-03:
If you have dedicated plans for an application, feel free to join the OpenSCAD IRC channel for some developer discussion.