openscad/openscad #5130: is there a way to get a degree symbol to display in a text function?  I have tried creating a string of chr(186) and printing that , but it gives something like a square.  Thanks

daackm, 2024-05-10:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

MethylBromide, 2024-05-10:
I believe you are getting the right character code, but using a font (maybe the default font?) that doesn't contain an image for that character.

t-paul, 2024-05-10:
Yes, try with this: `text(str("23", chr(186)), font = "Liberation Sans:style=Regular");`

[image]

What Operating System are you using?

daackm, 2024-05-10:
Thanks for the help.  I got it to work.  
 
I use OpenSCAD to design structures for model railroads.  I build front end programs which collect the design parameters for walls, roofing, windows and so forth, but was stuck on how to place text on a tool for designing a circus tent (visit http://daackm.github.io to see what I do).
 
I appreciate the information you shared with me.
 
Dave Ackmann

daackm, 2024-05-10:
Thanks for the help.  I got it to work.  
 
I use OpenSCAD to design structures for model railroads.  I build front end programs which collect the design parameters for walls, roofing, windows and so forth, but was stuck on how to place text on a tool for designing a circus tent (visit http://daackm.github.io to see what I do).
 
I appreciate the information you shared with me.
 
Dave Ackmann

 

image.png (view on web) <https://github.com/openscad/openscad/assets/1330241/2d4f01cd-b772-4c45-abb6-7fe2092c5b82>

t-paul, 2024-05-11:
Once the remaining issues are solved, #5114 will provide a way to filter for fonts having specific glyphs.

closed 2024-05-11 by t-paul