openscad/openscad #5131: Selection of line thickness for SVG/PDF export

Virindi-AC2, 2024-05-11:
I often export vector images of projections for the purpose of printing paper templates. While the thickness of exported lines seems to be a good default, it would be nice to be able to make them thinner in certain situations for more accuracy.

"Line width" seems like it would be a good option to add to the PDF export dialog, and perhaps future export dialogs.

jordanbrown0, 2024-05-14:
I'd think the default should be "hairline".

AaronVerDow, 2025-04-19:
Where should this option be defined?

I'm thinking of adding new special variables like $fn so the option could be defined through the CLI and within scripts.

AaronVerDow, 2025-04-19:
Per IRC discussion this will be an export option.

t-paul, 2025-04-19:
Unfortunately SVG has no export options yet, so that's going to be a bit more work. I started a while ago adding this, but that only went so far as an initial GUI file, so that's probably not worth much. I'll attach it here anyway, just in case...

[ExportSvgDialog.ui.txt](https://github.com/user-attachments/files/19822933/ExportSvgDialog.ui.txt)

jordanbrown0, 2025-04-21:
If we ever do
[OEP7: Mixed Dimension Geometry Support](https://github.com/openscad/openscad/wiki/OEP7%3A-Mixed-Dimension-Geometry-Support)
then line widths would be a good component.

t-paul, 2025-04-21:
> I'm digging into how 3MF handles colors to see if multi-colored output based on SCAD code would be possible. I'm still pretty new to the code base, so I may leave multi-color mode as a future improvement.

At this point, the 2D subsystem does not support retaining color across CSG operations.

AaronVerDow, 2025-04-22:
Thanks for the tip on 2D not supporting color.

I have this working via CLI using `parse_hex_color`, but I would like to leverage webcolors as well. `builtin_color` from `core/ColorNode.cc` seems to have the logic I want, but it's not for anything but the `color()` scad command.

Should I stick to hex, expose `builtin_color`, or is there some solution I'm not aware of?

t-paul, 2025-04-22:
Exposing `builtin_color` directly is probably not useful due to the specialized module invocation parameters. But extracting all the string->color code and the `webcolors` list itself and moving it into `ColorUtil.cc` seems like a viable option. It looks like `pyfunctions.cc` is poking directly into the `webcolors` list too. I hope that would easily translate to the new general string->color mapping too.

closed 2025-07-20 by t-paul

AaronVerDow, 2025-07-20:
SVG options still need to be done, I plan on starting on that next.