openscad/openscad #5132: Unable to launch

Jlc4BigRed, 2024-05-11:
When I launch Openscad it initially launches but when I either click new or load up a pre existing file it will open it but then immediately freeze.

t-paul, 2024-05-11:
1) Check if "Design -> Automatic Reload and Preview" is checked, if it is, try disabling it.

2) Please add at least the following information from the bug report form:

**Environment and Version info (please complete the following information):**
 - OS: [e.g. Windows 10, macOS 12 Monterey, Ubuntu 22.04]
 - System: [e.g. Intel PC 64-bit, Raspberry PI 4]
 - OpenSCAD Version [e.g. 2021.01, 2023.04.16] and where it was installed from

**Library & Graphics card information**
If applicable, add the content of the Help->Library Info window, or the output of running the `openscad --info` command.

Jlc4BigRed, 2024-05-11:
I am unable to get into any windows/menus. The only one I am able to use is the Welcome window. Once I click either new or load a pre-existing it just freezes and then unable to do anything. 

Windows 10 Pro
64 Bit
OpenSCAD 2021.01

[image]

t-paul, 2024-05-12:
This sounds like an issue with the graphics driver. Did it work before or is that the first installation of OpenSCAD? Did any Windows updates run lately and was the system rebooted afterwards? Is that a "normal" PC or a virtualized system of some sort?

It would be helpful to see the info output:

1) In a CMD or Powershell window, navigate to the folder where OpenSCAD is installed.
2) Run `openscad.com --info > info.txt`
3) Attach the file here, or open it in a text editor and copy&paste the content (use the "slash commands" / "code block" from the editor menu icons for formatting before pasting)

Jlc4BigRed, 2024-05-12:
It has run on this PC at one point. 

OpenSCAD Version: 2021.01
System information: Windows(TM) 10.0 SP 0.0 NTW 1 MSDN 724833 12 CPUs 15.90 GB RAM
User Agent: OpenSCAD/2021.01 (Windows(TM) 10.0 SP 0.0 NTW 1 MSDN 724833)
Compiler: GCC "7.4.0" 64bit
MinGW build: MingW64
Debug build: No
Boost version: 1_60
Eigen version: 3.3.7
CGAL version, kernels: 4.11, Cartesian<Gmpq>, Extended_cartesian<Gmpq>, Epeck
OpenCSG version: OpenCSG 1.4.2
Qt version: 5.13.0
QScintilla version: 2.11.2
InputDrivers: 
GLib version: 2.50.2
lodepng version: 20180910
libzip version: 1.5.2
fontconfig version: 2.13.1
freetype version: 2.10.1
harfbuzz version: 2.4.0
cairo version: 1.16.0
lib3mf version: 1.8.0
Application Path: C:/Program Files/OpenSCAD
Documents Path: C:\Users\Curti\OneDrive\Documents
User Documents Path: C:\Users\Curti\OneDrive\Documents
Resource Path: C:/Program Files/OpenSCAD
User Library Path: C:/Users/Curti/OneDrive/Documents/OpenSCAD/libraries
User Config Path: C:\Users\Curti\AppData\Local/OpenSCAD
Backup Path: C:/Users/Curti/OneDrive/Documents/OpenSCAD/backups
OPENSCADPATH: <not set>
OpenSCAD library path:
  C:/Users/Curti/OneDrive/Documents/OpenSCAD/libraries
  C:/Program Files/OpenSCAD\libraries

OPENSCAD_FONT_PATH: <not set>
OpenSCAD font path:
  C:/WINDOWS/fonts
  C:/Users/Curti/.local/share/fonts
  C:/Users/Curti/.fonts

GLEW version: 1.12.0
OpenGL Version: 4.6.0 NVIDIA 546.65
GL Renderer: NVIDIA GeForce RTX 3060/PCIe/SSE2
GL Vendor: NVIDIA Corporation
RGBA(8888), depth(24), stencil(8)
GL_ARB_framebuffer_object: yes
GL_EXT_framebuffer_object: yes
GL_EXT_packed_depth_stencil: yes
GL context creator: WGL
PNG generator: lodepng
OS info: Microsoft(TM) Windows(TM) 10 0 19045 0x360ecc4 amd64
Machine: 8664

Jlc4BigRed, 2024-05-17:
Just double checked and graphic drivers are all up to date.

kintel, 2024-05-19:
Could you also try the latest development snapshot for Windows?

t-paul, 2024-05-19:
There were cases reported where the application got stuck in a similar way after it ran on a 2nd monitor and saved that window position.

You could try removing the configuration from the Registry in `HKEY_CURRENT_USER\Software\OpenSCAD` and/or `HKEY_LOCAL_MACHINE\Software\OpenSCAD` (specifically the `window` settings).

Jlc4BigRed, 2024-05-19:
That fixed it!

Jlc4BigRed, 2024-05-19:
That fixed the issue.

closed 2024-05-19 by Jlc4BigRed