openscad/openscad #5137: Latest macOS snapshot crashes when rendering examples/Advanced/GEB with Manifold

kintel, 2024-05-16:
1. Download OpenSCAD-2024.05.13 for macOS and launch it
2. Open Examples/Advanced/GEB.scad
3. Enable Manifold
4. Render (F6)

-> Crash

kintel, 2024-05-17:
Reproducible also by running `ctest -C Examples`:
```
         55 - cgalpngtest_GEB (Failed)
        104 - opencsgtest_GEB (Failed)
        118 - opencsgtest_projection (Failed)
        120 - opencsgtest_text_on_cube (Failed)
        131 - opencsgtest_example009 (Failed)
        153 - throwntogethertest_GEB (Failed)
        167 - throwntogethertest_projection (Failed)
        180 - throwntogethertest_example009 (Failed)
        198 - csgpngtest_module_recursion (Failed)
        202 - csgpngtest_GEB (Failed)
        215 - csgpngtest_logo_and_text (Failed)
```

kintel, 2024-05-17:
We should probably run the examples tests automatically, perhaps nightly on the master branch if they're too heavy to run on each PR push

kintel, 2024-05-17:
The GEB crash seems to be a crash during a call to `Manifold::Project()`. Crash happens in the underlying Clipper2 library.

kintel, 2024-05-18:
Stack trace:
```
OpenSCAD!Clipper2Lib::ClipperBase::Split(Clipper2Lib::Active&, Clipper2Lib::Point<long long> const&) (/Users/kintel/code/OpenSCAD/openscad/build/_deps/clipper2-src/CPP/Clipper2Lib/src/clipper.engine.cpp:2746)
OpenSCAD!Clipper2Lib::ClipperBase::UpdateEdgeIntoAEL(Clipper2Lib::Active*) (/Users/kintel/code/OpenSCAD/openscad/build/_deps/clipper2-src/CPP/Clipper2Lib/src/clipper.engine.cpp:1742)
OpenSCAD!Clipper2Lib::ClipperBase::DoHorizontal(Clipper2Lib::Active&) (/Users/kintel/code/OpenSCAD/openscad/build/_deps/clipper2-src/CPP/Clipper2Lib/src/clipper.engine.cpp:2650)
OpenSCAD!Clipper2Lib::ClipperBase::ExecuteInternal(Clipper2Lib::ClipType, Clipper2Lib::FillRule, bool) (/Users/kintel/code/OpenSCAD/openscad/build/_deps/clipper2-src/CPP/Clipper2Lib/src/clipper.engine.cpp:2098)
OpenSCAD!Clipper2Lib::ClipperD::Execute(Clipper2Lib::ClipType, Clipper2Lib::FillRule, std::__1::vector<std::__1::vector<Clipper2Lib::Point<double>, std::__1::allocator<Clipper2Lib::Point<double> > >, std::__1::allocator<std::__1::vector<Clipper2Lib::Point<double>, std::__1::allocator<Clipper2Lib::Point<double> > > > >&, std::__1::vector<std::__1::vector<Clipper2Lib::Point<double>, std::__1::allocator<Clipper2Lib::Point<double> > >, std::__1::allocator<std::__1::vector<Clipper2Lib::Point<double>, std::__1::allocator<Clipper2Lib::Point<double> > > > >&) (/Users/kintel/code/OpenSCAD/openscad/build/_deps/clipper2-src/CPP/Clipper2Lib/include/clipper2/clipper.engine.h:602)
OpenSCAD!Clipper2Lib::ClipperD::Execute(Clipper2Lib::ClipType, Clipper2Lib::FillRule, std::__1::vector<std::__1::vector<Clipper2Lib::Point<double>, std::__1::allocator<Clipper2Lib::Point<double> > >, std::__1::allocator<std::__1::vector<Clipper2Lib::Point<double>, std::__1::allocator<Clipper2Lib::Point<double> > > > >&) (/Users/kintel/code/OpenSCAD/openscad/build/_deps/clipper2-src/CPP/Clipper2Lib/include/clipper2/clipper.engine.h:593)
OpenSCAD!Clipper2Lib::Union(std::__1::vector<std::__1::vector<Clipper2Lib::Point<double>, std::__1::allocator<Clipper2Lib::Point<double> > >, std::__1::allocator<std::__1::vector<Clipper2Lib::Point<double>, std::__1::allocator<Clipper2Lib::Point<double> > > > > const&, Clipper2Lib::FillRule, int) (/Users/kintel/code/OpenSCAD/openscad/build/_deps/clipper2-src/CPP/Clipper2Lib/include/clipper2/clipper.h:111)
OpenSCAD!manifold::CrossSection::CrossSection(std::__1::vector<std::__1::vector<glm::vec<2, float, (glm::qualifier)0>, std::__1::allocator<glm::vec<2, float, (glm::qualifier)0> > >, std::__1::allocator<std::__1::vector<glm::vec<2, float, (glm::qualifier)0>, std::__1::allocator<glm::vec<2, float, (glm::qualifier)0> > > > > const&, manifold::CrossSection::FillRule) (/Users/kintel/code/OpenSCAD/openscad/submodules/manifold/src/cross_section/src/cross_section.cpp:278)
OpenSCAD!manifold::Manifold::Impl::Project() const (/Users/kintel/code/OpenSCAD/openscad/submodules/manifold/src/manifold/src/face_op.cpp:319)
OpenSCAD!manifold::Manifold::Project() const (/Users/kintel/code/OpenSCAD/openscad/submodules/manifold/src/manifold/src/manifold.cpp:814)
OpenSCAD!ManifoldGeometry::project() const (/Users/kintel/code/OpenSCAD/openscad/src/geometry/manifold/ManifoldGeometry.cc:209)
OpenSCAD!GeometryEvaluator::projectionNoCut(ProjectionNode const&) (/Users/kintel/code/OpenSCAD/openscad/src/geometry/GeometryEvaluator.cc:974)
```

kintel, 2024-05-18:
@pca006132 I can open a Manifold ticket, unless you think this is a known issue

pca006132, 2024-05-19:
this is unknown to us.

kintel, 2024-05-19:
Opened Manifold issue: https://github.com/elalish/manifold/issues/813

It's currently unknown if this is a plain bug in Clipper, if Manifold's clipper is too old or if Manifold's usage of Clipper is causing this.

PR #5140 (2024-05-26): Update manifold
Fixes #5137 and #5142
closed 2024-05-26 by kintel