openscad/openscad #5145: Disable hardware GPU acceleration

ygoe, 2024-05-19:
**Is your feature request related to a problem? Please describe.**
OpenSCAD uses hardware GPU acceleration to render the preview. In theory, that's nice because it should speed things up. In practice, AMD's graphics driver is broken beyond repair. It crashes all the time when the GPU is actually used for a minute. And it won't recover until I reboot the entire system. So OpenSCAD is unusable on my new computer. I don't trust other devices and drivers as well. My new laptop (also AMD) has similar rendering errors (albeit not crashes) and my work laptop (Intel) also has render errors in Outlook and other apps all the time.

**Describe the solution you'd like**
Every software should have a toggle to turn off GPU acceleration and only use the CPU for calculations instead. Modern CPUs are also fast and way more stable than that experimental GPU stuff.

Almost all other apps I use do have this switch and it made them usable again. The system only crashes when I must use applications that do get through to the GPU. My CPU (Ryzen 7700) is fast enough, I haven't noticed any performance loss anywhere. GPU not needed.

**Describe alternatives you've considered**
None available. Newer AMD drivers have been even worse, crashing quicker and harder. Maybe buying new GPU hardware, but that'll be expensive because low-end discrete graphics cards are no longer available. You can only get high-end gamer stuff.

**Additional context**
I guess this feature request won't help me, unless it already exists, considering the release cycles of this tool.

I'd also be happy if someone could help me disabling all GPU usage on the system level (Windows 11) altogether, so that I don't need to tell it every app separately. But that also doesn't seem to exist. Uninstalling the driver doesn't help because Windows will automatically install it again within minutes.

pca006132, 2024-05-25:
I have never heard of GPU drivers being this bad on Windows... I usually encounter people complaining about the drivers on Linux, and one can change the mesa backend to llvmpipe on linux easily. Not sure about that for windows.

kintel, 2024-05-25:
The challenge is that, to disable GPU usage, we need to find a CPU driver for OpenGL. The one which ships with Windows is ancient and no usable, since Microsoft pretty much ditched OpenGL. It might be possible to install a CPU driver manually. If you're able to do that, we might be able to offer driver selection in OpenSCAD. Do you know how other apps do this?

ygoe, 2024-05-25:
No, never heard of this. I only know that I can select a software renderer in Affinity apps (called "WARP") and can disable hardware rendering in many other apps simply by selecting a checkbox or passing a commandline argument. Including all web browsers. No manual installations or drivers needed. .NET WPF also allows this by an option, it's based on DirectX so I'd have thought that it's a standard feature. But I know nothing about OpenGL.

Scopeuk, 2024-05-27:
I believe that it is possible to use https://github.com/pal1000/mesa-dist-win to redirect opengl rendering to a software renderer (this can be done at application launch overriding the system opengl bindings).
I am currently using AMD gpu drivers on windows with openscad without issue.

ygoe, 2024-05-27:
@Scopeuk May I ask which GPU model you're using? Mine is the integrated Ryzen 7700 CPU/GPU.

Scopeuk, 2024-05-27:
I am running a vega56

closed 2026-01-11 by kintel