openscad/openscad #5147: Customizer precision syntax causes default value to be changed

MethylBromide, 2024-05-20:
**Describe the bug**
Customizer "step size" specification is sometimes ignored in displaying the variable value, showing a full 7 digits precision when only (for example) two are needed.

**To Reproduce**
Steps to reproduce the behavior:
1. Create a new script in the OpenSCAD client and enter the following code:
`root = 0.28; // 0.01`
2. Observe what appears in the Customizer panel:
[image]
The desired precision as expressed through the step size is ignored, showing a precision of 7 digits after the decimal point, even though two digits is enough to display any valid value.

Note: the spinner controls do adjust the amount by the expected value.

**Expected behavior**
If the step size in the Customizer formatting comment contains X digits following the decimal point, X digits precision should be displayed.

This issue doesn't occur if the value is one that can be precisely expressed in floating point, e.g. .25 rather than .28. So I believe this is due to trying to compare floating point numbers without taking rounding errors into account.

**Environment and Version info (please complete the following information):**
 - OS: Windows 11
 - System: Intel PC 64-bit
 - OpenSCAD Version: 2024.03.22 (git 968835698) from OpenSCAD website.

UBaer21, 2024-05-20:
confirmed with 2024.05.11 only happens with certain floating values
like 0.07 or 0.081 0.082  interesting is that after you change the value in the customizer ("new set") changing it in the script still changes the number of digits. Which is much more to worry about
`root = +0.07;`  will make the customizer step 10⁻⁷  changing the value in the customizer to 0.071 will `echo(root)` confirm 0.071 if you now change the value in the code the echoed value changes to 0.07 (or 0.069 to 0.07) which is ok to see the customizer is limited and round to the precision given in the code. 
But it is a bit confusing that the stored set from the customizer is changed and rounded when you change the value in the code.

e.g 
`root = 1.2345;` ↦ customizer change to 1.5  ↦ code change to ` root=1; `  value is now 2

krhackbarth, 2025-02-15:
I don't know if this violates the assumptions above, but I have lines of code that look like this:

```
status_bar_height = 0.0; // .1
```
 
I see this as:
**status bar height**            0.0

But I have a customer who sees this as:
**status bar height**            0.0000000

It would seem that 0.0 can be "precisely expressed".

closed 2026-09-04 by github-actions[bot]