openscad/openscad #5148: Functions as input for the scale parameter in linear_extrude

Nexusnui, 2024-05-20:
**Is your feature request related to a problem? Please describe.**
Sometimes I want to extrude a 2D shape and want it to transiton in a curve.

**Describe the solution you'd like**
Let the scale parameter accept a function as input. The a provided function should accept either a progress or heigth parameter as input and return a scale for that heigth.

**Additional context**
I have implemented a version of this in the library "Variable Extrude" on Thingiverse, but a native version would be an improvement.
[The Library on Thingiverse](https://www.thingiverse.com/thing:6629632)

Some example pictures using `sin((z*360+45)*2)*0.25+0.75` as a scaling function, where z is the progress variable:
Square only scaled on x axis:
[image]
Square  scaled on both axes:
[image]
Circle only scaled on x axis:
[image]
Circle scaled on both axes:
[image]

Some other examples that also use the twist parameter:
[image]
[image]

Nexusnui, 2024-11-22:
This might be interesting, since this could be used for other manipulations, but I am not sure how slices should be generated, if top and bottom are different shapes(wich is possible with your suggestion).
The function should probably take take two arguments total_heigth and current_height.
I have not updated the file posted on Thingiverse yet, but I have changed the function to accept two parameters.

p-v-n, 2024-11-23:
Yeah, the above is just a rough suggestion to illustrate the idea.
Supplying total_height and current_height makes perfect sense, and would allow the function to produce the shape it is designed to.

gsohler, 2024-11-23:
This is yet already possible with PythonSCAD (pythonscad.org)

https://imgur.com/a/MmxRMFk

[image: image.png]

Restriction here is that each slice should have same number of points
If you cannot satisfy this restriction, PythonSCAD has a loft function, too.

jordanbrown0, 2024-11-23:
It's also not that hard to DIY a linear extrude that consumes a point list, and then you can do whatever you like to the point list on each layer.  If you're interested, send e-mail on the mailing list and I'll post a sample implementation.