openscad/openscad #5150: textmetrics does not work from command line

ostat, 2024-05-25:
**Describe the bug**
the **Textmetics** function works when the UI is open but when executed from the command line it returns undefined.

**To Reproduce**
Steps to reproduce the behavior:
1. create a script that uses the **textmetrics** function
```js
textvalue = "sample text!!!!";
font = "Liberation:style=Bold";
fontSize = 6.5;
ver = version();
matrics = textmetrics(text=textvalue,size=fontSize,font=font);
echo(matrics=matrics, version=ver);

linear_extrude(5)
text(
  str(textvalue, " size='", matrics.size, "' version=", ver),
  size=fontSize,
  font=font);
```
3. Click render in the UI and see that the size is returned as expected.
[image]
4. Create a script (in my example powershell) to run the above openscad script via the command line.
```ps
$script:ScadExePath = 'C:\Program Files\OpenSCAD\openscad.exe'
$script:SourceFolder = (Get-Item $MyInvocation.MyCommand.Source).Directory.FullName
$script:ScadScriptPath = Join-Path $SourceFolder 'test_text_length.scad'
$script:target = Join-Path $SourceFolder 'test_text_length.stl'

$args = ""
$args += "-o `"$($script:target)`""
$args += " $($script:ScadScriptPath)"

write-host $args

$executionTime = $args | Measure-Command { 
    write-host 'starting'
    $output = ''
    Start-Process $script:ScadExePath -ArgumentList $_ -wait | Tee-Object -Variable output
    Write-Host "done`r`n$($output)"
}
```
6. See that the text size is not rendered.
[image]

**Expected behavior**
I expected to be able to get the text metrics while executing the script via the command line

**Environment and Version info (please complete the following information):**
 - OS: Windows 11
 - System: Intel PC 64-bit
 - OpenSCAD Version 2023.05.20 installed from official site

t-paul, 2024-05-25:
It's flagged experimental, you need to `--enable textmetrics`.

closed 2024-05-25 by t-paul

ostat, 2024-05-25:
> It's flagged experimental, you need to `--enable textmetrics`.

Thanks that worked.

```ps
$script:ScadExePath = 'C:\Program Files\OpenSCAD\openscad.exe'
$script:SourceFolder = (Get-Item $MyInvocation.MyCommand.Source).Directory.FullName
$script:ScadScriptPath = Join-Path $SourceFolder 'test_text_length.scad'
$script:target = Join-Path $SourceFolder 'test_text_length.stl'

$args = ""
$args += "-o `"$($script:target)`""
$args += " --enable textmetrics"
$args += " $($script:ScadScriptPath)"

write-host $args

$executionTime = $args | Measure-Command { 
    write-host 'starting'
    $output = ''
    Start-Process $script:ScadExePath -ArgumentList $_ -wait | Tee-Object -Variable output
    Write-Host "done`r`n$($output)"
}
```

[image]

jordanbrown0, 2024-05-25:
We should really un-experimental-ize it.