openscad/openscad #5151: Render missing parts of the model

tcurdt, 2024-05-27:
**Describe the bug**

The model works fine in the preview.
The render does not give an error - it's just that the thread disappears.

**To Reproduce**

Open the file. Press Preview. Press Render.

**Expected behavior**

It should render.

**Code reproducing the issue**

It's not self-contained. It depends on list comprehension and scad utils.

**Screenshots**

Preview:
[image]

Render:
[image]

**Environment and Version info:**
 - OS: macOS 14.4.1 (23E224)
 - System: M1
 - OpenSCAD Version 2024.05.24 (git 8a890a9e1) (weekly)

[testcase.scad.zip](https://github.com/openscad/openscad/files/15449840/testcase.scad.zip)

kintel, 2024-05-27:
This fails with the following error:
```
ERROR: CGAL error in CGALUtils::applyUnion3D: CGAL ERROR: precondition violation!
Expr: ! _is_valid (_succ2_P) || comp_f (node1_P->object, _succ2_P->object) != LARGER
File: /Users/kintel/code/OpenSCAD/libraries/install/include/CGAL/Multiset.h
Line: 2346
```

Not sure exactly what caused CGAL to barf on it, but I rendered the same with Manifold enabled without any issues.

tcurdt, 2024-05-27:
I saw that error only once. I guess some kind of caching?

Oh, cool. I didn't know about the manifold.
That does indeed seems to fix it.

tcurdt, 2024-05-27:
Not sure whether that means to close the ticket or not.

kintel, 2024-05-27:
Let's keep it open. Until Manifold becomes the default, it's useful to track known CGAL issues.

jordanbrown0, 2024-11-23:
@tcurdt When you're having mysterious stuff happen, do Design/Flush Caches before test runs.  Unfortunately, caching caches the results (usually empty) of cases where there were problems, with the effect being that a subsequent render silently has missing pieces.