openscad/openscad #5160: can not open some *off file

sugizo, 2024-05-30:
**Describe the bug**
can not open some *off file

**To Reproduce**
convert stl (1.8 mb) 
into off file (1.2 mb) 
to make file size smaller

**Expected behavior**
can open off file converted, like in meshlab

**Screenshots**
[image]

**Environment and Version info (please complete the following information):**
 - OS: windows 11
 - System: Intel PC 64-bit
 - OpenSCAD Version 2021.01

**Additional context**
can open with meshlab

[image]

kintel, 2024-05-30:
Can you provide an example?

sugizo, 2024-05-30:
attached the off files that can not be opened in openscad
but
can opened in meshlab
[rotterdam.zip](https://github.com/openscad/openscad/files/15501100/rotterdam.zip)

and here is the screen capture the same file with *.stl formated 
that can be opened by openscad 
[image]

tldr
after several process in the end, 
can save the processed in some format, 
when try *.stl it can be opened in openscad 
but 
when choose *.off it can not be opened in openscad (in meshlab is success opened), 

tried several sources too for another city : 
den haag, montreal, vienna, etc
all of it can be opened when save as *.stl
but 
can not be opened when save as *.off

trying several converter too, 
but again same result

problem with the *.off file generated ?
no, 
because the same file *.off can be opened with meshlab, 
like posted on first reported above

best regards

kintel, 2024-05-30:
Did you click on View All after importing and rendering?

sugizo, 2024-05-30:
already click `view all (ctrl + shift + v)` 
in openscad
but still can't view *.off file

kintel, 2024-05-30:
This works for me using the development snapshot. Could you try with the latest snapshot too?

sugizo, 2024-05-31:
fixed in development version, 
thanks n best regards

closed 2024-05-31 by sugizo