openscad/openscad #5166: Customizer doesn´t allow fractions for parameters which are integers

nomike, 2024-06-08:
**Describe the bug**
OpenSCAD tries to guess the type of a variable and uses it for the customizer. If a parameter's value in the code is "false" for example, the customizer detects this as a boolean and displays a check-box.

Strings are represented with a text-input field.

For numbers, it is distinguished between integers and floats. This however seems to happen by checking the value rather than the way it is written in the code.

So if the design value is 10.0 for example, the customizer will display it as an integer and will not allow you to specify commas.

Rather than using some hacks like setting the design value to 10.000001 to force the customizer to treat it as a float, there is no way to set the value I intend.

**To Reproduce**
Steps to reproduce the behavior:
1. Create a design with a variable and set it's value to 10.0
2. Try to change the value to 10.3 in the customizer

**Expected behavior**
I would expect openscad to honor the "10.0" original format and treat the field as a float.

**Code reproducing the issue**

```openscad
size = 10.0;
sphere(size);
```

**Screenshots**
Not applicable.

**Environment and Version info (please complete the following information):**
 - OS: Ubuntu 23.04
 - System: Lenovo Thinkpad T14 Gen1 AMD
 - OpenSCAD Version: version 2024.05.24, installed from source

**Library & Graphics card information**
Not applicable.

**Additional context**
Not applicable.

UBaer21, 2024-06-08:
`size = 10; //.1`  or `//(.1)`
https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/Customizer#Spinbox

nomike, 2024-11-09:
Ah, yes, I overlooked that. Thank you.

closed 2024-11-09 by nomike