openscad/openscad #5169: Unable to return to "Preview" (F9) view mode after "Thrown Together" (F12) view mode is selected.

gargagecan, 2024-06-09:
**Describe the bug**
Unable to return to "Preview" (F9) view mode after "Thrown Together" (F12) view mode is selected.

**To Reproduce**
Steps to reproduce the behavior:
1. Start OpenSCAD
2. Select a project that already exists
3. Project will be displayed in "Preview" view mode
5. Press F12 key for "Thrown Together" view mode
6. Press F9 to return to "Preview mode
7. View mode will not change. Additionally, Clicking on "View" menu option will show that "Preview F9" option is greyed out.

**Expected behavior**
Pressing F9 should cause view mode to return to "Preview" view mode.

**Environment and Version info (please complete the following information):**
 - OS: Windows 10 Pro
 - System: Intel 64-bit PC
 - OpenSCAD: Version 2024.06.04 (git 6db5adc0c).  Nightly snapshot from OpenSCAD.org

**Library & Graphics card information**
Output from `openscad --info` command:

OpenSCAD Version: 2024.06.04 (git 6db5adc0c)
System information: Microsoft Windows 10 (10.0.19045) x86_64 20 CPUs 63.92 GB RAM
User Agent: OpenSCAD/2024.06.04 (git 6db5adc0c) (Microsoft Windows 10 (10.0.19045) x86_64)
Compiler: GCC "13.2.0" 64bit
MinGW build: MingW64
Debug build: No
Boost version: 1_81
Eigen version: 3.4.0
CGAL version, kernels: 5.5, Cartesian<Gmpq>, Extended_cartesian<Gmpq>, Epeck
OpenCSG version: OpenCSG 1.5.0
Qt version: 5.15.11
QScintilla version: 2.11.2
InputDrivers: 
GLib version: 2.70.2
lodepng version: 20230410
libzip version: 1.5.2
fontconfig version: 2.14.2
freetype version: 2.13.2
harfbuzz version: 7.3.0
cairo version: 1.16.0
lib3mf version: 1.8.1
Features: fast-csg, fast-csg-safer, fast-csg-debug, manifold, roof, input-driver-dbus, lazy-union, vertex-object-renderers-indexing, textmetrics, import-function, predictible-output
Application Path: C:/Program Files/OpenSCAD
Documents Path: C:\Users\merlin\Documents
User Documents Path: C:\Users\merlin\Documents
Resource Path: C:/Program Files/OpenSCAD
User Library Path: C:/Users/merlin/Documents/OpenSCAD/libraries
User Config Path: C:\Users\merlin\AppData\Local/OpenSCAD
Backup Path: C:/Users/merlin/Documents/OpenSCAD/backups
OPENSCADPATH: C:\Program Files\OpenSCAD\libraries;My Documents\OpenSCAD\libraries
OpenSCAD library path:
  C:/Program Files/OpenSCAD/libraries
  C:/Windows/System32/My Documents/OpenSCAD/libraries
  C:/Users/merlin/Documents/OpenSCAD/libraries
  C:/Program Files/OpenSCAD\libraries

OPENSCAD_FONT_PATH: <not set>
OpenSCAD font path:
  C:/Windows/fonts
  C:/Users/merlin/AppData/Local/Microsoft/Windows/Fonts
  C:/Users/merlin/.local/share/fonts
  C:/usr/local/share/fonts
  C:/usr/share/fonts
  C:/Users/merlin/.fonts
  C:/usr/X11/lib/X11/fonts
  C:/System/Library/Fonts
  C:/Library/Fonts
  C:/Users/merlin/Library/Fonts

GL context creator: WGL (old)
PNG generator: lodepng

GLEW version: 2.1.0
OpenGL Version: 4.6.0 NVIDIA 536.23
GL Renderer: NVIDIA GeForce RTX 2060 SUPER/PCIe/SSE2
GL Vendor: NVIDIA Corporation
RGBA(8888), depth(24), stencil(8)
GL_ARB_framebuffer_object: yes
GL_EXT_framebuffer_object: yes
GL_EXT_packed_depth_stencil: yes

mbenert, 2024-10-30:
I have the same issue on MacOS with OpenSCAD-2024.10.28.
```
OpenSCAD --info
QObject::startTimer: Timers can only be used with threads started with QThread
OpenSCAD Version: 2024.10.28 (git 7f05c6294)
System information: macOS 14.6.1 arm64 MacBookPro18,2 10 CPUs 64.00 GB RAM 
User Agent: OpenSCAD/2024.10.28 (git 7f05c6294) (macOS 14.6.1 arm64 MacBookPro18,2)
Compiler: Clang "15.0.0 (clang-1500.3.9.4)"
MinGW build: No
Debug build: No
Boost version: 1_81
Eigen version: 3.4.0
CGAL version, kernels: 6.0, Cartesian<Gmpq>, Extended_cartesian<Gmpq>, Epeck
OpenCSG version: OpenCSG 1.6.0
Qt version: 5.15.13
QScintilla version: 2.13.3
InputDrivers: 
GLib version: 2.81.0
lodepng version: 20230410
libzip version: 1.9.2
fontconfig version: 2.14.1
freetype version: 2.12.1
harfbuzz version: 6.0.0
cairo version: 1.18.0
lib3mf version: 2.3.1
Features: roof, input-driver-dbus, lazy-union, vertex-object-renderers-indexing, textmetrics, import-function, predictible-output
Application Path: /Applications/OpenSCAD-2024.10.28.app/Contents/MacOS
Documents Path: /Users/mike/Documents
User Documents Path: /Users/mike/Documents
Resource Path: /Applications/OpenSCAD-2024.10.28.app/Contents/Resources
User Library Path: /Users/mike/Documents/OpenSCAD/libraries
User Config Path: /Users/mike/Library/Application Support/OpenSCAD
Backup Path: /Users/mike/Documents/OpenSCAD/backups
OPENSCADPATH: <not set>
OpenSCAD library path:
  /Users/mike/Documents/OpenSCAD/libraries
  /Applications/OpenSCAD-2024.10.28.app/Contents/Resources/libraries

OPENSCAD_FONT_PATH: <not set>
OpenSCAD font path:
  /System/Library/Assets/com_apple_MobileAsset_Font3
  /System/Library/Assets/com_apple_MobileAsset_Font4
  /Users/mike/.local/share/fonts
  /usr/local/share/fonts
  /usr/share/fonts
  /Users/mike/.fonts
  /usr/X11/lib/X11/fonts
  /System/Library/Fonts
  /Library/Fonts
  /Users/mike/Library/Fonts
  /System/Library/Fonts/Supplemental

GL context creator: CGL (new)
PNG generator: Core Foundation

GLAD version: 2.0.4
OpenGL Version: 2.1 Metal - 88.1
GL Renderer: Apple M1 Max
GL Vendor: Apple
RGBA(8888), depth(24), stencil(8)
GL_ARB_framebuffer_object: yes
GL_EXT_framebuffer_object: yes
GL_EXT_packed_depth_stencil: yes
```

jordanbrown0, 2025-05-20:
Not reproduced on 2025.04.23 on Windows 11.

kintel, 2025-05-22:
Likely fixed by the menu cleanup in #5585

closed 2025-05-22 by kintel