openscad/openscad #5170: need to clean out old openscad installer and reset back to clean sheet

seanmahrt, 2024-06-10:
**Describe the bug**
the old install of openscad had some weird folder for my library and I moved the folder in windows and now openscad is angry.  Well, there's no way to change it, so I figured I'd just uninstall/reinstall.    yeah, even a hard drive search of openscad files and deleting them still has an angry setup.  not being able to define/redefine default folders even via a config text file is silly, but not being able to reset to factory default and get it to work is totally broken.

**To Reproduce**
windows 11
change some parameter or remove the library folder
uninstall
reinstall
cannot render or open the library folder

mation):**
 - OS: Windows 11
 - System: intel pc 64bit
 - OpenSCAD Version 2021.01

seanmahrt, 2024-06-10:
Trying 2024.06.04

jordanbrown0, 2024-06-11:
Your phasing is not conducive to convincing people who are all *volunteers* to help you.  You might want to remember that you can get a full refund of every penny that you paid for the program, at any time.

That said... OpenSCAD libraries on Windows live in one of three places:
* The directory named by the OPENSCADPATH environment variable.
* Documents\OpenSCAD\libraries
* \Program Files\OpenSCAD\libraries

Of those, only the first is directly under user control.  (Windows may allow you to move the Documents directory, and the installation lets you move the OpenSCAD installation directory, and of course moving those moves the "libraries" directory that is under them.)

Which one(s) did you delete or rename?

I have tried renaming the two automatically-created ones, and setting $OPENSCADPATH to a nonexistent path, and OpenSCAD 2021.01 still seems to run fine.

You say that OpenSCAD is "angry".  What exactly does it do?

seanmahrt, 2024-06-13:
Yeah, I'm part of the volunteer groups for a number of things and I get you.  Been a long day and I likely took some of that out on my bug report. 

I think the documents one got munged up with OneDrive triyng to be super pervasive and reinstalling itself every time I turned around and set the path into that folder.  the 2024 one worked out of the box so that's weird.

I've not seen anything about the environment variable in windows when I scoured for similar bugs.  I may try to put the old 2021 on there.

What seemed to happen is there was an OS error (probably non-existant path) error in the box and it refused to even render.  My guess is it wasn't happy it couldn't create a backup file maybe?

I'm trying to finish my project first and then I'll try reverting back to 2021.   maybe 2024 broke it out of it's funk....

Sean

jordanbrown0, 2024-06-13:
I tried renaming Documents\OpenSCAD, and on starting OpenSCAD it happily created a new one.

However, I could easily believe that if you moved Documents itself, without appropriately changing the Windows configuration, all kinds of chaos could result.  I think we do expect that the directory that Windows tells us to use for documents already exists.

seanmahrt, 2024-06-14:
Right.  I remapped the documents folder to a shared offline system.  I was pretty sure I didn’t do that after I installed openscad but OneDrive does all sorts of goofy things with my documents folder.  A config file (or a better documented environment variable) would likely let me fix it. 2024 is behaving like a champ.

Sean

jordanbrown0, 2024-06-14:
`$OPENSCADPATH` is documented on the Libraries page of the User Manual:  https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/Libraries

Environment variables are inconvenient enough on Windows that I think I'd prefer that `$OPENSCAD` was replaced by a preference item.  Other than that, I think there's a good mix of options:  a location completely under your control, a location relative to the user's system-standard home directory, and a location relative to the application installation directory.  Maybe there should be a system-wide location that's independent of the particular installation, but that's getting pretty exotic.

On Windows, I don't believe OpenSCAD uses *any* config files.  Preferences are all stored in the Registry.  Besides... if it was stored in a config file, where should OpenSCAD look for the config file?