openscad/openscad #5172: Allow selecting STL files to import via a file dialog

keithel, 2024-06-13:
**Is your feature request related to a problem? Please describe.**
It is very non-intuitive how to import an STL file into your project. If you don't know that OpenSCAD can do this, from the interface, you'd have no idea you can, or how to do so.

**Describe the solution you'd like**
I would like a way to bring up a QFileDialog that has STL files in a filter so that STL files will show in the dialog. Choosing an STL and accepting the file dialog would add the import statement to the editor's cursor's current line.

**Describe alternatives you've considered**
This could be implemented very simply by just adding a file filter for STL files to the openFile() utility function, however that may confuse the user, since it isn't exactly opening a new file.
Alternatively, a separate menu item and toolbar button could be created for importing STLs (File->Import probably), which would call a different utility function that shows a file dialog specifically for importing.

**Additional context**
https://github.com/openscad/openscad/issues/4791 asks for any file to be openable in the editor. If for some reason the user wanted to open an STL file and have the STLs file contents shown in the editor window, this feature I am proposing would need to be implemented in some way other than with the existing openFile() utility function (and thus not through File->Open, or the open tool button).

keithel, 2024-06-13:
I have implemented alternative # 1 listed above. I will initiate the PR for it momentarily.

jordanbrown0, 2024-06-13:
>If you don't know that OpenSCAD can do this, from the interface, you'd have no idea you can, or how to do so.

While that's true, it's also true of literally everything else having to do with creating an OpenSCAD model, starting with how to add a cube to the model.  You must read the documentation and write a program.

I'm not really happy with the idea of saying that "Open" adds an `import()` statement.  I don't mind having some *other* menu item add an `import()` statement.  However, why stop at an `import()` statement?  Why not have a menu item that prompts for the parameters and adds a cube?

I'd say that the ease-of-use issues that you describe are way, way down the list of things that make OpenSCAD hard to use.

Note also that a scheme using a file-open dialog does not know whether it makes sense to use relative or absolute paths.  If it picks wrong, your model may stop working when you move it from one directory to another.  (E.g.:  suppose that you have a directory for your model, and under it you have an `scad` directory for the OpenSCAD sources and a `stl` directory for the STLs.  If the tool uses an absolute path for the STL, moving that hierarchy somewhere else will fail.  Now suppose that you have a library of stock STL files in some directory, and models elsewhere.  If the tool uses relative paths, again if you move the models they may fail.  You might want an option in the dialog that says whether to use an absolute or relative path.

keithel, 2024-06-13:
Having it be another menu item seems fine to me (and probably more appropriate) -- I can certainly change my PR to do that.

"why stop there"? - file paths are something that GUIs are already good at handling. I think it's natural to assume you should be able to add an import with the help of the GUI, instead of just having to type it. If not, why does OpenSCAD have a GUI at all? why isn't it just fully command line, with it just spitting out an STL, STEP file, or some other 3d model file format as it's output, and perhaps a way to have it spit out an image that you can view with some other program?

Relative vs absolute paths -- This is something that the user can tweak after they've done the import. I certainly do this. IMHO it isn't a good reason for opposing adding a menu item and/or toolbar button that will import a file for you.

jordanbrown0, 2024-06-14:
I don't agree on the value, but if it doesn't overload "Open", I guess I don't care enough to argue about it.

kintel, 2024-06-17:
If we do this, we should probably do it for all the file formats supported by `import()`, not just STL.

keithel, 2024-06-18:
@kintel Yes, probably the right thing to do.

When I get a chance, I'll update my PR. I'll change it to having a separate File->Import menu item + toolbar button

t-paul, 2024-06-18:
I'd say no toolbar button, it's already crowded and that's not a hugely common feature I assume. We want to have the toolbar customizable at some point anyway, so users who use it often could then add the action themselves.

If it's a new menu entry anyway, would it still always open a new document or just insert the import()/surface()/use<> at the current cursor position in the open document just like drag&drop?

kintel, 2024-06-18:
I'd say import at cursor. But it's tricky to build this right, as it may not be a good place to put an import statement. It kind of feels like this sort of functionality would be the beginning of some sort of "shape palette" for people to be able to insert objects without having to read the manual first. Which could have good value, but needs significant design thinking.

t-paul, 2024-06-18:
I see the concerns, but then it's not uncommon to have an "insert external stuff" menu item for inserting images, links, ... If you have the cursor at the wrong place, you know the next time :grin:.

I do like the idea of a shape palette, like an extended version of the "insert code snippet" we have already.

keithel, 2024-06-19:
I think it should insert at cursor. Maybe it can insert on the line before/after instead of right at the column position where the cursor is.

I like the thoughts on the toolbar - makes sense to leave it out, and instead provide a means for the user to customize the toolbar the way they want. I might be able to help with the customizable toolbar. (I work for Qt btw - though this work is just my work in my spare time - I'm not intentionally representing Qt with this work).

I also *really* like the idea of a shape palette. But I agree - that will require significant design.

keithel, 2024-06-19:
Oh, ok - I think you must be referring to <Right Click context menu>->Insert Template.

jordanbrown0, 2025-10-24:
An extension of templates might be a good answer for the "add an import operation" goal.  Have a template that has special markup for "prompt the user to select a file, and put the file name here".

jordanbrown0, 2025-11-07:
See PR #6341 for a draft of this file-selecting-template mechanism.