openscad/openscad #5174: OpenSCAD does not respect macOS system keyboard shortcuts.

ipmcc, 2024-06-13:
**Describe the bug**

macOS has a feature where, even if the app you're using doesn't support custom key bindings, if something is in a menu, regardless of its preset keyboard shortcut you can create alternate mappings, per application. It's wildly useful for crusty old farts like me who have 35 years of muscle memory to contend with. 

This functionality can be found in **System Preferences > Keyboard > Shortcuts > App Shortcuts**.  Once there, you can add shortcuts to (most) apps. For instance I've been using other IDEs for years where the indent command is Cmd-Shift-Right, and dedent is Cmd-Shift-Left. OpenSCAD's defaults are indent as Cmd-I and dedent as Cmd-Shift-I.

When I set up these overrides, I can even see the key combo change in the menu, and the menu get activated when I press them, but the app doesn't respond, although weirdly, the initial key combos do. What this makes me suspect is that keyboard commands are not using the standard macOS responder chain correctly, but is rather trying to manage keyboard shortcuts itself.

**To Reproduce**
Steps to reproduce the behavior:
0. Note the before state of the menu:
[image]
1. Create a shortcut for Indent as described above in the system-native keyboard shortcuts.
2. Note the after state of the menu:
[image]

3. Enter 1 line of code `cube(10);` for instance  
4. Try to indent and dedent that line of code, using the new key combos.
6. Observe that the Edit menu flashed (meaning that the shortcut has sent that message up the responder chain through the macOS menu subsystem.)
7. Observe that the indents and dedents don't happen in the editor.

**Expected behavior**
I expect the new key combos to actually cause the desired indent and dedent operations in the editor

**Code reproducing the issue**
One line like:
`cube(10);` 
will allow you to repro this.

**Screenshots**
See screenshots above in the repro instructions.

**Environment and Version info (please complete the following information):**
 - OS: macOS 12.5
 - System: Retina 5K iMac (circa 2014/15), 4GHz quad-core i7
 - OpenSCAD Version: 2021.01 installed with `brew install openscad`

Obscuretone, 2024-06-15:
Confirmed on macOS 14.1 MacBook Air.  OpenSCAD 2024.06.14

[image]

Shift Command Left / Right does a "select everything to the left or right" in every application I've tried including Chrome and can't be overridden. 

Setting it to any unused shortcut works correctly in OpenSCAD
[image]

That shortcut does work correctly in Textedit.app, an Apple application. So it's _possible_

[image]

ipmcc, 2024-06-15:
I haven't looked at the OpenSCAD source code, but I spent the better part of a decade developing Mac applications at Apple. If I had to hazard a guess, just based on the behavior here, that OpenSCAD (or whatever UI framework it's based on) probably has its own key handling code, and handles key combinations without using the normal macOS mechanism. Also, it probably means that actions (i.e. Indent, etc) aren't being dispatched with messages being sent through the responder chain, as in a normal Mac App.

kintel, 2024-06-15:
OpenSCAD uses Qt as a UI framework and QScintilla as an editor component. Not sure how these handle system-level overrides.

ipmcc, 2024-06-15:
I assumed it was something like that. I've used Qt before (much to my chagrin). Qt is like all-season tires: passable, but not as good as summer and winter tires. (i.e. separate apps per platform/season.) [This thread](https://forum.qt.io/topic/97699/how-create-a-mac-document_based-app-use-qt-on-mac) digs into it a little, but not enough to tell you how to DO handle the responder chain in Qt. I suspect it will require the touch of someone versed in Mac app development. I haven't written code (other than some shell scripts) for a decade, so it probably isn't me. Sadly, it probably needs to be an "old fart" since Swift has changed so many things on Apple platforms, and people coming to the platform in the last 7 years or so aren't going to know Objective-C.

mnemnion, 2025-02-16:
This seems like the right place to add the following report, since it's likely to be the same underlying issue:

The 'globe' key which pops up the character picking menu does not load when pressed in OpenSCAD. It produces a rather long spinning beachball, but nothing else.  

Interestingly, the operating system thinks it's displayed, the old way to load it is a menu icon and that says either "Show emoji & symbols" or "Hide".  That toggles but the picker doesn't display.

Writing strings out in another editor and pasting them in is pretty straightforward, just something to check if and when someone wants to tackle this bug. 

I see that upgrading QT is one of the blocking issues for the next release, decent odds the problem goes away when that happens.

kintel, 2025-02-16:
@mnemnion We already upgraded to Qt6 for the macOS development snapshots; that doesn't fix this issue.

Most tricky key bindings in OpenSCAD are handled by our QScintilla-based editor component, and it's often hard to move the needle there. One good starting point could be to check whether the issue is present in other QScintilla-based programs, as that could be an indication that the issue is somewhat out of our control.

ipmcc, 2025-02-16:
Bear with me for a minute, because I'm going to say something that might feel a little inflammatory: In my experience, relying on a 'least common denominator' cross-platform framework like Qt tends to yield apps with 'least common denominator' user experiences. 

Has the team thought about integrating with a cross-platform programming UI with **much** more steam behind it: **VSCode**? This sort of integration could be relatively simple, with relatively little UI impact: A language binding/definition for their editor, a few hooks to watch files on disk for changes, and perhaps a few more hooks to switch back and forth between OpenSCAD and VSCode at the right times ought to do this trick. This could really be a match made in heaven! 

I recognize that OpenSCAD has _stunningly_ rich support for running on virtually any OS/container/hardware/etc out there, and VSCode is 'limited' to just Linux (Ubuntu & RedHat), Windows, MacOS, and various other uses via a variety of ancillary integrations, but I have to imagine that the overlap between the two would cover a _large_ majority of OpenSCAD users. Qt and its ilk have never seemed to care much about producing apps that feel platform native, whereas VSCode seems to have 'swallowed the hook' on being best-in-class with native feel on all its platforms, not to mention deep customization support. It has truly impressed me many times over the years.

Just a thought...

nophead, 2025-02-16:
You can use any external editor for OpenSCAD. It does look for file changes and render if any change. I use an editor I wrote myself.

kintel, 2025-02-16:
As nophead points out, there is already reasonably good support for external editors. There are also multiple LSP plugins available. The small OpenSCAD dev team has limited resources to work directly with individual editor extensions, but adding better support for external editors in OpenSCAD itself is definitely on the table. One of the main challenges is to choose a good, multi-platform, IPC mechanism for signaling between OpenSCAD and external editors and getting external editors to support that.

mnemnion, 2025-02-16:
@kintel I did some searching with Google and on GitHub, it didn't turn up a program which both uses QScintilla with QT6 and supports macOS at all. If you happen to know of another such program besides OpenSCAD, I would be happy to download it and try this.  I'd also be willing to try and build such a program locally, if it were known to work on macOS.

Personally I plan to get Neovim working comfortably with an OpenSCAD workflow and proceed on that basis.   My 'dream stack' for OpenSCAD would take the kernel and build a new UI using a platform like Tauri to run on the OS-native WebView, but I'm aware of how much work that would be and I'm definitely not asking 😅.  

Of the options which aren't that, Qt is probably the best cross-platform framework available.  Least-common-denominator, sure, but it's unreasonable to expect a volunteer-lead free software project like OpenSCAD to write three or more GUIs.

kintel, 2025-02-17:
Looks like this uses QScintilla: https://eric-ide.python-projects.org/
Might be a bit of a big tool though.

The good thing about OpenSCAD is that the GUI code is insulated from the rest of the code, so it should be doable to experiment with alternative solutions for GUI. In terms of using a web view, I think one of the challenges would be to get the GL code to work well with a webview-based viewer component.