openscad/openscad #5177: Can't build on OS X

Obscuretone, 2024-06-14:
**Environment**
Compiling on MacOS 14.1 M2 MacBook Air. Dependencies installed with homebrew. Followed instructions at https://github.com/openscad/openscad?tab=readme-ov-file#building-for-macos.

Any help appreciated

**Issue**

I can't run `./scripts/macosx-build-dependencies.sh` succesfully. 

`==> cairo: stable 1.18.0 (bottled), HEAD`
`==> freetype: stable 2.13.2 (bottled)` 

```
...
Found ninja-1.12.1 at /opt/homebrew/bin/ninja
INFO: autodetecting backend as ninja
INFO: calculating backend command to run: /opt/homebrew/bin/ninja -C /Users/obscuretone/Code/libraries/src/cairo-1.18.0/build-arm64
ninja: Entering directory `/Users/obscuretone/Code/libraries/src/cairo-1.18.0/build-arm64'
[116/148] Compiling C object src/libcairo.2.dylib.p/cairo-colr-glyph-render.c.o
FAILED: src/libcairo.2.dylib.p/cairo-colr-glyph-render.c.o 
cc -Isrc/libcairo.2.dylib.p -Isrc -I../src -I. -I.. -I/opt/homebrew/Cellar/fontconfig/2.15.0/include -I/opt/homebrew/opt/freetype/include/freetype2 -I/opt/homebrew/opt/libpng/include/libpng16 -I/opt/homebrew/Cellar/pixman/0.42.2/include/pixman-1 -I/Users/obscuretone/Code/libraries/install/include -fvisibility=hidden -fdiagnostics-color=always -Wall -Winvalid-pch -Wextra -O0 -g -Wmissing-declarations -Werror-implicit-function-declaration -Wpointer-arith -Wwrite-strings -Wsign-compare -Wpacked -Wswitch-enum -Wmissing-format-attribute -Wvolatile-register-var -Wstrict-aliasing=2 -Winit-self -Wno-missing-field-initializers -Wno-unused-parameter -Wno-attributes -Wno-long-long -Winline -Wno-unused-but-set-variable -Wno-enum-conversion -fno-strict-aliasing -fno-common -D_GNU_SOURCE -arch arm64 -mmacosx-version-min=10.13 -Wno-attributes -Wno-unused-but-set-variable -Wno-missing-field-initializers -Wno-unused-parameter -Wno-long-long -D_REENTRANT -DCAIRO_COMPILATION -MD -MQ src/libcairo.2.dylib.p/cairo-colr-glyph-render.c.o -MF src/libcairo.2.dylib.p/cairo-colr-glyph-render.c.o.d -o src/libcairo.2.dylib.p/cairo-colr-glyph-render.c.o -c ../src/cairo-colr-glyph-render.c
In file included from ../src/cairo-colr-glyph-render.c:37:
../src/cairo-ft-private.h:87:30: error: unknown type name 'FT_Color'
                             FT_Color               *palette,
                             ^
1 error generated.
[118/148] Compiling C object src/libcairo.2.dylib.p/cairo-ft-font.c.o
FAILED: src/libcairo.2.dylib.p/cairo-ft-font.c.o 
cc -Isrc/libcairo.2.dylib.p -Isrc -I../src -I. -I.. -I/opt/homebrew/Cellar/fontconfig/2.15.0/include -I/opt/homebrew/opt/freetype/include/freetype2 -I/opt/homebrew/opt/libpng/include/libpng16 -I/opt/homebrew/Cellar/pixman/0.42.2/include/pixman-1 -I/Users/obscuretone/Code/libraries/install/include -fvisibility=hidden -fdiagnostics-color=always -Wall -Winvalid-pch -Wextra -O0 -g -Wmissing-declarations -Werror-implicit-function-declaration -Wpointer-arith -Wwrite-strings -Wsign-compare -Wpacked -Wswitch-enum -Wmissing-format-attribute -Wvolatile-register-var -Wstrict-aliasing=2 -Winit-self -Wno-missing-field-initializers -Wno-unused-parameter -Wno-attributes -Wno-long-long -Winline -Wno-unused-but-set-variable -Wno-enum-conversion -fno-strict-aliasing -fno-common -D_GNU_SOURCE -arch arm64 -mmacosx-version-min=10.13 -Wno-attributes -Wno-unused-but-set-variable -Wno-missing-field-initializers -Wno-unused-parameter -Wno-long-long -D_REENTRANT -DCAIRO_COMPILATION -MD -MQ src/libcairo.2.dylib.p/cairo-ft-font.c.o -MF src/libcairo.2.dylib.p/cairo-ft-font.c.o.d -o src/libcairo.2.dylib.p/cairo-ft-font.c.o -c ../src/cairo-ft-font.c
In file included from ../src/cairo-ft-font.c:46:
../src/cairo-ft-private.h:87:30: error: unknown type name 'FT_Color'
                             FT_Color               *palette,
                             ^
1 error generated.
[120/148] Compiling C object src/libcairo.2.dylib.p/cairo-svg-glyph-render.c.o
FAILED: src/libcairo.2.dylib.p/cairo-svg-glyph-render.c.o 
cc -Isrc/libcairo.2.dylib.p -Isrc -I../src -I. -I.. -I/opt/homebrew/Cellar/fontconfig/2.15.0/include -I/opt/homebrew/opt/freetype/include/freetype2 -I/opt/homebrew/opt/libpng/include/libpng16 -I/opt/homebrew/Cellar/pixman/0.42.2/include/pixman-1 -I/Users/obscuretone/Code/libraries/install/include -fvisibility=hidden -fdiagnostics-color=always -Wall -Winvalid-pch -Wextra -O0 -g -Wmissing-declarations -Werror-implicit-function-declaration -Wpointer-arith -Wwrite-strings -Wsign-compare -Wpacked -Wswitch-enum -Wmissing-format-attribute -Wvolatile-register-var -Wstrict-aliasing=2 -Winit-self -Wno-missing-field-initializers -Wno-unused-parameter -Wno-attributes -Wno-long-long -Winline -Wno-unused-but-set-variable -Wno-enum-conversion -fno-strict-aliasing -fno-common -D_GNU_SOURCE -arch arm64 -mmacosx-version-min=10.13 -Wno-attributes -Wno-unused-but-set-variable -Wno-missing-field-initializers -Wno-unused-parameter -Wno-long-long -D_REENTRANT -DCAIRO_COMPILATION -MD -MQ src/libcairo.2.dylib.p/cairo-svg-glyph-render.c.o -MF src/libcairo.2.dylib.p/cairo-svg-glyph-render.c.o.d -o src/libcairo.2.dylib.p/cairo-svg-glyph-render.c.o -c ../src/cairo-svg-glyph-render.c
In file included from ../src/cairo-svg-glyph-render.c:39:
../src/cairo-ft-private.h:87:30: error: unknown type name 'FT_Color'
                             FT_Color               *palette,
                             ^
1 error generated.
[121/148] Compiling C object src/libcairo.2.dylib.p/cairo-script-surface.c.o
FAILED: src/libcairo.2.dylib.p/cairo-script-surface.c.o 
cc -Isrc/libcairo.2.dylib.p -Isrc -I../src -I. -I.. -I/opt/homebrew/Cellar/fontconfig/2.15.0/include -I/opt/homebrew/opt/freetype/include/freetype2 -I/opt/homebrew/opt/libpng/include/libpng16 -I/opt/homebrew/Cellar/pixman/0.42.2/include/pixman-1 -I/Users/obscuretone/Code/libraries/install/include -fvisibility=hidden -fdiagnostics-color=always -Wall -Winvalid-pch -Wextra -O0 -g -Wmissing-declarations -Werror-implicit-function-declaration -Wpointer-arith -Wwrite-strings -Wsign-compare -Wpacked -Wswitch-enum -Wmissing-format-attribute -Wvolatile-register-var -Wstrict-aliasing=2 -Winit-self -Wno-missing-field-initializers -Wno-unused-parameter -Wno-attributes -Wno-long-long -Winline -Wno-unused-but-set-variable -Wno-enum-conversion -fno-strict-aliasing -fno-common -D_GNU_SOURCE -arch arm64 -mmacosx-version-min=10.13 -Wno-attributes -Wno-unused-but-set-variable -Wno-missing-field-initializers -Wno-unused-parameter -Wno-long-long -D_REENTRANT -DCAIRO_COMPILATION -MD -MQ src/libcairo.2.dylib.p/cairo-script-surface.c.o -MF src/libcairo.2.dylib.p/cairo-script-surface.c.o.d -o src/libcairo.2.dylib.p/cairo-script-surface.c.o -c ../src/cairo-script-surface.c
In file included from ../src/cairo-script-surface.c:88:
../src/cairo-ft-private.h:87:30: error: unknown type name 'FT_Color'
                             FT_Color               *palette,
                             ^
1 error generated.
[125/148] Compiling C object src/libcairo.2.dylib.p/cairo-pdf-surface.c.o
ninja: build stopped: subcommand failed.
```

Obscuretone, 2024-06-14:
Seems to be this bug https://gitlab.freedesktop.org/cairo/cairo/-/issues/792

Obscuretone, 2024-06-15:
I was able to work through it and apply the patch with homebrew on Mac. There's already a bug open with cairo.

`brew tap Obscuretone/homebrew-cairo `
`brew install Obscuretone/homebrew-cairo/cairo --build-from-source`

Obscuretone, 2024-06-15:
3rd party issue

Obscuretone, 2024-06-15:
That doesn't resolve the issue as the build script grabs it's own copy. Making a PR for that.

Obscuretone, 2024-06-15:
Success. 

See https://github.com/openscad/openscad/pull/5178

closed 2024-06-15 by Obscuretone

kintel, 2024-06-15:
It seems to me that the real reason this fails is that FreeType is installed with Homebrew, while you build Cairo from source. The real bug is probably that `macosx-build-dependencies.sh` picks up the Homebrew build of FreeType instead of the one build from source.

Obscuretone, 2024-06-16:
the problem exists in both the homebrew build and the source build. I was able to patch both though.

kintel, 2024-06-16:
Out of curiosity, which Xcode version / compiler version are you using?
I've successfully installed the dependencies on both Homebrew and from source on up to macOS 14.5 and Xcode 15.4, so I wonder what causes your environment to fail.

I'm not saying your patch isn't the correct solution, it's just curious that I cannot reproduce it.

Obscuretone, 2024-06-16:
It's not even my patch, I found it on the cairo mailing list. One of the maintainers confirms this is the correct behaviour, but they haven't updated it. https://gitlab.freedesktop.org/cairo/cairo/-/issues/792#note_2053751

```
pple clang version 15.0.0 (clang-1500.3.9.4)
Target: arm64-apple-darwin23.1.0
Thread model: posix
InstalledDir: /Library/Developer/CommandLineTools/usr/bin
```

kintel, 2024-06-16:
Thanks, I see what's going on now: I think my initial guess was correct: The Cairo build system most likely picks up a newer version of FreeType from your Homebrew installation, yielding a mix of homebrew dependencies and source dependencies. This this should be fine in most cases, it's not a recommended setup as different build systems may pick up libraries from different locations, e.g. Cairo may end up using homebrew FreeType while OpenSCAD may use its own FreeType.

The patch you suggested is requires to build Cairo-1.18 against FreeType 2.13. We're still on FreeType 2.12, but if I upgrade `scripts/macosx-build-dependencies.sh` to FreeType-2.13, I can reproduce the issue.