openscad/openscad #5187: Configurable GUI font/icon size

ouroborus, 2024-06-21:
On a 4K monitor, on Windows 10, I haven't been able to figure out an exe file properties configuration that makes the GUI text reasonably sized. (OpenSCAD seems to ignore Windows' DPI settings. Setting up various compatibility modes does make the items slightly larger but they're still too small and it sacrifices resolution, giving blurring/pixelation.)

I'd like a configuration option for this.

As far as I can tell, Qt does have [high DPI support](https://doc.qt.io/qt-5/highdpi.html). No idea if that's in the version OpenSCAD uses or what it takes to enable that. So an alternative is to arrange things so that OpenSCAD naturally follows the system DPI via Qt.

(I'm experiencing OpenSCAD 2021.01)

jordanbrown0, 2024-06-23:
Several of the font sizes are configurable (though some only in recent builds).  Which precise Windows configuration item have you tried, and exactly which text are you unhappy with?

OpenSCAD *does* respond to Windows font size selections, though sometimes not very well and it sometimes requires a restart to take effect.

jordanbrown0, 2024-06-23:
I use a 43" 4K monitor, at ~100dpi. and find the defaults OK, but preferences and monitors vary.

kintel, 2024-06-23:
@ouroborus Try the latest development snapshot. The 2021.01 is a few years old by now.

ouroborus, 2024-06-23:
@kintel I found and installed the nightly 2024.06.23 from the [downloads page](https://openscad.org/downloads.html). Icons are larger and the output and customizer fonts are configurable. This is a lot better, thank you.

closed 2024-06-23 by ouroborus