openscad/openscad #5188: No experimental features in windows build

ThreeD-Michael, 2024-06-21:
When building from Github for Linux I see the configuration tab for experimental features, but when cross-compiling for Windows this tab is missing. Is there a way to enable these experimental features for the Windows version too?

t-paul, 2024-06-21:
`cmake -DSNAPSHOT=ON -DEXPERIMENTAL=ON`

ThreeD-Michael, 2024-06-22:
Thanks, but unfortunately that didn't help.

First of all: I am not familiar with this build environment, so please bear with me if I write nonsense.

After looking into release-common.sh I think I need to start a snapshot build with "./scripts/release-common.sh mingw64 **snaphot**". But then the build process aborts with "Package 'tbb', required by 'world', not found."

The Ubuntu repository doesn't know "tbb", so what should I do?

t-paul, 2024-06-22:
I'd suggest start at the beginning then. Why build yourself? What goal are you trying to reach? Why do you need the cross-build? There are options to build natively on Windows too.

What are the specific commands and what exact platform (e.g. which Ubuntu version) you are building on.

ThreeD-Michael, 2024-06-22:
I'm building on a freshly installed and updated Ubuntu 22.04 LTS inside a Hyper-V Machine, following the instructions in this repository. When starting the build with "./scripts/release-common.sh mingw64" I get a working windows executable, but without the experimental features.

When adding "snapshot" to the build command the build fails with "Package 'tbb', required by 'world', not found.". It appears that "tbb" is required for "manifold" so it should have been installed before.

kintel, 2024-06-22:
MXE cross compile doesn't work out of the box as MXE doesn't yet contain all needed packages. To work around that, we have our own MXE fork where we (temporarily) maintain these packages, and we build Docker images to avoid rebuilding all the time. If you look at our snapshot build script you should be able to follow the same steps to reproduce that build: https://github.com/openscad/openscad/blob/master/.circleci/config.yml

..but as @t-paul indicates, this build already exists at https://openscad.org/downloads.html, so the value of creating your own may be limited.

ThreeD-Michael, 2024-06-22:
Ok, thanks.

pca006132, 2024-07-13:
After the recent updates, manifold dropped the thrust dependency. If tbb is not present, manifold can still be built without parallelization. Will that solve this issue?

closed 2026-10-03 by t-paul