openscad/openscad #5200: Unable to render a union with a SVG path when curve self-intersects.

lubosz, 2024-07-01:
**Describe the bug**
For paths that contain self-intersections caused by an inverted handle direction, rendering a union with extruded path fails with:
```
ERROR: The given mesh is not closed! Unable to convert to CGAL_Nef_Polyhedron.
```
SVGs in the wild can contain this issue without being apparent due to the path's complexity.

**To Reproduce**
Steps to reproduce the behavior:
1. Clone the repro repo https://github.com/lubosz/openscad-svg-bug/tree/main
2. Open `svg-bug.scad`
3. Render the mesh
4. See error

**Expected behavior**
The mesh should be able to render, as correctly visible in the preview.  It would be nice for OpenSCAD to handle this or to have a way to analyze which nodes caused the issue.

**Code reproducing the issue**
```
// Produces "ERROR: The given mesh is not closed! Unable to convert to CGAL_Nef_Polyhedron. "
svg_path = "broken/a.svg";

// Works
// svg_path = "working/a.svg";

union() {
  // Commenting the cube line avoids the issue
  cube([50, 50, 1], center=true);

  linear_extrude(2) import(svg_path, center=true);
}
```

**Screenshots**

| A bezier path node with a handle directions that cause self-intersection. |
|:--| 
| [image] | 

| Fixed bezier path node. |
|:--| 
| [image] | 

| Preview works without errors. |
|:--| 
| [image] | 

| Without creating a union the mesh can be rendered. |
|:--| 
| [image] | 

**Environment and Version info (please complete the following information):**
 - OS: Arch Linux
 - System: AMD PC 64-bit
 - OpenSCAD Version: Built from source using git master at `d4c36aa2bb7c20d9e3dd847ce96ebe6011bae39b` and
Version `2021.01-11` installed from Arch Linux

nophead, 2024-07-01:
I think you can fix it by applying a miniscule `offset()`. That passes it through the 2D Clipper library that is very good at fixing broken polygons.

lubosz, 2024-07-01:
> `offset`

@nophead You are right! Applying `offset(0.00001)` to the import avoided the issue!
An offset of `0.000001` was not enough, `0.00001` seems to be needed, which did not introduce any noticeable visible change.

nophead, 2024-07-01:
You can always offet again with -0.0001 to not change the dimensions.

kintel, 2024-07-01:
Closing as duplicate of #5053.
Note: This works when using the Manifold backend

closed 2024-07-01 by kintel

lubosz, 2024-07-01:
I can confirm that the issue does not occur using the manifold backend.

To enable the backend, OpenSCAD needs to be compiled with `-DEXPERIMENTAL=On`, which makes the "Features" tab appear under preferences. Manifold can be enabled there.

kintel, 2024-07-01:
Or you can use a precompiled development snapshot instead of building it yourself