openscad/openscad #5205: Rendering with substraction gets wrong results.

Dapid, 2024-07-06:
**Describe the bug**
I have a design with an inclined tube, cut at the bottom by the z=0 plane. When rendering (either in code or with the button), the last subtraction is not applied.

**To Reproduce**
See the code: https://gist.github.com/Dapid/4bd8f271f7ea43
[image]
cea5c1ee46be36e219

Here is what I see without render (correct results):
[image]
[image]

And here is what I get with rendering:
[image]
[image]

**Environment and Version info:**
```
OpenSCAD Version: 2024.06.21 [built from source]
System information: Linux 6.8.11-300.fc40.x86_64 #1 SMP PREEMPT_DYNAMIC Mon May 27 14:53:33 UTC 2024 x86_64 Fedora Linux 40 (Forty) 8 CPUs 15.38 GB RAM
User Agent: OpenSCAD/2024.06.21 (Linux x86_64; Fedora Linux 40 (Forty))
Compiler: GCC "14.1.1 20240607 (Red Hat 14.1.1-5)" 64bit
MinGW build: No
Debug build: No
Boost version: 1_83
Eigen version: 3.4.0
CGAL version, kernels: 5.6.1, Cartesian<Gmpq>, Extended_cartesian<Gmpq>, Epeck
OpenCSG version: OpenCSG 1.5.0
Qt version: 6.7.2
QScintilla version: 2.13.4
InputDrivers: 
GLib version: 2.80.2
lodepng version: 20230410
libzip version: 1.10.1
fontconfig version: 2.15.0
freetype version: 2.13.2
harfbuzz version: 8.4.0
cairo version: 1.18.0
lib3mf version: 2.2.0
GLEW version: 2.2.0
OpenGL Version: 4.6 (Compatibility Profile) Mesa 24.1.2
GL Renderer: Mesa Intel(R) UHD Graphics 620 (KBL GT2)
GL Vendor: Intel
RGBA(8880), depth(0), stencil(0)
GL_ARB_framebuffer_object: yes
GL_EXT_framebuffer_object: yes
GL_EXT_packed_depth_stencil: yes
```

**Additional context**
The rendering log shows CSG is being used:
```
Compiling design (CSG Tree generation)...
Compiling design (CSG Products generation)...
Geometries in cache: 128
Geometry cache size in bytes: 7935336
CGAL Polyhedrons in cache: 67
CGAL cache size in bytes: 97510424
Compiling design (CSG Products normalization)...
Normalized tree has 2 elements!
Compile and preview finished.
Total rendering time: 0:00:00.109
```

jordanbrown0, 2024-07-06:
The gist link was damaged, had the image inserted in the middle of it.  I took the liberty of fixing it.

jordanbrown0, 2024-07-06:
One hint:  the cylinder that you are subtracting in main_body2() starts at the same point as the cylinder you are subtracting it from.  Don't do that.  Always make sure that the negative part is larger than the positive part.  Otherwise, OpenSCAD often can't decide whether the last little bit of material should be subtracted away, or if there should be an infinitesimal amount left.

jordanbrown0, 2024-07-06:
Adding "center=true" is a good way to address that issue in this particular case.

jordanbrown0, 2024-07-06:
As for the not-subtracting part, I don't duplicate that with 2024.07.03, either with or without Manifold enabled.

thehans, 2024-07-06:
Duplicate of #4748

You can use any of the following workarounds: 
- Don't put camera inside negative objects (Don't grossly oversize negative objects)
- switch to orthographic view mode
- use `render()` to force geometry calculation.

thehans, 2024-07-06:
Sorry, I may have misundertsood this report, but I am unable to reproduce the issue as reported.  The only wrong appearance I could reproduce was F5 preview with `render()` commented out, which relates to the issue linked above.

Your version says "[built from source]".  Was that not downloaded from https://openscad.org/downloads#snapshots ?  I would recommend trying one of those builds first to verify the issue.  Also are you enabling any experimental features?

Dapid, 2024-07-06:
@jordanbrown0: Thanks for fixing the link. And I am aware of the substraction, I just tried to simplify the model as much as possible.

After recompiling the latest version, it went away. Maybe I got unlucky with the version.

@thehans in the faulty version, the exported mesh was also incorrect, since it followed whatever was displayed upon rendering.
I installed from source because that is what it refers on the README.

```-DCMAKE_INSTALL_PREFIX=/home/david/.local/  -DENABLE_PYTHON=ON -DMANIFOLD_EXPORT=ON -DUSE_QT6=True  -DCRYPTOPP_LIBRARIES=/usr/lib64/libcryptopp.so.8 -DEXPERIMENTAL=1 
```

Should in the future get the source from snapshots, or just for bug hunting?

closed 2024-07-06 by Dapid

thehans, 2024-07-06:
The snapshots are automatically generated from the head of the repo main branch.  There is nothing special about the revisions used for them.

The main reason I recommend using the pre-built snapshots is that they come from a known setup (the same CI servers which build and test every PR etc.)  
On the other hand if you are building from source on your own machine, that introduces a large number of unknowns.