openscad/openscad #5206: Update To:  Saving file on network drive writes NUL to file in Windows #3078

mwhead, 2024-07-06:
I added to a closed issue, ref: #3078 and should have re-started it so here it is:

"nul" files whilst saving to a NAS drive is not solved. It has just happened to me again. It is very, very occassional and so is extremely difficult if not impossible to replicate. It does not happen with any other program that I use when storing to the NAS drive. It happened to me a while ago and I always create separate copies now but this needs resolving. (Easy for me to say, I don't know the answer!)
Using Windows 10 and Openscad version 2021.01. Also happened on Windows 7 with earlier version of Openscad.
Thanks and regards

rbarbrow, 2024-07-06:
win 11 23h2   saved both scad and stl files to network drive could not replicate

mwhead, 2024-07-06:
Exactly. Therein lies the problem as it happens very, very rarely and so, as I said originally, it is almost impossible to replicate. I have seen the problem with different computers and with different NAS drives but over a very long time. I have raised it again in order to keep it in people's minds just in case by some chance someone notices what is happening.

rbarbrow, 2024-07-06:
Has anyone other than you had this issue?  (genuinely asking I don't know the full history)

mwhead, 2024-07-07:
Have a look at the original issue reference #3078. When the problem occurred with me I went looking and this was the only reference I could find. Are we just 2 unlucky folks? Maybe others have just shrugged and used a backup thinking it was "just-one-of-those-things". As I have said I don't know the answer but Openscad is the only program that has given me this problem on a range of different computers, OSs and NAS drives. Very difficult to solve I guess.

nophead, 2024-07-07:
Perhaps nobody else works on a NAS. I use my NAS for backup and I use GoogleDrive and NextCloud to synchronise my PCs so never save directly to a NAS.

mwhead, 2024-07-07:
I save many different program outputs to NAS drives and also backup to additional NAS drives. I need to be able to access the output from different computers in different locations. Surely I'm not the only one? Well, maybe I am!

nophead, 2024-07-07:
I tried saving a 9KB file over and over again on my Seagate NAS using CTRL-S and it always writes a 9KB file. I did notice it seems to write it  with a different name first and then renames it. It was so quick I had to video it to see what it was.

[image]

mwhead, 2024-07-07:
It normally works perfectly well, it is just, say, 1 in 1000 that are incorrectly saved hence the difficulty in fault finding. Also, the file could be zero bytes but also it can be the right size but ful of nuls instead of the correct data.

nophead, 2024-07-07:
I doubt it is an OpenSCAD problem, how would it know it was a NAS and write NULLs? Seems like it is a NAS problem that is only triggered by something OpenSCAD does.  Or maybe OpenSCAD makes some assumption about the file system that doesn't hold true for your NAS.

mwhead, 2024-07-07:
I take your point but it has happened over the years with different NAS drives and computers + operating systems. Openscad may well "make" an assumption about the FS but then that would be an Openscad problem. Regrettably we shall not get anywhere until it becomes a major problem so I think I have to leave it there.

t-paul, 2024-07-07:
> I tried saving a 9KB file over and over again on my Seagate NAS using CTRL-S and it always writes a 9KB file. I did notice it seems to write it with a different name first and then renames it. It was so quick I had to video it to see what it was.

Yes, the editor uses [QSaveFile](https://doc.qt.io/qt-6/qsavefile.html#commit) which does that internally.

nophead, 2024-07-07:
I suspect the problem is a race condition on the NAS where a rename immediately after a write causes a bug.

jordanbrown0, 2024-07-07:
I've seen a symptom like nophead describes, though I think it also involved a server crash at an inopportune moment.  (The operations got reordered so that the rename hit the disk before the write did.)

We should check that the file is properly closed, and probably even a (platform-specific) forced sync complete, before doing the rename.

nophead, 2024-07-07:
Yes but one would expect filesystem calls to be synchronous, so the write and file close should not return until the file has been written. However l have found a similar problem on a local file system using python on windows. If I delete a tree of files and then recreate the top level directory again it can fail on a fast machine. The deltree returns before the directory has been deleted so l had to insert a 100ms delay. Bonkers.

jordanbrown0, 2024-07-08:
It is generally *not* a good idea to assume that file system calls are synchronous - certainly not that a sequence of writes will be synchronous or even in order, and often not even that a close will be synchronous.  If you need synchronous operation, you need to use the explicit options and calls provided.

(References below are to C/POSIX documentation, but I expect that they apply equally to C++.)

Ref `O_SYNC` in https://www.gnu.org/software/libc/manual/html_node/Operating-Modes.html and https://pubs.opengroup.org/onlinepubs/9699919799/functions/open.html and `fsync()` in https://www.gnu.org/software/libc/manual/html_node/Synchronizing-I_002fO.html and https://pubs.opengroup.org/onlinepubs/9699919799/functions/fsync.html .

The standard for close() https://pubs.opengroup.org/onlinepubs/9699919799/functions/close.html does not discuss synchronization for normal files, neither promising synchronous operation nor warning that it might not be synchronous.  However, note https://linux.die.net/man/2/close :

> A successful close does not guarantee that the data has been successfully saved to disk, as the kernel defers writes. It is not common for a file system to flush the buffers when the stream is closed. If you need to be sure that the data is physically stored use [fsync](https://linux.die.net/man/2/fsync)(2). (It will depend on the disk hardware at this point.)

jordanbrown0, 2024-07-08:
I agree that file system calls should be (and I think are required to be) synchronous and in-order with respect to other operations on the system.  It is only in system-failure cases that the underlying asynchrony should come into play.

nophead, 2024-07-08:
Yes it might not be written to disk and be cached but a subsequent rename
should use the cached version. If it doesn't then it is a bug in the
filesystem, not the client.

DanielJoyce, 2024-07-31:
In general remote file systems break a lot of the assumptions that people make about local file systems. It's likely QSaveFile is not robust enough to handle the rare occassion remote network writes faile, likely swallowing an error somehwere or ignoring a return code.

degroof, 2024-12-28:
I think I'm running into a similar issue. I've got a Linux machine connected to a Windows shared folder via smbnetfs. Most apps (LibreOffice, PrusaSlicer, etc) can read and write just fine. OpenSCAD, though, is failing to write with a "Failed to open file for writing (Function not implemented)" error. 

Anyone happen to know of a workaround or a smbnetfs setting that might help?

[image]