openscad/openscad #5207: Crash nightly version 2024.07.03 by using roof

SteveLelievre, 2024-07-08:
**Describe the bug**
OpenSCAD crashes with certain roof construct

**To Reproduce**

Attempt to preview the sample code given below

**Expected behavior**
I would expect either the model to be displayed or an error message output

**Code reproducing the issue**

roof()difference(){square(10);translate([3,3])circle(3);}

**Screenshots**
Not applicable

**Environment and Version info (please complete the following information):**
 - Windows 10 Home 
 - System: Intel i5-7200U CPU @ 2.50GHz   2.71 GHz
 - OpenSCAD Version: 2024.07.03 from https://files.openscad.org/snapshots/OpenSCAD-2024.07.03-x86-64-Installer.exe
 - 
**Library & Graphics card information**
If applicable, add the content of the Help->Library Info window, or the output of running the `openscad --info` command.

OpenSCAD Version: 2024.07.03 (git 1d5586dda)
System information: Microsoft Windows 10 (10.0.19045) x86_64 4 CPUs 7.88 GB RAM
User Agent: OpenSCAD/2024.07.03 (git 1d5586dda) (Microsoft Windows 10 (10.0.19045) x86_64)
Compiler: GCC "13.2.0" 64bit
MinGW build: MingW64
Debug build: No
Boost version: 1_81
Eigen version: 3.4.0
CGAL version, kernels: 5.5, Cartesian<Gmpq>, Extended_cartesian<Gmpq>, Epeck
OpenCSG version: OpenCSG 1.5.0
Qt version: 5.15.11
QScintilla version: 2.11.2
InputDrivers: 
GLib version: 2.70.2
lodepng version: 20230410
libzip version: 1.5.2
fontconfig version: 2.14.2
freetype version: 2.13.2
harfbuzz version: 7.3.0
cairo version: 1.16.0
lib3mf version: 1.8.1
Features: fast-csg*, fast-csg-safer*, fast-csg-debug, manifold*, roof*, input-driver-dbus, lazy-union*, vertex-object-renderers-indexing*, textmetrics*, import-function*, predictible-output
Application Path: C:/Program Files/OpenSCAD
Documents Path: C:\Users\steve\Documents
User Documents Path: C:\Users\steve\Documents
Resource Path: C:/Program Files/OpenSCAD
User Library Path: C:/Users/steve/Documents/OpenSCAD/libraries
User Config Path: C:\Users\steve\AppData\Local/OpenSCAD
Backup Path: C:/Users/steve/Documents/OpenSCAD/backups
OPENSCADPATH: <not set>
OpenSCAD library path:
C:/Users/steve/Documents/OpenSCAD/libraries
C:/Program Files/OpenSCAD\libraries

OPENSCAD_FONT_PATH: <not set>
OpenSCAD font path:
C:/WINDOWS/fonts
C:/Users/steve/.local/share/fonts
C:/usr/local/share/fonts
C:/usr/share/fonts
C:/Users/steve/.fonts
C:/usr/X11/lib/X11/fonts
C:/System/Library/Fonts
C:/Library/Fonts
C:/Users/steve/Library/Fonts
GLEW version: 2.1.0
OpenGL Version: 4.6.0 - Build 27.20.100.9664
GL Renderer: Intel(R) HD Graphics 620
GL Vendor: Intel
RGBA(8888), depth(24), stencil(8)
GL_ARB_framebuffer_object: yes
GL_EXT_framebuffer_object: yes
GL_EXT_packed_depth_stencil: yes

Qt graphics widget: QOpenGLWidget
QSurfaceFormat: RGBA(8888), depth(24), stencil(8)

GL Extensions:
GL_3DFX_texture_compression_FXT1
GL_AMD_depth_clamp_separate
GL_AMD_vertex_shader_layer
GL_AMD_vertex_shader_viewport_index
GL_ARB_ES2_compatibility
GL_ARB_ES3_1_compatibility
GL_ARB_ES3_compatibility
GL_ARB_arrays_of_arrays
GL_ARB_base_instance
GL_ARB_bindless_texture
GL_ARB_blend_func_extended
GL_ARB_buffer_storage
GL_ARB_cl_event
GL_ARB_clear_buffer_object
GL_ARB_clear_texture
GL_ARB_clip_control
GL_ARB_color_buffer_float
GL_ARB_compatibility
GL_ARB_compressed_texture_pixel_storage
GL_ARB_compute_shader
GL_ARB_conditional_render_inverted
GL_ARB_conservative_depth
GL_ARB_copy_buffer
GL_ARB_copy_image
GL_ARB_cull_distance
GL_ARB_debug_output
GL_ARB_depth_buffer_float
GL_ARB_depth_clamp
GL_ARB_depth_texture
GL_ARB_derivative_control
GL_ARB_direct_state_access
GL_ARB_draw_buffers
GL_ARB_draw_buffers_blend
GL_ARB_draw_elements_base_vertex
GL_ARB_draw_indirect
GL_ARB_draw_instanced
GL_ARB_enhanced_layouts
GL_ARB_explicit_attrib_location
GL_ARB_explicit_uniform_location
GL_ARB_fragment_coord_conventions
GL_ARB_fragment_layer_viewport
GL_ARB_fragment_program
GL_ARB_fragment_program_shadow
GL_ARB_fragment_shader
GL_ARB_fragment_shader_interlock
GL_ARB_framebuffer_no_attachments
GL_ARB_framebuffer_object
GL_ARB_framebuffer_sRGB
GL_ARB_geometry_shader4
GL_ARB_get_program_binary
GL_ARB_get_texture_sub_image
GL_ARB_gl_spirv
GL_ARB_gpu_shader5
GL_ARB_gpu_shader_fp64
GL_ARB_half_float_pixel
GL_ARB_half_float_vertex
GL_ARB_indirect_parameters
GL_ARB_instanced_arrays
GL_ARB_internalformat_query
GL_ARB_internalformat_query2
GL_ARB_invalidate_subdata
GL_ARB_map_buffer_alignment
GL_ARB_map_buffer_range
GL_ARB_multi_bind
GL_ARB_multi_draw_indirect
GL_ARB_multisample
GL_ARB_multitexture
GL_ARB_occlusion_query
GL_ARB_occlusion_query2
GL_ARB_pipeline_statistics_query
GL_ARB_pixel_buffer_object
GL_ARB_point_parameters
GL_ARB_point_sprite
GL_ARB_polygon_offset_clamp
GL_ARB_post_depth_coverage
GL_ARB_program_interface_query
GL_ARB_provoking_vertex
GL_ARB_query_buffer_object
GL_ARB_robust_buffer_access_behavior
GL_ARB_robustness
GL_ARB_robustness_isolation
GL_ARB_sample_shading
GL_ARB_sampler_objects
GL_ARB_seamless_cube_map
GL_ARB_seamless_cubemap_per_texture
GL_ARB_separate_shader_objects
GL_ARB_shader_atomic_counter_ops
GL_ARB_shader_atomic_counters
GL_ARB_shader_bit_encoding
GL_ARB_shader_draw_parameters
GL_ARB_shader_group_vote
GL_ARB_shader_image_load_store
GL_ARB_shader_image_size
GL_ARB_shader_objects
GL_ARB_shader_precision
GL_ARB_shader_stencil_export
GL_ARB_shader_storage_buffer_object
GL_ARB_shader_subroutine
GL_ARB_shader_texture_image_samples
GL_ARB_shading_language_100
GL_ARB_shading_language_420pack
GL_ARB_shading_language_packing
GL_ARB_shadow
GL_ARB_spirv_extensions
GL_ARB_stencil_texturing
GL_ARB_sync
GL_ARB_tessellation_shader
GL_ARB_texture_barrier
GL_ARB_texture_border_clamp
GL_ARB_texture_buffer_object
GL_ARB_texture_buffer_object_rgb32
GL_ARB_texture_buffer_range
GL_ARB_texture_compression
GL_ARB_texture_compression_bptc
GL_ARB_texture_compression_rgtc
GL_ARB_texture_cube_map
GL_ARB_texture_cube_map_array
GL_ARB_texture_env_add
GL_ARB_texture_env_combine
GL_ARB_texture_env_crossbar
GL_ARB_texture_env_dot3
GL_ARB_texture_filter_anisotropic
GL_ARB_texture_float
GL_ARB_texture_gather
GL_ARB_texture_mirror_clamp_to_edge
GL_ARB_texture_mirrored_repeat
GL_ARB_texture_multisample
GL_ARB_texture_non_power_of_two
GL_ARB_texture_query_levels
GL_ARB_texture_query_lod
GL_ARB_texture_rectangle
GL_ARB_texture_rg
GL_ARB_texture_rgb10_a2ui
GL_ARB_texture_stencil8
GL_ARB_texture_storage
GL_ARB_texture_storage_multisample
GL_ARB_texture_swizzle
GL_ARB_texture_view
GL_ARB_timer_query
GL_ARB_transform_feedback2
GL_ARB_transform_feedback3
GL_ARB_transform_feedback_instanced
GL_ARB_transform_feedback_overflow_query
GL_ARB_transpose_matrix
GL_ARB_uniform_buffer_object
GL_ARB_vertex_array_bgra
GL_ARB_vertex_array_object
GL_ARB_vertex_attrib_64bit
GL_ARB_vertex_attrib_binding
GL_ARB_vertex_buffer_object
GL_ARB_vertex_program
GL_ARB_vertex_shader
GL_ARB_vertex_type_10f_11f_11f_rev
GL_ARB_vertex_type_2_10_10_10_rev
GL_ARB_viewport_array
GL_ARB_window_pos
GL_ATI_separate_stencil
GL_EXT_abgr
GL_EXT_bgra
GL_EXT_blend_color
GL_EXT_blend_equation_separate
GL_EXT_blend_func_separate
GL_EXT_blend_minmax
GL_EXT_blend_subtract
GL_EXT_clip_volume_hint
GL_EXT_compiled_vertex_array
GL_EXT_direct_state_access
GL_EXT_draw_buffers2
GL_EXT_draw_range_elements
GL_EXT_fog_coord
GL_EXT_framebuffer_blit
GL_EXT_framebuffer_multisample
GL_EXT_framebuffer_object
GL_EXT_geometry_shader4
GL_EXT_gpu_program_parameters
GL_EXT_gpu_shader4
GL_EXT_multi_draw_arrays
GL_EXT_packed_depth_stencil
GL_EXT_packed_float
GL_EXT_packed_pixels
GL_EXT_polygon_offset_clamp
GL_EXT_rescale_normal
GL_EXT_secondary_color
GL_EXT_separate_specular_color
GL_EXT_shader_framebuffer_fetch
GL_EXT_shader_integer_mix
GL_EXT_shadow_funcs
GL_EXT_stencil_two_side
GL_EXT_stencil_wrap
GL_EXT_texture3D
GL_EXT_texture_array
GL_EXT_texture_compression_s3tc
GL_EXT_texture_edge_clamp
GL_EXT_texture_env_add
GL_EXT_texture_env_combine
GL_EXT_texture_filter_anisotropic
GL_EXT_texture_integer
GL_EXT_texture_lod_bias
GL_EXT_texture_rectangle
GL_EXT_texture_sRGB
GL_EXT_texture_sRGB_decode
GL_EXT_texture_shared_exponent
GL_EXT_texture_snorm
GL_EXT_texture_storage
GL_EXT_texture_swizzle
GL_EXT_timer_query
GL_EXT_transform_feedback
GL_IBM_texture_mirrored_repeat
GL_INTEL_conservative_rasterization
GL_INTEL_fragment_shader_ordering
GL_INTEL_framebuffer_CMAA
GL_INTEL_map_texture
GL_INTEL_multi_rate_fragment_shader
GL_INTEL_performance_query
GL_KHR_blend_equation_advanced
GL_KHR_blend_equation_advanced_coherent
GL_KHR_context_flush_control
GL_KHR_debug
GL_KHR_no_error
GL_KHR_shader_subgroup
GL_KHR_shader_subgroup_arithmetic
GL_KHR_shader_subgroup_ballot
GL_KHR_shader_subgroup_basic
GL_KHR_shader_subgroup_clustered
GL_KHR_shader_subgroup_quad
GL_KHR_shader_subgroup_shuffle
GL_KHR_shader_subgroup_shuffle_relative
GL_KHR_shader_subgroup_vote
GL_KHR_texture_compression_astc_hdr
GL_KHR_texture_compression_astc_ldr
GL_NV_blend_square
GL_NV_conditional_render
GL_NV_primitive_restart
GL_NV_texgen_reflection
GL_SGIS_generate_mipmap
GL_SGIS_texture_edge_clamp
GL_SGIS_texture_lod
GL_SUN_multi_draw_arrays
GL_WIN_swap_hint
WGL_EXT_swap_control

kintel, 2024-07-20:
Thanks for the report - I can reproduce this locally.
Note: roof() is a relatively unstable feature and needs a bit of love. We also don't have any unit tests for it, so there are likely a few hidden bugs there, including this one.

For known issues, see https://github.com/openscad/openscad/issues?q=is%3Aissue+is%3Aopen+label%3A%22Feature%3A+Roof%22

coryrc, 2025-11-13:
Some backtrace:

```
#0  roof_vd::vd_inner_faces
    (vd=..., segments=std::vector of length 14, capacity 16 = {...}, fa=fa@entry=12, fs=fs@entry=268435456) at /opt/prg/openscad/src/geometry/roof_vd.cc:266
#1  0x0000555555caf0b0 in roof_vd::voronoi_diagram_roof (poly=<optimized out>, fa=12, fs=2)
    at /opt/prg/openscad/src/geometry/roof_vd.cc:360
#2  0x0000555555b4d5cc in roofOverPolygon (node=..., poly=<optimized out>)
    at /opt/prg/openscad/src/geometry/GeometryEvaluator.cc:1001
#3  GeometryEvaluator::visit (this=0x7fffffffbe90, state=..., node=...)
    at /opt/prg/openscad/src/geometry/GeometryEvaluator.cc:1023
#4  0x0000555555a5e88a in NodeVisitor::traverse
    (this=this@entry=0x7fffffffbe90, node=..., state=<optimized out>)
    at /opt/prg/openscad/src/core/NodeVisitor.cc:30
#5  0x0000555555b48d72 in GeometryEvaluator::evaluateGeometry
    (this=this@entry=0x7fffffffbe90, node=..., allownef=allownef@entry=false)
    at /opt/prg/openscad/src/geometry/GeometryEvaluator.cc:80
```

```
info locals
twin_cell = 0x54d0cbda030
twin_segment = <optimized out>
v0 = {<Eigen::PlainObjectBase<Eigen::Matrix<double, 2, 1, 0, 2, 1> >> = {<Eigen::MatrixBase<Eigen::Matrix<double, 2, 1, 0, 2, 1> >> = {<Eigen::DenseBase<Eigen::Matrix<double, 2, 1, 0, 2, 1> >> = {<Eigen::DenseCoeffsBase<Eigen::Matrix<double, 2, 1, 0, 2, 1>, 3>> = {<Eigen::DenseCoeffsBase<Eigen::Matrix<double, 2, 1, 0, 2, 1>, 1>> = {<Eigen::DenseCoeffsBase<Eigen::Matrix<double, 2, 1, 0, 2, 1>, 0>> = {<Eigen::EigenBase<Eigen::Matrix<double, 2, 1, 0, 2, 1> >> = {<No data fields>}, <No data fields>}, <No data fields>}, <No data fields>}, <No data fields>}, <No data fields>}, m_storage = {m_data = {array = {-0, 
          402653184}}}}, <No data fields>}
discr = std::vector of length 63531586549274212, capacity -108086026775033080 = {
  <error reading variable discr (Cannot access memory at address 0x1800000000000000)>
twin_point = {a = 0, b = 402653184}
v1 = {<Eigen::PlainObjectBase<Eigen::Matrix<double, 2, 1, 0, 2, 1> >> = {<Eigen::MatrixBase<Eigen::Matrix<double, 2, 1, 0, 2, 1> >> = {<Eigen::DenseBase<Eigen::Matrix<double, 2, 1, 0, 2, 1> >> = {<Eigen::DenseCoeffsBase<Eigen::Matrix<double, 2, 1, 0, 2, 1>, 3>> = {<Eigen::DenseCoeffsBase<Eigen::Matrix<double, 2, 1, 0, 2, 1>, 1>> = {<Eigen::DenseCoeffsBase<Eigen::Matrix<double, 2, 1, 0, 2, 1>, 0>> = {<Eigen::EigenBase<Eigen::Matrix<double, 2, 1, 0, 2, 1> >> = {<No data fields>}, <No data fields>}, <No data fields>}, <No data fields>}, <No data fields>}, <No data fields>}, m_storage = {m_data = {array = {
          39414498.938777998, 153799831.03080887}}}}, <No data fields>}
edge = 0x54d0f9a0060
cell_index = <optimized out>
segment = @0x54d0e0d3420: {p0 = {a = 0, b = 1342177280}, p1 = {a = 0, b = 0}}
cell = @0x54d0cbda018: {source_index_ = 2, incident_edge_ = 0x54d0f9a0660, color_ = 9}
__for_range = std::vector of length 28, capacity 42 = {{source_index_ = 2, 
    incident_edge_ = 0x54d0f9a0120, color_ = 2}, {source_index_ = 2, incident_edge_ = 0x54d0f9a0660, 
    color_ = 9}, {source_index_ = 6, incident_edge_ = 0x54d0f9a01e0, color_ = 1}, {source_index_ = 1, 
    incident_edge_ = 0x54d0f9a07b0, color_ = 2}, {source_index_ = 3, incident_edge_ = 0x54d0f9a15f0, 
    color_ = 8}, {source_index_ = 6, incident_edge_ = 0x54d0f9a05a0, color_ = 8}, {source_index_ = 5, 
    incident_edge_ = 0x54d0f9a0510, color_ = 9}, {source_index_ = 1, incident_edge_ = 0x54d0f9a18f0, 
    color_ = 9}, {source_index_ = 4, incident_edge_ = 0x54d0f9a0630, color_ = 2}, {source_index_ = 7, 
    incident_edge_ = 0x54d0f9a06c0, color_ = 1}, {source_index_ = 4, incident_edge_ = 0x54d0f9a1140, 
    color_ = 9}, {source_index_ = 7, incident_edge_ = 0x54d0f9a1170, color_ = 8}, {source_index_ = 4, 
    incident_edge_ = 0x54d0f9a0a50, color_ = 1}, {source_index_ = 7, incident_edge_ = 0x54d0f9a0b40, 
    color_ = 2}, {source_index_ = 13, incident_edge_ = 0x54d0f9a11a0, color_ = 9}, {
    source_index_ = 8, incident_edge_ = 0x54d0f9a11d0, color_ = 8}, {source_index_ = 13, 
    incident_edge_ = 0x54d0f9a0e10, color_ = 1}, {source_index_ = 8, incident_edge_ = 0x54d0f9a1380, 
    color_ = 2}, {source_index_ = 12, incident_edge_ = 0x54d0f9a1200, color_ = 9}, {
    source_index_ = 9, incident_edge_ = 0x54d0f9a17d0, color_ = 8}, {source_index_ = 12, 
    incident_edge_ = 0x54d0f9a1410, color_ = 1}, {source_index_ = 10, incident_edge_ = 0x54d0f9a1770, 
    color_ = 1}, {source_index_ = 11, incident_edge_ = 0x54d0f9a1650, color_ = 9}, {
    source_index_ = 10, incident_edge_ = 0x54d0f9a1710, color_ = 8}, {source_index_ = 11, 
    incident_edge_ = 0x54d0f9a16b0, color_ = 1}, {source_index_ = 0, incident_edge_ = 0x54d0f9a1560, 
    color_ = 1}, {source_index_ = 0, incident_edge_ = 0x54d0f9a1860, color_ = 8}, {source_index_ = 0, 
    incident_edge_ = 0x54d0f9a18c0, color_ = 2}}
__for_begin = <optimized out>
__for_end = <optimized out>
ret = {faces = std::vector of length 1, capacity 1 = {std::vector of length 6, capacity 8 = {
      {<Eigen::PlainObjectBase<Eigen::Matrix<double, 2, 1, 0, 2, 1> >> = {<Eigen::MatrixBase<Eigen::Matrix<double, 2, 1, 0, 2, 1> >> = {<Eigen::DenseBase<Eigen::Matrix<double, 2, 1, 0, 2, 1> >> = {<Eigen::DenseCoeffsBase<Eigen::Matrix<double, 2, 1, 0, 2, 1>, 3>> = {<Eigen::DenseCoeffsBase<Eigen::Matrix<double, 2, 1, 0, 2, 1>, 1>> = {<Eigen::DenseCoeffsBase<Eigen::Matrix<double, 2, 1, 0, 2, 1>, 0>> = {<Eigen::EigenBase<Eigen::Matrix<double, 2, 1, 0, 2, 1> >> = {<No data fields>}, <No data fields>}, <No data fields>}, <No dat [... cut ...]  = {<Eigen::DenseCoeffsBase<Eigen::Matrix<double, 2, 1, 0, 2, 1>, 1>> = {<Eigen::DenseCoeffsBase<Eigen::Matrix<double, 2, 1, 0, 2, 1>, 0>> = {<Eigen::EigenBase<Eigen::Matrix<double, 2, 1, 0, 2, 1> >> = {<No data fields>}, <No data fields>}, <No data fields>}, <No data fields>}, <No data fields>}, <No data fields>}, m_storage = {m_data = {array = {
              74883498.609569266, 74883498.609569266}}}}, <No data fields>}] = 74883498.609569266}}
cell_contains_boundary_point = <optimized out>
```

coryrc, 2025-11-13:
I think the code lines are one of these:

```
          Vector2d v0(edge->vertex0()->x(), edge->vertex0()->y()),
            v1(edge->vertex1()->x(), edge->vertex1()->y());
```

which makes it likely #6291 points out the cause?

coryrc, 2025-11-13:
So if you want to know what's going on:

Read the first paragraph: https://www.boost.org/doc/libs/latest/libs/polygon/doc/voronoi_diagram.htm

Edges are the black lines, some of which stretch off into infinity.

then look at this code:

```
        if (edge->is_linear()) {  // linear edge is simple
          Vector2d p(edge->vertex1()->x(), edge->vertex1()->y());
          ret.faces.back().push_back(p);
          ret.heights[p] = distance_to_segment(p, segment);
        } else {  // discretize a parabolic edge
          const voronoi_diagram::cell_type *twin_cell = edge->twin()->cell();
          if (!(twin_cell->contains_point())) {
            RAISE_ROOF_EXCEPTION("Voronoi error");
          }
          Segment twin_segment = segments[twin_cell->source_index()];
          Point twin_point =
            (twin_cell->source_category() == ::boost::polygon::SOURCE_CATEGORY_SEGMENT_START_POINT)
              ? twin_segment.p0
              : twin_segment.p1;
          Vector2d v0(edge->vertex0()->x(), edge->vertex0()->y()),
            v1(edge->vertex1()->x(), edge->vertex1()->y());
          
          std::vector<Vector2d> discr = discretize_arc(twin_point, segment, v1, v0, discretizer);
          std::reverse(discr.begin(), discr.end());
          for (std::size_t k = 1; k < discr.size(); k++) {
            ret.faces.back().push_back(discr[k]);
            ret.heights[discr[k]] = distance_to_segment(discr[k], segment);
          }
        }
```

We're trying to use both vertices from an edge, but edges don't always have two vertices. It returns NULL in that case. I'm adding a commit in to raise an exception in that case, which is obviously better than crashing, but certainly there is something more useful that can be done. 

Aha, this is happening when there's a loop or something causes a shared point to be used for both sides of the outline:

[image]

Shifting the circle the tiniest bit left or right doesn't crash.