openscad/openscad #5211: SIGSEGV with Minkovski operation using `manifold` (Snapshot version)

pludi, 2024-07-11:
**Describe the bug**
Rendering the preview of the attached file results in a `SIGSEGV` on macOS (2024.07.08) and Linux (2024.07.09).

**To Reproduce**
Steps to reproduce the behavior:
1. Enter the below code
1. Let a preview render
1. Enable `manifold` feature & restart
1. Enter or reload the code

**Expected behavior**
Preview rendered.

**Code reproducing the issue**
```openscad
difference() {
  minkowski() {
    union() {
      translate([0, .6, 0])
        cube([29.75, .5, 99.5], center = true);
      translate([-12.5, -.6, 0])
        cube([4.75, .5, 99.5], center = true);
      translate([-14.5, 0, 0])
        cube([.875, 1.2, 99.5], center = true);
    }
    sphere(.25);
  }
  translate([-15.375, 0, 0])
    cube([1, 2.2, 100], center = true);
}
```

**Environment and Version info (please complete the following information):**
- macOS:
   - OS: macOS Sonoma 14.5
   - System: MacBook Pro M3
   - OpenSCAD Version 2024.07.08, downloaded from openscad.org
- Linux:
    - OS: EndeavourOS / Arch Linux
    - System: Intel Core i7 system
    - OpenSCAD Version 2024.07.09, compiled locally from AUR

**Additional context**
Keeping the 3 `union` cubes non-overlapping by changing the last cube to
```openscad
cube([.875, .2, 99.5], center = true);
```
does not lead to this issue.

pca006132, 2024-07-11:
maybe you can compile with debug information (RelWithDebInfo) and attach the stack trace? it will be very helpful.

pludi, 2024-07-12:
Full output of the debug build:

```
> ./openscad
Could not initialize localization.
Error reading examples.json: examples.json: cannot open file
ERROR: Cannot open shader source file: 'MouseSelector.vert'
ERROR: Cannot open shader source file: 'MouseSelector.frag'
/usr/src/debug/openscad-git/openscad/src/gui/MouseSelector.cc: OpenGL vertex shader Error:
0:1(1): error: syntax error, unexpected end of file

/usr/src/debug/openscad-git/openscad/src/gui/MouseSelector.cc: OpenGL fragment shader Error:
0:1(1): error: syntax error, unexpected end of file

/usr/src/debug/openscad-git/openscad/src/gui/MouseSelector.cc: OpenGL Program Linker Error:
error: linking with uncompiled/unspecialized shadererror: linking with uncompiled/unspecialized shader

/usr/src/debug/openscad-git/openscad/src/gui/MouseSelector.cc: OpenGL symbol retrieval went wrong, id is -1

/usr/include/c++/14.1.1/bits/shared_ptr_base.h:1350: std::__shared_ptr_access<_Tp, _Lp, <anonymous>, <anonymous> >::element_type& std::__shared_ptr_access<_Tp, _Lp, <anonymous>, <anonymous> >::operator*() const [with _Tp = ManifoldGeometry; __gnu_cxx::_Lock_policy _Lp = __gnu_cxx::_S_atomic; bool <anonymous> = false; bool <anonymous> = false; element_type = ManifoldGeometry]: Assertion '_M_get() != nullptr' failed.
```

pca006132, 2024-07-12:
Is `manifold` the only feature you enabled? For some reason I cannot reproduce your error with current master (1577b12). It renders fine for me.

kintel, 2024-07-13:
I also tried to reproduce using the official macOS-2024.07.08 snapshot without success (on M1 mac on macOS 13.6 and 14.5).

pludi, 2024-07-13:
Yes, `manifold` is the only enabled feature. I noticed though that it seems I need to enable the feature & restart OpenSCAD for it to produce this effect.

kintel, 2024-07-13:
Do you have a crash report and/or error output for the macOS crash as well?

pludi, 2024-07-14:
I can try again on macOS on July 22nd, when I'm back from vacation.

sesse, 2024-07-18:
I also cannot reproduce with this test file, but I can with a much more complicated test file (sorry :-) ). I've tried to simplify it as much as possible, so it doesn't really resemble anything useful anymore when rendered:

```
$fa=12;
$fs=10;

minkowski() {
  union() {
    shape(0.0004, 0.0010, 4, 6);
    shape(0.0010, 0.0021, 6, 8);
  }
  cube([1,1,1]);
}

module shape(y1, y2, x1, x2) {
  translate([0, 0, -y2]) {
    linear_extrude(y2-y1, scale=x1/x2) {
      scale(x2/119) {
        translate([62.5, 62.5,0])
          circle(r=48);
        translate([62.5, 0,0])
          circle(r=48);
      }
    }
  }
}
```

On a Linux debug ASan build (I had to disable mimalloc to get ASan to run), it gives:

```
kos:~/nmu/openscad> ./build/openscad --enable manifold -o test.stl test.scad
CGAL Cache insert: union(){multmatrix([[1,0,0,0],[0,1,0,0], (0 bytes)
WARNING: [manifold] Minkowski failed with error, falling back to Nef operation: CGAL ERROR: assertion violation!
File: /usr/include/CGAL/Convex_decomposition_3/SM_walls.h
Line: 440
Explanation: wrong handle

ERROR: The given mesh is not closed! Unable to convert to CGAL_Nef_Polyhedron.
AddressSanitizer:DEADLYSIGNAL
=================================================================
==11716==ERROR: AddressSanitizer: SEGV on unknown address 0x000000000008 (pc 0x5651b68852c6 bp 0x7ffc098e75e0 sp 0x7ffc098e75d0 T0)
==11716==The signal is caused by a READ memory access.
==11716==Hint: address points to the zero page.
    #0 0x5651b68852c6 in Geometry::Geometry(Geometry const&) /home/sesse/nmu/openscad/src/geometry/Geometry.h:28
    #1 0x5651b688532b in ManifoldGeometry::ManifoldGeometry(ManifoldGeometry const&) /home/sesse/nmu/openscad/src/geometry/manifold/ManifoldGeometry.h:28
    #2 0x5651b687cd41 in ManifoldGeometry::minkowski(ManifoldGeometry const&) const /home/sesse/nmu/openscad/src/geometry/manifold/ManifoldGeometry.cc:300
    #3 0x5651b6a02974 in ManifoldUtils::applyOperator3DManifold(std::__cxx11::list<std::pair<std::shared_ptr<AbstractNode const>, std::shared_ptr<Geometry const> >, std::allocator<std::pair<std::shared_ptr<AbstractNode const>, std::shared_ptr<Geometry const> > > > const&, OpenSCADOperator) /home/sesse/nmu/openscad/src/geometry/manifold/manifold-applyops.cc:63
    #4 0x5651b6a0cb26 in ManifoldUtils::applyMinkowskiManifold(std::__cxx11::list<std::pair<std::shared_ptr<AbstractNode const>, std::shared_ptr<Geometry const> >, std::allocator<std::pair<std::shared_ptr<AbstractNode const>, std::shared_ptr<Geometry const> > > > const&) /home/sesse/nmu/openscad/src/geometry/manifold/manifold-applyops-minkowski.cc:217
    #5 0x5651b84d8c63 in applyMinkowski(std::__cxx11::list<std::pair<std::shared_ptr<AbstractNode const>, std::shared_ptr<Geometry const> >, std::allocator<std::pair<std::shared_ptr<AbstractNode const>, std::shared_ptr<Geometry const> > > > const&) /home/sesse/nmu/openscad/src/geometry/boolean_utils.cc:110
    #6 0x5651b84727ea in GeometryEvaluator::applyToChildren3D(AbstractNode const&, OpenSCADOperator) /home/sesse/nmu/openscad/src/geometry/GeometryEvaluator.cc:146
    #7 0x5651b8471b17 in GeometryEvaluator::applyToChildren(AbstractNode const&, OpenSCADOperator) /home/sesse/nmu/openscad/src/geometry/GeometryEvaluator.cc:112
    #8 0x5651b8482f1f in GeometryEvaluator::visit(State&, CgalAdvNode const&) /home/sesse/nmu/openscad/src/geometry/GeometryEvaluator.cc:1082
    #9 0x5651b7f3e260 in Response BaseVisitable::acceptImpl<CgalAdvNode>(State&, CgalAdvNode const&, BaseVisitor&) /home/sesse/nmu/openscad/src/core/BaseVisitable.h:32
    #10 0x5651b7f3dc32 in CgalAdvNode::accept(State&, BaseVisitor&) const /home/sesse/nmu/openscad/src/core/CgalAdvNode.h:16
    #11 0x5651b82c277c in NodeVisitor::traverse(AbstractNode const&, State const&) /home/sesse/nmu/openscad/src/core/NodeVisitor.cc:30
    #12 0x5651b82c25bf in NodeVisitor::traverse(AbstractNode const&, State const&) /home/sesse/nmu/openscad/src/core/NodeVisitor.cc:20
    #13 0x5651b8470be5 in GeometryEvaluator::evaluateGeometry(AbstractNode const&, bool) /home/sesse/nmu/openscad/src/geometry/GeometryEvaluator.cc:67
    #14 0x5651b7d9d731 in do_export(CommandLine const&, RenderVariables const&, FileFormat, SourceFile*) /home/sesse/nmu/openscad/src/openscad.cc:580
    #15 0x5651b7d9a057 in cmdline(CommandLine const&) /home/sesse/nmu/openscad/src/openscad.cc:451
    #16 0x5651b7da9558 in main /home/sesse/nmu/openscad/src/openscad.cc:1229
    #17 0x7f0dbaa42c89 in __libc_start_call_main ../sysdeps/nptl/libc_start_call_main.h:58
    #18 0x7f0dbaa42d44 in __libc_start_main_impl ../csu/libc-start.c:360
    #19 0x5651b6832150 in _start (/home/sesse/nmu/openscad/build/openscad+0x6e6150) (BuildId: fb7cecece1127e07c5320f2a5797c4cc236a52df)

AddressSanitizer can not provide additional info.
SUMMARY: AddressSanitizer: SEGV /home/sesse/nmu/openscad/src/geometry/Geometry.h:28 in Geometry::Geometry(Geometry const&)
==11716==ABORTING
```

Depending on how much I simplify, I can get a variety of different warnings about the Minkowski operation, but I guess it's somehow the same root bug. The immediate cause of SIGSEGV is, of course, that minkowskiOp() returns nullptr and then ManifoldGeometry::minkowski() assumes it's safe to dereference the return value (and so it tries to copy-construct a null object).

This is current master (2e045800).

sesse, 2024-07-18:
Simpler repro:

```
minkowski() {
  union() {
    linear_extrude(0.001, scale=0.6666667) {
      circle(r=8);
    }
    translate([0, 0, -0.001]) {
      linear_extrude(0.001, scale=0.75) {
        circle(r=8);
      }
    }
  }
  cube([1,1,1]);
}
```

kintel, 2024-07-18:
Thanks! I can reproduce this

kintel, 2024-07-18:
Looks like Manifold produces a geometry which CGAL isn't able to represent as a Nef polyhedron, and it bails.
Quick-fix: Catch exception and print error.
Long-term fix: Implement convex decomposition outside of CGAL.
While at it: Create some more corner-case minkowski test for corner-case manifold objects

sesse, 2024-07-18:
FWIW, this is common enough that I just had to turn off Manifold entirely, so I'm not sure if the quick-fix is much of a fix :-) Though I guess it only affects minkowski(), so if you don't use that, I suppose you're fine.

kintel, 2024-07-18:
Yeah, the quick-fix is just a way of making the process not crash, so at least it doesn't take your work down with it. I'm not yet sure what a medium-term fix would look like.

kintel, 2024-07-18:
In terms of it being "common enough", feel free to dump other examples here in case there are nuances of failure modes we need to be aware of.

sesse, 2024-07-18:
> Yeah, the quick-fix is just a way of making the process not crash, so at least it doesn't take your work down with it. 

The backup feature has saved me a couple of times already, so thanks to whoever added that. :-)

I'll be on the lookout for other models where I see manifold crash, and post them here if they seem reasonably different (to the degree I can figure that out). I do sometimes get different crash messages just by messing with $fa and $fs, but I don't know if they're qualitatively different.

kintel, 2024-07-18:
I'm not even sure if the crash you reported is the same as originally reported, but we can deduplicate as it becomes more clear

kintel, 2024-07-20:
It should no longer crash now. Waiting for @pludi for confirmation of whether the originally reported crash changed.

I'll track the issue that minkowski doesn't always yield correct results with Manifold separately and link it here.

sesse, 2024-07-20:
Just so I understand it right: It will now not crash, but the returned object will be empty? So I should hold off on torture-testing it.

kintel, 2024-07-20:
Exactly. Torture-testing for crashes is ok, but don't expect anything sane to come out of it.

pludi, 2024-07-22:
Attached is the macOS crash log for 2024.07.08. I can confirm that with 2024.07.21 it does not crash anymore, instead I get the same error message as in this comment https://github.com/openscad/openscad/issues/5211#issuecomment-2236758207

[openscad-2024.07.08-crash.txt](https://github.com/user-attachments/files/16328931/openscad-2024.07.08-crash.txt)

closed 2024-09-20 by kintel

sesse, 2024-09-23:
2024.09.22 still happily segfaults on minkowski() operations with Manifold :-) Should I file a new bug? Reopen this one? (I have an example, but it's not a small example, unfortunately.)

kintel, 2024-09-23:
@sesse If it doesn't crash with the initial example in this issue, but only in a new example from you, please open a new ticket.
..and if you're able to slim it down to as few lines and dependencies as possible, that would help a lot.

sesse, 2024-09-24:
Well, the good news is that it's got zero external dependencies, the bad news is that it's 250 lines and takes an hour or so to render before it crashes :-) I'm trying to reduce some resolution parameters and such to try to get to a faster crash, but it's a bit stubborn.

pca006132, 2024-09-24:
ideally, run it with gdb and see where the crash it