openscad/openscad #5213: CMake installs nothing on macOS

thecaralice, 2024-07-13:
**Describe the bug**
`cmake --install build --prefix PREFIX` leaves `PREFIX` empty

**To Reproduce**
1. Build OpenSCAD following the instructions in README
2. Run `cmake --install build`

**Expected behavior**
`OpenSCAD.app` is installed in the specified prefix

**Environment and Version info (please complete the following information):**
 - OS: macOS Sonoma 14.4.1
 - System: MacBook Pro with Apple M1
 - OpenSCAD Version: 1577b123cc3680115a8ee7f2bc077e16a0056be9

**Additional context**
I don't know a lot about CMake, but it seems that `install(…)` is only run if not building for Apple or when using APPLE_UNIX - that is, every case except macOS Bundle.

kintel, 2024-07-13:
We don't support installing macOS app bundles directly using cmake. Making a self-contained app bundle takes a few more steps: You can see how e.g. the development snapshot binaries are built here: https://github.com/openscad/openscad/blob/master/.circleci/config.yml#L118

thecaralice, 2024-07-13:
> We don't support installing macOS app bundles directly using cmake.

Is there a reason for that?

kintel, 2024-07-13:
> > We don't support installing macOS app bundles directly using cmake.
> 
> Is there a reason for that?

Not really, apart from it being tricky to get right.

thecaralice, 2024-07-14:
Seems to work with the following patch:
```patch
diff --git a/CMakeLists.txt b/CMakeLists.txt
index ef91d6c1a..28fbb70ef 100644
--- a/CMakeLists.txt
+++ b/CMakeLists.txt
@@ -1350,6 +1350,10 @@ if(NOT APPLE OR APPLE_UNIX)
   install(DIRECTORY ${CMAKE_CURRENT_LIST_DIR}/locale DESTINATION "${OPENSCAD_INSTALL_RESOURCEDIR}" FILES_MATCHING PATTERN "*/LC_MESSAGES/openscad.mo")
   install(DIRECTORY ${CMAKE_CURRENT_LIST_DIR}/shaders DESTINATION "${OPENSCAD_INSTALL_RESOURCEDIR}")
   install(DIRECTORY ${CMAKE_CURRENT_LIST_DIR}/templates DESTINATION "${OPENSCAD_INSTALL_RESOURCEDIR}" FILES_MATCHING PATTERN "*.json")
+else()
+  install(TARGETS ${CMAKE_PROJECT_NAME}
+      BUNDLE DESTINATION ${CMAKE_INSTALL_PREFIX}
+  )
 endif()
 
 # Packaging: CPACK_* settings should be configured before `include(CPack)`
```
However, as I've already said, I know nothing about CMake, so I can't guarantee this is the right way to do this.

kintel, 2024-07-14:
I'm sure the patch does _something_. What is your use-case? Build for distribution, build for homebrew, or just build for your own usage? How do you manage library dependencies, are they built manually in a separate location, or do they get automatically fully bundled with the app in this case?

thecaralice, 2024-07-14:
> Build for distribution, build for homebrew, or just build for your own usage?

Ideally, I would like to add proper macOS support for OpenSCAD in nixpkgs. Nix manages pretty much everything in this case. This exact patch is just for my own usage until OpenSCAD supports building macOS bundles or the proper way to do that becomes clear.

kintel, 2024-07-14:
Oh, cool. That sounds like a worthwhile effort. Can you open a Pull Request with your patch? Even if it's suboptimal, it's likely a positive baby step towards your goal, and it's unlikely to do any damage I think.

PR #5214 (2024-07-14): fix(cmake): allow installing macOS bundle with CMake
Fixes #5213.
closed 2024-07-14 by kintel