openscad/openscad #5216: color() without arguments renders black with Manifold

kintel, 2024-07-14:
This minimal example renders a black cube in Manifold mode, but a cube with default color in preview mode.
```
color() cube();
```

See also https://github.com/openscad/openscad/blob/master/tests/data/scad/3D/features/color-tests.scad

UBaer21, 2024-08-24:
I assume replacing "undef" with black ([0,0,0]) is also the reason why  color(alpha=.5)color("blue"); works only in render but not in preview.  As alpha can be set without defining a color but can't overwrite existing alphas without defining also a color. Would be nice if that would be possible too.

kintel, 2024-10-01:
@UBaer21 I think specifying a partial color was never intended, so if it works anywhere that's a coincidence :)
..but it's an interesting feature request. It's different from this issue though.

commit 591e34fe (2024-10-01): Don't apply invalid colors in VBORenderer. Fixes #5216
commit 202844c1 (2024-10-01): Don't apply invalid colors in VBORenderer. Fixes #5216 (#5347)
closed 2024-10-01 by kintel