openscad/openscad #5218: Subtracted minkowski isn't using the subtraction color in Manifold mode

kintel, 2024-07-15:
The following design subtracts a minkowski object from a simple positive object. The cutout surface should be rendered using the subtractions color (defaults to green), but is instead rendered with the default positive color (yellow):
```
difference() {
  cube(10, center=true);
  translate([5,5,5]) minkowski(convexity=2) {
    cube(8, center=true);
    cube(1, center=true);
  }
}
```

pca006132, 2024-09-19:
Probably missing an AsOriginal call after minkowski.

kintel, 2024-09-19:
This was already fixed by #5234 - forgot to close it.

@pca006132 Did you observe similar wrong behavior related to your latest commit in #5282?

pca006132, 2024-09-19:
yes but it is now fixed as well

kintel, 2025-04-17:
Reopening this: This is still broken in throwntogether mode, but we didn't include a test for that. Will create a test shortly.

kintel, 2025-04-19:
How to reproduce:
* Re-enable disabled test here: https://github.com/openscad/openscad/blob/7f35682b4bf278e3d33e3bb8293b5907ec356a29/tests/CMakeLists.txt#L1628
* `ctest -R throwntogether-manifold_issue5218`