openscad/openscad #5220: roof example broken in CGAL mode

kintel, 2024-07-16:
The `roof` example is broken in CGAL mode since #5185

Expected:
[image]

Actual:
[image]

MichaelAtOz, 2026-10-12:
This is still a problem, the OP didn't detail it.
Examples/Basics/Roof, with CGAL, this bit:
```
// A nice application is beveling of fonts:
translate([6,0,0])
roof(method = "straight")
text("straight skeleton", size = 2, halign = "left", valign = "center");
cube(); // to force a CGAL union
```
Gets
> Rendering Polygon Mesh using CGAL...
ERROR: CGAL error in CGAL_Nef_polyhedron3(): CGAL ERROR: assertion violation!
Expr: ss_circle.has_on(sp_0)
File: /mxe/usr/x86_64-w64-mingw32.static.posix/include/CGAL/Nef_3/polygon_mesh_to_nef_3.h
Line: 291

Possibly a release blocker (?), IFF roof() is considered for release.

MichaelAtOz, 2026-10-12:
(2026.10.05 W11)