openscad/openscad #5221: name polyhedron points

yukulele, 2024-07-17:
**Is your feature request related to a problem? Please describe.**
it's not easy to identify the points with their positions in the table.

**Describe the solution you'd like**
the points could be named.
This makes it much easier to combine points into polygons, but also to correct clockwise rotation.

**Describe alternatives you've considered**
add comments on each points.

example:

```scad
module house() {
  polyhedron(
    [
      [ "frontBtmLeft", [ 0, -5, 0 ] ],
      [ "frontTopLeft", [ 0, -5, 10 ] ],
      [ "frontTopRight", [ 10, -5, 10 ] ],
      [ "frontBtmRight", [ 10, -5, 0 ] ],

      [ "backBtmLeft", [ 0, 5, 0 ] ],
      [ "backBtmRight", [ 10, 5, 0 ] ],
      [ "backTopRight", [ 10, 5, 10 ] ],
      [ "backTopLeft", [ 0, 5, 10 ] ],

      [ "roofLeft", [ 0, 0, 15 ] ],
      [ "roofRight", [ 0, 0, 15 ] ],
    ],
    [
      [
        "frontBtmLeft",
        "frontTopLeft",
        "frontTopRight",
        "frontBtmRight",
      ],
      [
        "backTopRight",
        "backTopLeft",
        "backBtmLeft",
        "backBtmRight",
      ],
      [
        "backBtmLeft",
        "backTopLeft",
        "roofLeft",
        "frontTopLeft",
        "frontBtmLeft",
      ],
      [
        "frontBtmRight",
        "frontTopRight",
        "roofRght",
        "backTopRight",
        "backBtmRight",
      ],
      [
        "frontTopLeft",
        "roofLeft",
        "roofRight",
        "frontTopRight",
      ],
      [
        "backTopRight",
        "roofRight",
        "roofLeft",
        "backTopLeft",
      ]
    ]
  );
```

This makes it much easier to combine points into polygons, but also to correct clockwise rotation.

nophead, 2024-07-17:
You could use a let() statement to name the points and then use them in the lists, I think.

yukulele, 2024-07-17:
> You could use a let() statement to name the points and then use them in the lists, I think.

I don't think so, because `faces` expects indexes from the `points` list.

UBaer21, 2024-07-17:
```
module house() {

frontBtmLeft= [ 0, -5, 0 ] ;
frontTopLeft= [ 0, -5, 10 ] ;
frontTopRight= [ 10, -5, 10 ] ;
frontBtmRight= [ 10, -5, 0 ] ;

backBtmLeft= [ 0, 5, 0 ] ;
backBtmRight= [ 10, 5, 0 ] ;
backTopRight= [ 10, 5, 10 ] ;
backTopLeft= [ 0, 5, 10 ] ;

roofLeft= [ 0, 0, 15 ] ;
roofRight= [ 10, 0, 15 ] ;

  polyhedron(
   points = [
      frontBtmLeft,    // 0
      frontTopLeft,   // 1
      frontTopRight, // 2
      frontBtmRight,// 3

      backBtmLeft,    // 4
      backBtmRight,  // 5
      backTopRight, // 6
      backTopLeft, // 7

      roofLeft,   // 8
      roofRight, // 9
    ],
    faces = [
      // floor
      [3,5,4,0],
      // wall
      [0,1,2,3], // front
      [4,5,6,7], // back
      [4,7,8,1,0], // left
      [6,5,3,2,9], // right
      // roof
      [8,9,2,1],
      [7,6,9,8]

    ]
  );
  }
  
  house();
```

Here a working example. While the naming might help you a bit in this example, most of the time using polyhedra you will use functions to calculate thousand of points.

You could also name the faces but the problem is the order of them you might use a "search" to find the name in your array and getting the position - however if you work a bit with points you will notice that naming points is not that helpful. 
What is, is knowing the number of the points which can be done with a small function.

  
```
frontBtmLeft= [ 0, -5, 0 ] ;
frontTopLeft= [ 0, -5, 10 ] ;
frontTopRight= [ 10, -5, 10 ] ;
frontBtmRight= [ 10, -5, 0 ] ;

backBtmLeft= [ 0, 5, 0 ] ;
backBtmRight= [ 10, 5, 0 ] ;
backTopRight= [ 10, 5, 10 ] ;
backTopLeft= [ 0, 5, 10 ] ;

roofLeft= [ 0, 0, 15 ] ;
roofRight= [ 10, 0, 15 ] ;
     points = [
      frontBtmLeft,    // 0
      frontTopLeft,   // 1
      frontTopRight, // 2
      frontBtmRight,// 3

      backBtmLeft,    // 4
      backBtmRight,  // 5
      backTopRight, // 6
      backTopLeft, // 7

      roofLeft,   // 8
      roofRight, // 9
    ];
    
    color("red",.5)for (pos=[0:len(points)-1])translate(points[pos])rotate($vpr)linear_extrude(5,convexity=5,center=true)text(str(pos),size=2);
```
now it is quite easy to build the faces with the right number, and it is less writing too.