openscad/openscad #5222: export()

yahbluez, 2024-07-18:
I know that any kind of save() is claimed as harmful in the openscad context.

I like to discuss a secure way to implement a save export() method to openscad.
The central idea is to implement an export() module that did not write to the filesystem but into an archive which maybe given in the preferences. That way the export() can not break out or write / overwrite someting.

And writing the export into an zip archive can't harm the filesystem.

Today workaround with bash and python add a lot of workload that could be done inside openscad if this kind of export() would exist.

gsohler, 2024-07-18:
Of course the archive could then contain not one, but many exported data items.
If I had to choose between bash script and openscad archive, I would still prefer the bash script(or windows powershell)  method, because during usage you always need to extract the data to proceed,(this is the time when the evil would escape),
for the bash script: you spend the effort only once but you never have to extract the data again in case the model changes:
PLUS: even if we had a working PR now, there might be a looonger ETA to merge as time is really limited, its a "hot" topic and right now are are many features waiting, which are way more attractive.

yahbluez, 2024-07-18:
A new export() will not remove the old way to use an external script language to work around the miss of an internal export().

I try to give some examples why an internal export() can be superior over the external use.

Think about a parametric model you like to generate in several versions at once.

Or one that needs heavy and long manifold calculations, with an internal export() this calculations may be cached while they need to be done several times while scripting from outside because that way each export is a new and need rendering of everything. That can eat hours even with the dev version of openscad.

I have a few projects where i used bash but moved over to python because of more easy handling of any kind of data.
I could move over to one of the python/openscad or cadquery things but i like for many reasons stay in openscad because the idea of openscad is so great.

jordanbrown0, 2025-03-29:
Related to #1041.