openscad/openscad #5225: Kill AMF support

kintel, 2024-07-19:
Suggestion: Kill AMF support:
* The AMF file format seems to be pretty dead.
* OpenSCAD has some known AMF import/export issues which are not seeing any traction
* Keeping patching AMF support with low usage adds to maintenance burden

Main question is when to pull the plug.

t-paul, 2024-07-19:
Perfect option:
* Mark deprecated in next release, kill in the one following (timeline?: :scream:)

In my view still ok as AMF seems very dead:
* Kill as soon as convenient and someone has time to do it, next release will not have it

For some limited potential feedback, we could ask via Mastodon.

kintel, 2024-07-19:
As it's not a language feature, we could kill it directly IMO, but let's let it stew for a bit in case someone comes up with some good arguments (and dev resources) for keeping it.
I'll consider marking it as deprecated and issuing a warning on usage immediately to signal any dev snapshot power users.

t-paul, 2026-08-29:
Any last words for AMF?

jordanbrown0, 2026-08-29:
RIP.

closed 2026-08-30 by t-paul