openscad/openscad #5236: Openscad-nightly 2024-07-19 does not draw picture anymore.

timonoko, 2024-07-22:
Openscad-nightly 2024-07-19 does not draw picture anymore.

Maybe because I have 2012-era graphics card.  [GeForce GTX 760] 

---
Confirmed. Works fine on newer machine.

closed 2024-07-22 by timonoko

timonoko, 2024-07-22:
In Ubuntu 18.03 and Snap. So old machine with old operating system.  Not an issue Imho.

t-paul, 2024-07-22:
Hmm, ok, that's a bit surprising. The "newer" version of OpenGL we need is still ancient. But it could be some incompatibility with the older drivers.

kintel, 2024-07-22:
The only thing I can think of is that our VBO renderer inadvertently requires something that nVidia doesn't provide, or triggers a bug in the driver. If you build OpenSCAD yourself, you can test this by building with `cmake -DUSE_LEGACY_RENDERERS=ON`. Beware: This is not very well tested.