openscad/openscad #5237: VBO has changed how "click on the Model Jump to the file" behave

damienmarchal, 2024-07-22:
**Describe the bug**
In https://github.com/openscad/openscad/pull/3276 was introduced a click on the model to jump to file feature. 
It seems that VBO and Legacy rendering behave differently regarding this feature. The consequence is that pop up menu that is filled using the rendering index show a different content between the two version. I'm not sure the change of UX behavior when VBO was made the default renderer was intended. 

**To Reproduce**
Steps to reproduce the behavior:
0. Compile master, 
1. Open openSCAD & open the "logo.scad" file 
2. Right click on the 3D view in one of the hole of the sphere. 
3. The Pop-up menu should display as in screenshot (1) 

4. Recompile same version with USE_LEGACY_RENDERERS set to TRUE
5. repeat step 1,2,3, the menu should show screenshot(2). 

Screenshot 1:
Recent master branch
[image]

Screenshot 2:
Recent master branch compiled with USE_LEGACY_RENDERERS set to TRUE
[image]

NB: 
I'm working on a PR with some part of https://thomaspietrzak.com/bibliography/gonzalez23.pdf and any help on having the legacy behavior using the VBO renderer would be appreciated.

kintel, 2024-07-22:
This was likely unintended. We don't have automated tests for UI features so it's tricky to get everything right. I did a bunch of cleanup of the VBO renderer earlier, but it's still a bit messy and needs more cleanup. I'm away for the summer but can take a look at this in the fall.

damienmarchal, 2024-08-13:
Hi @kintel  thanks for the reply.