openscad/openscad #5240: complete freeze on opening

kuldeepdhaka, 2024-07-25:
**Describe the bug**
OpenSCAD do not work at all when opening the file 'bla.scad'

**To Reproduce**
Steps to reproduce the behavior:
1. Download both bla.scad and polyround.scad (too obvious to say rename .txt? :)
2. Open bla.scad
3. Amazed?

**Expected behavior**
I should be able to work

**Code reproducing the issue**
Dummy code presented

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Environment and Version info (please complete the following information):**
- Linux 127.0.0.1localhost 6.9.1-arch1-2 #1 SMP PREEMPT_DYNAMIC Wed, 22 May 2024 13:47:07 +0000 x86_64 GNU/Linux
 - OpenSCAD version 2021.01

**Additional context**
Add any other context about the problem here.
[polyround.scad.txt](https://github.com/user-attachments/files/16383338/polyround.scad.txt)
[bla.scad.txt](https://github.com/user-attachments/files/16383339/bla.scad.txt)

t-paul, 2024-07-25:
Enabling Preferences -> Advanced -> "Stop on first warning" guards against such issues.

kuldeepdhaka, 2024-07-26:
@t-paul I understand your suggestion (thanks!) but as a user I do expect some form of "reliability" when I open a file.

thehans, 2024-07-26:
FWIW the reason for the slowness (as far as I can recall from last time I looked at these sort of performance issues) is that the "Error Log" handles excessive numbers of warnings/errors extremely poorly.  I think it rebuilds the whole grid for each addition, giving it a O(n^2) kind of performance.

The given code produces over 31K warnings due to undefined values.  You can get through all of them in a flash if you run on the command line
```
$ openscad -o bla.stl bla.scad 2> log.txt
$ wc -l log.txt
31398 log.txt
```

Also see: https://github.com/openscad/openscad/issues/4459 

Adding a hard cap on number of allowable errors before terminating is one suggested solutiont that would allow the UI to finish.

pca006132, 2024-07-29:
Maybe we can only update the error log once when the evaluation is finished? Or add a cap and switch to this lazy rendering when there are too many errors.

thehans, 2024-08-07:
Closing as duplicate of #4459

closed 2024-08-07 by thehans