openscad/openscad #5249: QCssParser::parseHexColor: Unknown color name '#30306' with dark theme

mmuman, 2024-08-04:
**Describe the bug**

`ViewportControl::yellowHintBackground()` returns a boggus hex color value wen using a dark theme.

The code in `src/gui/ViewportControl.cc` misses a zero, not sure where exactly.

**To Reproduce**

Use a dark theme.
Run from a terminal.
Zoom in until distance in the Aspect Ratio panel shows less than 1.0.

**Expected behavior**
The proper color should be used as background on the text field instead of the Qt warning showing up.

**Environment and Version info (please complete the following information):**
 - OS: Debian Linux Sid
 - System: AMD64
 - OpenSCAD Version 3816e45d4b8297dd68742ecd8ba9776df0af01c5

t-paul, 2024-08-04:
It's coming directly from that function https://github.com/openscad/openscad/blob/master/src/gui/ViewportControl.cc#L55

Looking at the values, I assume it is supposed to be `#303006` giving a very dark yellow-ish color. I can't test as on my system it seems to completely ignore any setting of dark-mode including setting `QT_QPA_PLATFORMTHEME`.

mmuman, 2024-08-04:
Ah yes, makes more sense than `#030306` or `#303060`  which didn't look convincing:

[image]

PR #5251 (2024-08-05): Fix dark theme color
Fixes #5249
commit 1cda31e0 (2024-08-05): Fix dark theme color

Fixes #5249
closed 2024-08-05 by t-paul