openscad/openscad #5252: Artifacts/unexpected render when rendering with fillets3d.scad library using manifold

humneraudio, 2024-08-05:
I'm getting artifacts/unexpected render when rendering with the fillets3d.scad library using manifold.

I'm using this fillet library - https://github.com/ademuri/openscad-fillets

We can use this simple cube to replicate the problem

$fn= 30;
use <fillets3d.scad>;
topBottomFillet(b = 0, t = 10, r = 4, s = 24, e = 1)
cube([50,30,10]);

And enable "manifold" under the features.

[image]

This screenshot shows the result. Note that it will preview fine. top or bottom only fillets also have the same issue. 

I'm able to produce an expected render with manifold if the radius is exactly half the Z height(thickness) of the part.

Now if we enable fast-csg in stead, I get the expected render every time.

[image]

 - OS: Win 10
 - System: Intel 64bit i7-7700
 - OpenSCAD Version 2024.07.27

pca006132, 2024-08-05:
It seems to be https://github.com/openscad/openscad/issues/4871 plus some weird incorrect edge collapse. One way to fix it would be to change https://github.com/ademuri/openscad-fillets/blob/main/fillets3d.scad#L85 to smaller values, e.g. 100.

The more concerning issue is some triangles appear to be flipped... when I set `s = 500`, offset to 100, I got this

[image]

I should file an upstream issue later.

pca006132, 2024-08-05:
Closing as duplicate of #4871

closed 2024-08-05 by pca006132