openscad/openscad #5253: Openscad crash

praetp, 2024-08-06:
Hi,

I was using the nightly snapshot of June 15th and everything worked.
When I upstepped to Jul 27th (or even Jun 23rd), openscad started to crash for some workloads in my docker container (however, not on my desktop PC). So it does smell like some dependency issue.

Some output from openscad --info (from within docker container)

> OpenSCAD Version: 2024.08.05.ai20224 (git 8119ad428)
> System information: Linux 5.15.0-117-generic #127-Ubuntu SMP Fri Jul 5 20:13:28 UTC 2024 x86_64 Amazon Linux 2023.5.20240701 4 CPUs 15.57 GB RAM
> User Agent: OpenSCAD/2024.08.05.ai20224 (git 8119ad428) (Linux x86_64; Amazon Linux 2023.5.20240701)
> Compiler: GCC "9.4.0" 64bit
> MinGW build: No
> Debug build: No
> Boost version: 1_71
> Eigen version: 3.3.7
> CGAL version, kernels: 5.5.1, Cartesian<Gmpq>, Extended_cartesian<Gmpq>, Epeck
> OpenCSG version: OpenCSG 1.5.0
> Qt version: 5.15.4
> QScintilla version: 2.11.2
> InputDrivers: 
> GLib version: 2.64.6
> lodepng version: 20230410
> libzip version: 1.5.1
> fontconfig version: 2.13.1 (runtime: 2.13.94)
> freetype version: 2.10.1 (runtime: 2.13.0)
> harfbuzz version: 2.6.4 (runtime: 7.0.0)
> cairo version: 1.16.0 (runtime: 1.17.6)
> lib3mf version: 1.8.1
> Features: fast-csg, fast-csg-safer, fast-csg-debug, manifold, roof, input-driver-dbus, lazy-union, vertex-object-renderers-indexing, textmetrics, import-function, predictible-output
> Application Path: /openscad/tools/build/usr/bin
> Documents Path: /root/.local/share
> User Documents Path: /root
> Resource Path: /openscad/tools/build/usr/share/openscad
> User Library Path: 
> User Config Path: 
> Backup Path: 
> OPENSCADPATH: /openscad/tools/../lib/dotSCAD/src:/openscad/tools/../common:/openscad/tools/../lib
> OpenSCAD library path:
>   /openscad/tools/../lib/dotSCAD/src
>   /openscad/tools/../common
>   /openscad/tools/../lib
>   
>   /openscad/tools/build/usr/share/openscad/libraries
> 
> OPENSCAD_FONT_PATH: /openscad/tools/../fonts
> OpenSCAD font path:
>   /usr/share/X11/fonts/Type1
>   /usr/share/X11/fonts/TTF
>   /root/.local/share/fonts
>   /usr/local/share/fonts
>   /usr/share/fonts
>   /root/.fonts
>   /usr/X11/lib/X11/fonts
>   /System/Library/Fonts
>   /Library/Fonts
>   /root/Library/Fonts
>   /usr/share/fonts/google-noto-cjk
>   /usr/share/fonts/google-noto-vf
> 
> 
> GL context creator: GLX (new)
> GLX version: 1.4
> PNG generator: lodepng
> 
> GLEW version: 2.1.0
> OpenGL Version: 4.5 (Compatibility Profile) Mesa 22.3.3
> GL Renderer: llvmpipe (LLVM 15.0.6, 256 bits)
> GL Vendor: Mesa/X.org
> RGBA(8888), depth(24), stencil(8)
> GL_ARB_framebuffer_object: yes
> GL_EXT_framebuffer_object: yes
> GL_EXT_packed_depth_stencil: yes

Command-line args:
`openscad -o output.png --summary all --view axes,scales --colorscheme Nature --autocenter --viewall --imgsize=600,600 --enable all --hardwarnings [....]
`
Stacktrace (sorry , no debug symbols in this build - why not include them in every snapshot?)

> Program received signal SIGSEGV, Segmentation fault.
> 0x000055ee0b2addc3 in polygon2dToPolySet(Polygon2d const&) ()
> (gdb) bt
> #0  0x000055ee0b2addc3 in polygon2dToPolySet(Polygon2d const&) ()
> #1  0x000055ee0b2ae645 in CSGTreeEvaluator::evaluateCSGNodeFromGeometry(State&, std::shared_ptr<Geometry const> const&, ModuleInstantiation const*, AbstractNode const&) ()
> #2  0x000055ee0b2af21c in CSGTreeEvaluator::visit(State&, CgalAdvNode const&) ()
> #3  0x000055ee0b36f3ac in NodeVisitor::traverse(AbstractNode const&, State const&) ()
> #4  0x000055ee0b36f307 in NodeVisitor::traverse(AbstractNode const&, State const&) ()
> #5  0x000055ee0b36f307 in NodeVisitor::traverse(AbstractNode const&, State const&) ()
> #6  0x000055ee0b36f307 in NodeVisitor::traverse(AbstractNode const&, State const&) ()
> #7  0x000055ee0b36f307 in NodeVisitor::traverse(AbstractNode const&, State const&) ()
> #8  0x000055ee0b36f307 in NodeVisitor::traverse(AbstractNode const&, State const&) ()
> #9  0x000055ee0b36f307 in NodeVisitor::traverse(AbstractNode const&, State const&) ()
> #10 0x000055ee0b36f307 in NodeVisitor::traverse(AbstractNode const&, State const&) ()
> #11 0x000055ee0b36f307 in NodeVisitor::traverse(AbstractNode const&, State const&) ()
> #12 0x000055ee0b36f307 in NodeVisitor::traverse(AbstractNode const&, State const&) ()
> #13 0x000055ee0b36f307 in NodeVisitor::traverse(AbstractNode const&, State const&) ()
> #14 0x000055ee0b36f307 in NodeVisitor::traverse(AbstractNode const&, State const&) ()
> #15 0x000055ee0b36f307 in NodeVisitor::traverse(AbstractNode const&, State const&) ()
> #16 0x000055ee0b36f307 in NodeVisitor::traverse(AbstractNode const&, State const&) ()
> #17 0x000055ee0b36f307 in NodeVisitor::traverse(AbstractNode const&, State const&) ()
> #18 0x000055ee0b36f307 in NodeVisitor::traverse(AbstractNode const&, State const&) ()
> #19 0x000055ee0b36f307 in NodeVisitor::traverse(AbstractNode const&, State const&) ()
> #20 0x000055ee0b36f307 in NodeVisitor::traverse(AbstractNode const&, State const&) ()
> #21 0x000055ee0b36f307 in NodeVisitor::traverse(AbstractNode const&, State const&) ()
> #22 0x000055ee0b2abb8b in CSGTreeEvaluator::buildCSGTree(AbstractNode const&) ()
> #23 0x000055ee0b570681 in CsgInfo::compile_products(Tree const&) ()
> #24 0x000055ee0b56f824 in prepare_preview(Tree&, ViewOptions const&, Camera&) ()
> #25 0x000055ee0b23de0b in do_export(CommandLine const&, RenderVariables const&, FileFormat, SourceFile*) ()
> #26 0x000055ee0b23f505 in cmdline(CommandLine const&) ()
> #27 0x000055ee0acedfb7 in main ()

As it only crashes from within my docker container it is a bit hard to give a small reproducible scenario :-(
I am very sure it worked before however.

My application is trying to render some license plate. Important to note is that it does not crash for every every license plate. It crashes for "1ABC001", but it does not crash for "1ABC002". Very strange.

pca006132, 2024-08-07:
maybe related? https://github.com/openscad/openscad/pull/5164

and apparently you are not using manifold in your command line args? this means that it is something with CGAL?

praetp, 2024-08-07:
> maybe related? #5164
> 
> and apparently you are not using manifold in your command line args? this means that it is something with CGAL?

Indeed, could be related to that commit.
Yes, i am using manifold. Note the `--enable all`. Omitting manifold also does not make a difference. If I remove all features (except textmetrics - I did that one), it still fails.

Workaround for me is to now use an older binary...

pca006132, 2024-08-14:
Reproducer: with https://github.com/UBaer21/UB.scad,

```scad
use <ub.scad>

Rund(.5) Voronoi(30,dicke=1);
```

This will crash when you preview it.

kintel, 2024-08-25:
Some discoveries and thoughts:
* This issue was introduced by #5164, which changed triangulation using CGAL to triangulate using indexes, effectively not allowing it to insert new vertices. This is important for composing manifold volumes for e.g. linear_extrude of complex polygons.
* Even before #5164, tests similar to the one provided here yielded non-manifold volumes and caused CGAL to fail performing boolean ops, so reverting that PR isn't really the best solution.
* The root cause: 2D CSG (including offset) is performed using Clipper V1 in fixed-point `int64` object space, while polygon2d itself (and thus CGAL triangulation) uses `double` world-space. While Clipper guarantees no intersections and overlaps, converting from int64 to double may result in precision loss and introduce aliasing which yields tiny intersections.

kintel, 2024-08-25:
Ideally, we could fix this by using a triangulator which doesn't introduce new vertices but still allows intersections. Manifold offers such a triangulator, but it's not available for non-experimental builds. This may become a longer-term possibility, but short-term it's not doable.

Another possible fix is to reduce the precision in int64 space to reduce loss when converting to double. I suggest starting with reducing the precision with a single bit and see where that leads.

kintel, 2024-09-15:
While #5271 made this a bit more robust, we still have occasional failures.
The following design processed as `openscad zebra.scad -o out.png` now crashes with `Assertion failed: (fit->vertex(1)->info().id != -1), function createTriangulatedPolySetFromPolygon2d, file cgalutils-triangulate.cc, line 151`.

```
$fn=64;
text(text="R",font="naturePro",size=5);
```

kintel, 2024-10-01:
Fixed by #5282

commit df9595de (2024-09-15): Reduce Clipper precision by one bit. Fixes #5253 (#5271)

* Unify Polygon2d creation
* Reduce clipper precision by one bit
* Safer handling and assertion of vertices inserted by CGAL's triangulator
closed 2024-10-01 by kintel