openscad/openscad #5256: Include BOSL2 into the build like it is done with MCAD since 2011.

yahbluez, 2024-08-10:
The headline says it all.

We have the unfinished outdated MCAD with deprecated modules as part of the distribution.
I like to recommend that the mighty BOSL2 gets part of the distribution.

t-paul, 2024-08-10:
Combining release timelines even with just a single library is difficult and would cause other issues like shipping an outdated BOSL2. If all goes well, a (minimalistic) [OpenSCAD Library Manager](https://summerofcode.withgoogle.com/programs/2024/projects/lnSRboG6) will go live in the not too distant future. Hopefully this will then be adopted and solve the problem in a more general way.

closed 2024-08-10 by t-paul

yahbluez, 2024-08-10:
A kind of lib manager is a nice idea. 

But, as we have today the totally outdated MCAD vs the frequently updated BOSL2, i think that especially this lib the BOSL2 should be part of the OpenSCAD package in the same way we have today the outdated MCAD "embedded".

Why BOSL2?
Because this huge lib will help many new user to use the full power of openscad without the frustration to reinvent the wheel each time.

It is a good idea to hold the language small and tight but a standardized  high quality lib would help many users to write better code.

nophead, 2024-08-10:
I personally don't use any libraries apart from my own so I don't want a huge library shipped with OpenSCAD. Probably MCAD should be removed as it isn't maintained.

adrianVmariano, 2024-08-11:
We think that it does not make sense to package BOSL2 with OpenSCAD because of the problem noted above about synchronization of releases.   It will tend to freeze people with a version of BOSL2 that doesn't include bug fixes, for example.   And BOSL2 is not the only library.  Who knows what the future may bring.  

I do agree that removing the outdated and unmaintained MCAD is probably worth considering.  That is, why give MCAD special status by shipping it with OpenSCAD given that it probably should only be used for legacy purposes these days.

gsohler, 2024-08-12:
> The argumentation, that the BOSL2 lib may be outdated is not valid.
>
> Today we see a 13 year old lib (MCAD) bundled with openscad and this lib
> still works but may break in the future because of some deprecated openscad
> features.
>
> I second the idea to remove MCAD, it is to old and miss leads new users
> into the wrong direction.
>
> BOSL2 is up to date and enhanced openscad by tons of every day needed
> functions and modules and that in a clean and proper way.
>
> This lib can help many users to write better cleaner code.
>
> If i compare the situation with C++ or Python both show us that including
> a kind of standard lib is a very good idea, i think we should do too with
> openscad and in my opinion BOSL2 should be this standard lib.

Every other lib listed in

https://openscad.org/libraries.html

could also claim rights to become the openscad std lib.

but they don't.

I could claim  my pylibfive which can do fantastic SDF merging to become
the standard lib, but I'd rather wait until it becomes compatible. :)

If openscad's authors are hard do convince to match to a certain "partner"
you might reconsider the matching parameters on a broader level ?

yahbluez, 2024-08-13:
My point was, that today we have a outdated by one decade lib, as part of the distribution. Even as part of the nightly dev  snapshots and i suggested to change that against the most universell lib we have today for openscad.

No more no less.

Finally that is a decision only the developers can make.

jordanbrown0, 2024-08-13:
Note that Adrian is one of the developers of BOSL2.  If they don't want it included, that's pretty much the end of the conversation.

A package manager may help, and perhaps some combination of the package manager and OpenSCAD itself could provide one- or two-click bootstrapping of "registered" libraries.

jordanbrown0, 2024-08-13:
Note also that there's no requirement that there only be one library included.  The distribution could include several. But the update problem that Adrian raises is real for any of them.

yahbluez, 2024-08-13:
Yah, that is also true. Adding the whole list would not be a big deal today.

Is the update problem real? Openscad is solid for years and code to check if a lib may need an update would be easy. Many software today checks daily for updates. 

I just wonder that this simple request gets so much negative opinions. 
I see BOSL2 as a huge improvement for openscad.

But never mind the developers can do what ever they want to do. Their code their rules.