openscad/openscad #5258: GPU Support

Pizzanomicon, 2024-08-11:
**Is your feature request related to a problem? Please describe.**
I'm always frustrated when previewing/rendering a drawing.  I feel like this could be a much faster process.

**Describe the solution you'd like**
Utilizing a GPU, perhaps specifically CUDA, could drastically speed up processes in this software by orders of magnitude.

pca006132, 2024-08-11:
Did you try the latest snapshot with manifold enabled?

And GPU acceleration is not that simple.

t-paul, 2024-08-11:
For some additional context Manifold had CUDA support and @pca006132 knows that very well including performance measurements. CUDA did not provide any reasonable benefit for the specific work load OpenSCAD generates. Also there's not enough dev resources for considering such a platform limited (Nvidia controlled) solution.
Manifold already brings speed improvement by (usually) multiple orders of magnitude including multithreading support. If you want to give that a try, use a dev snapshot and enable Manifold (Preferences -> Features -> Manifold).

closed 2024-08-11 by t-paul