openscad/openscad #5259: Crash in opening on Mac Sequoia Beta 15.0 24A5320a

wnc2, 2024-08-13:
**Describe the bug**
Openscad 2024-08-12 crashes on opening.

latest MacOs Sequoia Beta 15.0 24A5320a (updated today)

-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Incident Identifier: 04C09493-F5BD-4D4C-BD02-7339E5BBCB98
CrashReporter Key:   87B578F4-F403-F16E-7F9B-AAC282E6B5E1
Hardware Model:      iMac19,1
Process:             OpenSCAD [6993]
Path:                /Applications/OpenSCAD.app/Contents/MacOS/OpenSCAD
Identifier:          org.openscad.OpenSCAD
Version:             2024.8 (2024.08.12)
Code Type:           X86-64 (Native)
Role:                Default
Parent Process:      launchd [1]
Coalition:           org.openscad.OpenSCAD [4732]

Date/Time:           2024-08-13 12:52:15.2200 +0200
Launch Time:         2024-08-13 12:52:15.2082 +0200
OS Version:          macOS 15.0 (24A5320a)
Release Type:        User
Report Version:      104

Exception Type:  EXC_CRASH (SIGKILL (Code Signature Invalid))
Exception Codes: 0x0000000000000000, 0x0000000000000000
Termination Reason: CODESIGNING 4 Launch Constraint Violation

Triggered by Thread:  0

Thread 0 Crashed:
0   dyld_path_missing             	       0x116fd4000 _dyld_start + 0
1   main_executable_path_missing  	       0x107305000 ???

Thread 0 crashed with X86 Thread State (64-bit):
  rax: 0x0000000000000000  rbx: 0x0000000000000000  rcx: 0x0000000000000000  rdx: 0x0000000000000000
  rdi: 0x0000000000000000  rsi: 0x0000000000000000  rbp: 0x0000000000000000  rsp: 0x00007ff7b8bfabe8
   r8: 0x0000000000000000   r9: 0x0000000000000000  r10: 0x0000000000000000  r11: 0x0000000000000000
  r12: 0x0000000000000000  r13: 0x0000000000000000  r14: 0x0000000000000000  r15: 0x0000000000000000
  rip: 0x0000000116fd4000  rfl: 0x0000000000000200  cr2: 0x0000000000000000
  
Logical CPU:     0
Error Code:      0x00000000 
Trap Number:     0

Binary Images:
       0x116fcf000 -        0x11705bfff dyld_path_missing (*) <a132854e-5ad7-366c-ac92-eadf3567d70a> /dyld_path_missing
       0x107305000 -        0x108257fff main_executable_path_missing (*) <2b27ba8a-1c08-3fe5-a70e-54993a73e24e> /main_executable_path_missing
               0x0 - 0xffffffffffffffff ??? (*) <00000000-0000-0000-0000-000000000000> ???

Error Formulating Crash Report:
dyld_process_snapshot_get_shared_cache failed

EOF

-----------
Full Report
-----------

{"app_name":"OpenSCAD","timestamp":"2024-08-13 12:52:15.00 +0200","app_version":"2024.8","slice_uuid":"2b27ba8a-1c08-3fe5-a70e-54993a73e24e","build_version":"2024.08.12","platform":0,"bundleID":"org.openscad.OpenSCAD","share_with_app_devs":1,"is_first_party":0,"bug_type":"309","os_version":"macOS 15.0 (24A5320a)","roots_installed":0,"name":"OpenSCAD","incident_id":"04C09493-F5BD-4D4C-BD02-7339E5BBCB98"}
{
  "uptime" : 10000,
  "procRole" : "Default",
  "version" : 2,
  "userID" : 501,
  "deployVersion" : 210,
  "modelCode" : "iMac19,1",
[... 111 lines cut ...]
Bluetooth: Version (null), 0 services, 0 devices, 0 incoming serial ports
Network Service: Ethernet, Ethernet, en0
Network Service: Wi-Fi, AirPort, en1
USB Device: VLI Product String
USB Device: VLI Product String
USB Device: USB31Bus
USB Device: USB3.0 Hub
USB Device: USB3.0 Hub
USB Device: USB3.0 Hub
USB Device: USB3.0 Hub
USB Device: USB3.0 Hub
USB Device: USB3.0 Hub
USB Device: UHS-II SD Reader
USB Device: Cam Link 4K
USB Device: USB2.0 Hub
USB Device: iPad
USB Device: Yeti Stereo Microphone
USB Device: FaceTime HD Camera (Built-in)
USB Device: USB2.0 Hub
USB Device: USB2.0 Hub
USB Device: Stream Deck
USB Device: USB2.0 Hub
USB Device: USB Receiver
USB Device: VLI Product String
Thunderbolt Bus: iMac, Apple Inc., 47.5

kintel, 2024-08-15:
Did you authorize the app in System Settings as well, as shown in https://www.youtube.com/watch?v=Dykgs4bdjB0 ?

Fwiw, I just installed macOS 15.1 beta (Apple Silicon) in a VM and everything works fine there, both the 2021.01 release and 2024.08.12.

wnc2, 2024-08-20:
yes i authorized all, BTW the "old" official 2021.01 works great, i have problems with the last beta available

benjaminlau, 2024-08-21:
I experienced a similar crash with my iMac 2020 ( Intel ) running on OSX Sequoia Beta 4.

I have installed the OpenSCAD 2024.08.12 a few days ago and it run fine until today.
I updated to the latest OpenSCAD 2024.08.10 and it has a similar crash problem at startup.

Apple could not verify “OpenSCAD.2024.08.19” is free of malware that may harm your Mac or compromise your privacy.

I reinstalled the OpenSCAD 2021-01 stable and that opens fine.

Here are the crashed log associated with the crash
Path:                /Applications/OpenSCAD.app/Contents/MacOS/OpenSCAD
Identifier:          org.openscad.OpenSCAD
Version:             2024.8 (2024.08.19)
Code Type:           X86-64 (Native)
Role:                Default
Parent Process:      launchd [1]
Coalition:           org.openscad.OpenSCAD [34068]

Date/Time:           2024-08-21 14:28:20.6894 -0700
Launch Time:         2024-08-21 14:28:20.6705 -0700
OS Version:          macOS 15.0 (24A5320a)
Release Type:        User
Report Version:      104

Exception Type:  EXC_CRASH (SIGKILL (Code Signature Invalid))
Exception Codes: 0x0000000000000000, 0x0000000000000000
Termination Reason: CODESIGNING 4 Launch Constraint Violation

Triggered by Thread:  0

Thread 0 Crashed:
0   dyld_path_missing             	       0x116291000 _dyld_start + 0
1   main_executable_path_missing  	       0x1067e7000 ???

Thread 0 crashed with X86 Thread State (64-bit):
  rax: 0x0000000000000000  rbx: 0x0000000000000000  rcx: 0x0000000000000000  rdx: 0x0000000000000000
  rdi: 0x0000000000000000  rsi: 0x0000000000000000  rbp: 0x0000000000000000  rsp: 0x00007ff7b9718bd8
   r8: 0x0000000000000000   r9: 0x0000000000000000  r10: 0x0000000000000000  r11: 0x0000000000000000
  r12: 0x0000000000000000  r13: 0x0000000000000000  r14: 0x0000000000000000  r15: 0x0000000000000000
  rip: 0x0000000116291000  rfl: 0x0000000000000200  cr2: 0x0000000000000000
  
Logical CPU:     0
Error Code:      0x00000000 

======================

Please note that I have opened setting / privacy & security / Security  to allow for launch and it crashes anyway.

======================

Thank you for the great work on a wonderful product.  Hope this get resolved soon.

wnc2, 2024-08-27:
same problem with latest OpenSCAD-2024.08.26.dmg

kintel, 2024-08-27:
I wonder if this issue is limited to Intel Macs. Have anyone else experienced this, on Intel or Apple Silicon?

pkondratev, 2024-08-28:
> I wonder if this issue is limited to Intel Macs. Have anyone else experienced this, on Intel or Apple Silicon?

The same issue on Intel Mac with 15.0 (24A5327a)

kintel, 2024-08-28:
@pkondratev on bare metal or in a VM? I've only tested 15.1 in a VM and it worked fine.

pkondratev, 2024-08-28:
@kintel MacBook Pro 2018, Intel Core i5, bare metal

kintel, 2024-08-28:
Anyone experiencing this on 15.1 Beta?

benjaminlau, 2024-08-28:
> I wonder if this issue is limited to Intel Macs. Have anyone else experienced this, on Intel or Apple Silicon?

Experiencing it on Intel iMac 2020

kintel, 2024-09-03:
So far, all failures are on 15.0 beta on Intel.
Has anyone seen a failure on 15.0 on Apple Silicon, or a failure on 15.1 on any CPU?

ionif, 2024-09-13:
I'm testing on a 2021 Macbook with M1 Max and it doesn't crash but any time I try to interact with the app (resize, or modify anything) I get a long rainbow spinner. This laptop has 32 gb of ram so somethings definitely weird.

kintel, 2024-09-13:
@ionif Testing in a VM or on bare metal?

ionif, 2024-09-13:
I'm testing on a macbook so no vm and I'm on Monterey 12.5.1. Also I'm using the premade image, I didn't build from scratch.

kintel, 2024-09-13:
@ionif This issue is about macOS 15 Beta. For other topics, please open a new issue.

wnc2, 2024-09-13:
stil crashing on latest Sequoia 15.0 (24A335) and latest openscad 2024.09.08

the crash occurs just at start, then the auth request opens, ok authorized, but it crashes at every opening, same behaviors of previous version.

i can't test 15.1 due it is a production Mac, i can't risk a developer MacOs
-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Incident Identifier: 921165F4-CA69-4491-95EB-1978413A07B8
CrashReporter Key:   87B578F4-F403-F16E-7F9B-AAC282E6B5E1
Hardware Model:      iMac19,1
Process:             OpenSCAD [72970]
Path:                /Applications/OpenSCAD.app/Contents/MacOS/OpenSCAD
Identifier:          org.openscad.OpenSCAD
Version:             2024.9 (2024.09.08)
Code Type:           X86-64 (Native)
Role:                Default
Parent Process:      launchd [1]
Coalition:           org.openscad.OpenSCAD [24917]

Date/Time:           2024-09-13 14:06:22.5527 +0200
Launch Time:         2024-09-13 14:06:22.5380 +0200
OS Version:          macOS 15.0 (24A335)
Release Type:        User
Report Version:      104

Exception Type:  EXC_CRASH (SIGKILL (Code Signature Invalid))
Exception Codes: 0x0000000000000000, 0x0000000000000000
Termination Reason: CODESIGNING 4 Launch Constraint Violation

Triggered by Thread:  0

Thread 0 Crashed:
0   dyld_path_missing             	       0x1082b4000 _dyld_start + 0
1   main_executable_path_missing  	       0x104eb5000 ???

Thread 0 crashed with X86 Thread State (64-bit):
  rax: 0x0000000000000000  rbx: 0x0000000000000000  rcx: 0x0000000000000000  rdx: 0x0000000000000000
  rdi: 0x0000000000000000  rsi: 0x0000000000000000  rbp: 0x0000000000000000  rsp: 0x00007ff7bb04ab88
   r8: 0x0000000000000000   r9: 0x0000000000000000  r10: 0x0000000000000000  r11: 0x0000000000000000
  r12: 0x0000000000000000  r13: 0x0000000000000000  r14: 0x0000000000000000  r15: 0x0000000000000000
  rip: 0x00000001082b4000  rfl: 0x0000000000000200  cr2: 0x0000000000000000
  
Logical CPU:     0
Error Code:      0x00000000 
Trap Number:     0

Binary Images:
       0x1082af000 -        0x10833bfff dyld_path_missing (*) <e6056c94-fc2d-3517-b1e1-46d8eb58a10e> /dyld_path_missing
       0x104eb5000 -        0x105e09fff main_executable_path_missing (*) <6f8a437b-df5a-356a-8d0e-4e5a815105c2> /main_executable_path_missing
               0x0 - 0xffffffffffffffff ??? (*) <00000000-0000-0000-0000-000000000000> ???

Error Formulating Crash Report:
dyld_process_snapshot_get_shared_cache failed

EOF

-----------
Full Report
-----------

{"app_name":"OpenSCAD","timestamp":"2024-09-13 14:06:22.00 +0200","app_version":"2024.9","slice_uuid":"6f8a437b-df5a-356a-8d0e-4e5a815105c2","build_version":"2024.09.08","platform":0,"bundleID":"org.openscad.OpenSCAD","share_with_app_devs":1,"is_first_party":0,"bug_type":"309","os_version":"macOS 15.0 (24A335)","roots_installed":0,"name":"OpenSCAD","incident_id":"921165F4-CA69-4491-95EB-1978413A07B8"}
{
  "uptime" : 110000,
  "procRole" : "Default",
  "version" : 2,
  "userID" : 501,
  "deployVersion" : 210,
  "modelCode" : "iMac19,1",
[... 111 lines cut ...]
AirPort: spairport_wireless_card_type_wifi (0x14E4, 0x7BF), wl0: Jul 26 2024 22:09:35 version 9.30.514.0.32.5.94 FWID 01-47278712
AirPort: 
Bluetooth: Version (null), 0 services, 0 devices, 0 incoming serial ports
Network Service: Ethernet, Ethernet, en0
Network Service: Wi-Fi, AirPort, en1
USB Device: VLI Product String
USB Device: VLI Product String
USB Device: USB31Bus
USB Device: USB3.0 Hub
USB Device: USB3.0 Hub
USB Device: USB3.0 Hub
USB Device: USB3.0 Hub
USB Device: USB3.0 Hub
USB Device: USB3.0 Hub
USB Device: UHS-II SD Reader
USB Device: Cam Link 4K
USB Device: USB2.0 Hub
USB Device: Yeti Stereo Microphone
USB Device: FaceTime HD Camera (Built-in)
USB Device: USB2.0 Hub
USB Device: USB2.0 Hub
USB Device: Stream Deck
USB Device: USB2.0 Hub
USB Device: USB Receiver
Thunderbolt Bus: iMac, Apple Inc., 47.5

kintel, 2024-09-13:
When I find some time I'll give 15.0 a shot in a VM and see how 15.0 vs. 15.1 behaves for me (Apple Silicon only; I don't have an Intel machine)

rsnodgrass, 2024-09-14:
Same crashing issue on MacOS 15 :(

kintel, 2024-09-14:
@rsnodgrass Are you also on 15.0 on Intel?

rsnodgrass, 2024-09-14:
Yes!

pkondratev, 2024-09-23:
MacOS Version: Beta 15.1 (24B5046f)
Hardware Version: 2,3 GHz 4‑core Intel Core i5
OpenSCAD-2024.09.22
Same error. Does anybody know any fix for this problem?

<details>
<summary>MacOS Error Report</summary>

```

-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Incident Identifier: 2DE4D8C1-BC0A-4E06-A526-13BFF43E4EEC
CrashReporter Key:   9C6FF535-7198-C001-752D-5AD853F59C09
Hardware Model:      MacBookPro15,2
Process:             OpenSCAD [3080]
Path:                /Applications/OpenSCAD.app/Contents/MacOS/OpenSCAD
Identifier:          org.openscad.OpenSCAD
Version:             2024.9 (2024.09.22)
Code Type:           X86-64 (Native)
Role:                Default
Parent Process:      launchd [1]
Coalition:           org.openscad.OpenSCAD [2281]

Date/Time:           2024-09-23 15:19:37.8532 +0300
Launch Time:         2024-09-23 15:19:37.6375 +0300
OS Version:          macOS 15.1 (24B5046f)
Release Type:        User
Report Version:      104

Exception Type:  EXC_CRASH (SIGKILL (Code Signature Invalid))
Exception Codes: 0x0000000000000000, 0x0000000000000000
Termination Reason: CODESIGNING 4 Launch Constraint Violation

Triggered by Thread:  0

Thread 0 Crashed:
0   dyld_path_missing             	       0x116832000 _dyld_start + 0
1   ???                           	               0x1 ???

Thread 0 crashed with X86 Thread State (64-bit):
  rax: 0x0000000000000000  rbx: 0x0000000000000000  rcx: 0x0000000000000000  rdx: 0x0000000000000000
  rdi: 0x0000000000000000  rsi: 0x0000000000000000  rbp: 0x0000000000000000  rsp: 0x00007ff7b0712bd0
   r8: 0x0000000000000000   r9: 0x0000000000000000  r10: 0x0000000000000000  r11: 0x0000000000000000
  r12: 0x0000000000000000  r13: 0x0000000000000000  r14: 0x0000000000000000  r15: 0x0000000000000000
  rip: 0x0000000116832000  rfl: 0x0000000000000200  cr2: 0x0000000000000000
  
Logical CPU:     0
Error Code:      0x00000000 
Trap Number:     0

Binary Images:
       0x11682d000 -        0x1168b9fff dyld_path_missing (*) <ece08070-2194-3e61-9371-9216cc755f55> /dyld_path_missing
       0x10f7ed000 -        0x110752fff main_executable_path_missing (*) <bf1c4eb4-c6b6-3ead-b171-25f9a33809c3> /main_executable_path_missing
               0x0 - 0xffffffffffffffff ??? (*) <00000000-0000-0000-0000-000000000000> ???

Error Formulating Crash Report:
dyld_process_snapshot_get_shared_cache failed

EOF
```

</details>

Endymion42, 2024-09-30:
This seems to fix the problem for me:

codesign --force --deep --sign - /Applications/OpenSCAD.app

kintel, 2024-10-01:
Could someone give this one a try: https://output.circle-artifacts.com/output/job/048b2b16-9a8e-4b31-ab76-05f95b7594ff/artifacts/0/tmp/out/OpenSCAD-2024.10.01.dmg

Endymion42, 2024-10-01:
That version still crashes.  After doing codesign --force --deep --sign - /Applications/OpenSCAD.app it starts working.

kintel, 2024-10-01:
Thanks for the quick test. I'll see if I can codesign in a different way on the CI.

kintel, 2024-10-11:
Thanks for the exploration. It feels a bit like this is a bug in the Beta macOS release.

We should still sign the binary though :)

borisfeldman, 2024-10-14:
> That version still crashes. After doing codesign --force --deep --sign - /Applications/OpenSCAD.app it starts working.

This fixed the issue for me on Sequoia 15.0 Intel. No problems on Sequoia 15.0 ARM.

kintel, 2024-10-21:
@thestumbler This is the first I hear of such an issue. The summary of the issues reported above is that unsigned apps fail to load on Intel, but work on Apple Silicon. Cmd-line has so far not been reported to behave differently from GUI. I can run the 2024.10.20 development snapshot in cmd-line on a 15.1 beta VM without issues.

kintel, 2024-10-29:
Has anyone tested the release versions of 15.0 and 15.1 on Intel, to verify if this issue is still present?

Endymion42, 2024-10-29:
1) On 15.1 iMac Pro Intel the release version, , gives this popup when trying to launch:  The application “Finder” does not have permission to open “(null).”

2) If I then go to OpenSCAD.app/Contents/MacOS and try to run OpenSCAD, it warns me: Apple could not verify “OpenSCAD” is free of malware that may harm your Mac or compromise your privacy

3) At this point the “OpenSCAD” was blocked to protect your Mac  with the Open Anyway button appears in System Settings/Privacy & Security

4) Clicking on that pops up a Move to Trash/Open Anyway/Done dialog box.  Clicking on Open Anyway opens another box saying OpenSCAD.app Not Opened (cause couldn’t verify free of malware) with Done/Move to Trash buttons.  Done button clears box.

5) OpenSCAD still won’t open (get the could not verify free from malware box) when executed from Contents/MacOS.  From Applications directory trying to execute results in (1)

6) Running codesign --force --deep --sign - /Applications/OpenSCAD.app 
from the command line then running OpenSCAD from Contents/MacOs then starts a Open Anyway cycle that succeeds.   OpenScad will run that way.

7) OpenSCAD from the Applications directory still does (1).

They are not making it easy…

I will try the OpenSCAD-2023.10.27 release on 15.1 and report back.

Regards,
John

Endymion42, 2024-10-29:
With 10.27, running OpenSCAD from Applications results in a crash:

-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Incident Identifier: C00DE8DD-C539-48E2-9875-02F3BEC15B47
CrashReporter Key:   69E9916F-B9EB-4BEB-9F4D-9C40FE620B21
Hardware Model:      iMacPro1,1
Process:             OpenSCAD [3227]
Path:                /Applications/OpenSCAD.app/Contents/MacOS/OpenSCAD
Identifier:          org.openscad.OpenSCAD
Version:             2024.10 (2024.10.27)
Code Type:           X86-64 (Native)
Role:                Default
Parent Process:      launchd [1]
Coalition:           org.openscad.OpenSCAD [3664]

Date/Time:           2024-10-28 20:33:22.7829 -0700
Launch Time:         2024-10-28 20:33:22.7702 -0700
OS Version:          macOS 15.1 (24B83)
Release Type:        User
Report Version:      104

Exception Type:  EXC_CRASH (SIGKILL (Code Signature Invalid))
Exception Codes: 0x0000000000000000, 0x0000000000000000
Termination Reason: CODESIGNING 4 Launch Constraint Violation

Triggered by Thread:  0

Thread 0 Crashed:
0   dyld_path_missing             	       0x113a90000 _dyld_start + 0
1   main_executable_path_missing  	       0x10cb27000 ???

Thread 0 crashed with X86 Thread State (64-bit):
  rax: 0x0000000000000000  rbx: 0x0000000000000000  rcx: 0x0000000000000000  rdx: 0x0000000000000000
  rdi: 0x0000000000000000  rsi: 0x0000000000000000  rbp: 0x0000000000000000  rsp: 0x00007ff7b33d8b88
   r8: 0x0000000000000000   r9: 0x0000000000000000  r10: 0x0000000000000000  r11: 0x0000000000000000
  r12: 0x0000000000000000  r13: 0x0000000000000000  r14: 0x0000000000000000  r15: 0x0000000000000000
  rip: 0x0000000113a90000  rfl: 0x0000000000000200  cr2: 0x0000000000000000
  
Logical CPU:     0
Error Code:      0x00000000 
Trap Number:     0

Binary Images:
       0x113a8b000 -        0x113b17fff dyld_path_missing (*) <4d52bd1e-6a0e-31db-b564-9e2029fdcd6f> /dyld_path_missing
       0x10cb27000 -        0x10dab0fff main_executable_path_missing (*) <acafd0b4-22cb-32c4-b922-ab15342a75a7> /main_executable_path_missing
               0x0 - 0xffffffffffffffff ??? (*) <00000000-0000-0000-0000-000000000000> ???

Error Formulating Crash Report:
dyld_process_snapshot_get_shared_cache failed

EOF

Running from COntents/MacOS works after giving permission via System Settings and then Applications/OpenSCAD works too without crashing.

Confusing.

Regards,
John

splaisan, 2024-11-04:
> This seems to fix the problem for me:
> 
> codesign --force --deep --sign - /Applications/OpenSCAD.app

thanks for sharing this command
this solved it for me too as alternative solution to running the latest git version 2024_11-01 which also works on sequoia 15.1, M2 mac.

closed 2024-11-04 by kintel