openscad/openscad #5261: Axis numbering improvement suggestions

thehans, 2024-08-18:
After recent merge of #5247 I was checking how it behaved at various zoom levels.  
Overall this is a much welcomed improvement, but it drew my attention to some areas that I think could be further improved upon.

The textual content (specific formatting of numbers with exponents, etc) was not altered in the aforementioned PR, but I think doing this could offer even better results.

The new font results in very slightly more overlap in the most extreme cases.
Two particular problematic situations I noticed are around the transition to exponent display forms, at both large and small scales, as shown in the screenshots below.

Suggested changes:
* Currently the points that crossover to exponent form are `>= 1e6` and `<= 1e5`
    I would suggest reducing this by at least one order of magnitude to `>= 1e5` and `<= 1e4`
* Exponent numbers (after `e`) are given a leading zero, which I don't see a need for, it only adds unnecessary width.
* As seen in screenshots below, tick labels can end up with mixed formatting around these transition points.
  If possible, i think it would be better to decide overall formatting (exponent or not) based on the overall tick scale/spacing, and not the individual tick value, so these mixed formatting does not occur.
* The negative exponent symbol (at small scales) appears excessively wide, though I suspect this is some kind of illusion and the characters are actually fixed width?  In which case this last point is not really actionable without significant added complexity.

Large scale (new)
[image]
vs. old
[image]

And small scale new
[image]
vs. old
[image]

hzeller, 2024-08-18:
The new font rendering indeed encourages to look at axis numberings nobody would otherwise care to look a before :)

I think a simple way to reduce overlap' we could probably fix by slightly reducing the font size. The new font is more readable than before, so having it overall a tiny bit smaller probably doesn't harm.

The conversion of number to text is currently done (as before) uses the `STR()` function, which is based on the standard c++ `std::ostream` formatting. There is certainly room for 'nicer' rendering, e.g. without leading zeroes in the exponent, or we even could go fancy when writing things and drawing the exponent with a smaller font above as superscript ? `
So 4<sup>6</sup> instead of 4e+06.

One could also think of appending SI suffixes so `1e+06` would be `1M`. But I think that would be confusing as people often think in either millimeter (so 1 Mega-Millimeter would be 1000 Meter, but it is is somewhat unusual to write it like that).

I think I'd be in favor of maybe reducing the font size slightly for now and worry about different formatting of these values later (TBH: nobody would really model soo large thigns that the exponent representation really is encountered in the everyday use...).

jordanbrown0, 2024-08-18:
4<sup>6</sup> means 4 to the sixth power, `4*4*4*4*4*4`.  `4e+06` means 4 times ten to the sixth power, `4*10*10*10*10*10*10`.

If you wanted to use superscript notation, it'd be something like 4*10<sup>6</sup>, but that doesn't save any space.

hzeller, 2024-08-18:
yeah, I was typing too quickly before coffee :), of course I meant 4*10<sup>6</sup>. Anyway, was just a suggestion, but given that this is essentially never needed, the complexity does not warrant the result.
I think slightly making the font smaller should be good for now.