openscad/openscad #5263: Model corrupted with repeated translate (larger translation = more corruption)

blb75, 2024-08-19:
**Describe the bug**
I have a complex shape (tree structure) that is made with pairwise hulls (not sure if relevant but I couldn't reproduce this error with just a sphere).

I've attached an STL of it to make a minimal example that fails for debugging purposes, but please note that the issue occurs whether it is loaded from STL or from its native module.

**To Reproduce**
Steps to reproduce the behavior:
Repeat the STL with a translation. The larger the translation, the fewer repeats needed to create the corruption.
Even with many repeats, if the translation is small, the model remains as expected.
Even with a single repeat, if the translation is large, the model is corrupted.

**Expected behavior**
The model fidelity is independent of the translation magnitude.

**Code reproducing the issue**
//This will fail:
for (it_repeat=[0:1])
{
translate([100000*it_repeat,0,0])
    import("minimal_fail.stl");
}

//This will not, if it is alone (above code is commented out)
//If both for loops are run, this will also fail (whether this is run first or second)
for (it_repeat=[0:10])
{
translate([10*it_repeat,0,0])
    import("minimal_fail.stl");
    }

**Screenshots**
If applicable, add screenshots to help explain your problem.
Base model
[image]
Many small translation repeats
[image]
One large translation repeat
[image]
[minimal_fail.stl.zip](https://github.com/user-attachments/files/16665645/minimal_fail.stl.zip)

**Environment and Version info (please complete the following information):**
 - OS: macOS 13.6.7
 - System: Intel Mac
 - OpenSCAD Version 2024.08.11

Thank you!

UBaer21, 2024-08-19:
can't reproduce on windows

thehans, 2024-08-19:
I can reproduce on Linux, with manifold enabled.  
I was expecting the far away model to show more degradation, but surprisingly the origin-centered one is most affected.  (There is some artifact on the far object, but much less pronounced).
[image]

As far as I know, its an issue of precision, where the model coordinate values are converted to 32-bit float for manifold and this causes problems when combined with large absolute coordinates.

I think there is a possibility of manifold supporting 64bit float in the future which would alleviate this.

Seems like a duplicate of #4871

closed 2024-08-19 by thehans

pca006132, 2024-08-20:
Not that far in the future: https://github.com/elalish/manifold/pull/895

Do note that this is not really about repeated translates, I think if you do many small translates with the same resulting translation as the large translates, they should have similar effect.

There is also another way to alleviate this with lower precision, by normalizing each object (scale and translate to [-1, 1]) and maintain this normalization parameters. Probably not worth it though as we are switching to double precision.