openscad/openscad #5266: issue with difference()  with for loop with variable unused in body?

Parkinbot, 2024-08-22:
the following code bricks OpenSCAD 2024 (different builds) on my machine. Once I uncomment the `rotate(i)` there is no

```
difference()
{
  circle(100); 
  for(i=[0:60:359])
    //rotate(i)
    translate([50,0,0])
    difference()
    {
      circle(8); 
      circle(4); 
    }  
}  
```
**Environment and Version info (please complete the following information):**
 - OS: Windows 11 
 - System: Intel PC 64-bit
 - OpenSCAD Version 2024.08.18 (git a16ca2a67)

**Library & Graphics card information**
OpenSCAD Version: 2024.08.18 (git a16ca2a67)
System information: Microsoft Windows 11 (10.0.22631) x86_64 32 CPUs 63.75 GB RAM
User Agent: OpenSCAD/2024.08.18 (git a16ca2a67) (Microsoft Windows 11 (10.0.22631) x86_64)
Compiler: GCC "13.2.0" 64bit
MinGW build: MingW64
Debug build: No
Boost version: 1_81
Eigen version: 3.4.0
CGAL version, kernels: 5.5, Cartesian<Gmpq>, Extended_cartesian<Gmpq>, Epeck
OpenCSG version: OpenCSG 1.5.0
Qt version: 5.15.11
QScintilla version: 2.11.2
InputDrivers: 
GLib version: 2.70.2
lodepng version: 20230410
libzip version: 1.5.2
fontconfig version: 2.14.2
freetype version: 2.13.2
harfbuzz version: 7.3.0
cairo version: 1.16.0
lib3mf version: 1.8.1
**Additional context**
Add any other context about the problem here.

UBaer21, 2024-08-22:
[image]

kintel, 2024-08-22:
Not reproducible on macOS either.
@Parkinbot Does it crash with any error message?

Parkinbot, 2024-08-23:
Looks like my machine needed a reboot. Sorry guys.

closed 2024-08-23 by Parkinbot