openscad/openscad #5274: Non working contextual menu over 3d preview on Ubuntu+NVIDIA

damienmarchal, 2024-08-27:
**Describe the bug**
The right click on a 3d model does works properly when I'm using the Nvidia driver & GPU while it does when I'm using the Intel integrated graphics card. I wonder if this is something wrong on my system (like driver's or gl includes) or if someone can reproduce it. 

**To reproduce the expected behavior**
0 - start openscad forcing the intel IGP. 
1 - open the [provided file](https://github.com/user-attachments/files/16761259/problem.scad.txt)
2 - right click one of the logo, a menu with the construction stack pops up
3 - right click on the other logo, a menu with the construction stack pops up

**To Reproduce the bug**
0 - start openscad forcing the Nvidia GPU. 
1 - open the same file as before, 
2 - right click one of the logo, a menu with the construction stack pops up
3 - right click on the other logo, on that one no menu appears 
(appearance of step 2 & 3 can be flipped depending on the order you choose)

**Screenshots**
I made a small video in which I run the previously described scenario, after 0:30 I'm right clicking and nothing pops-up.
https://www.youtube.com/watch?v=9LJrF5N60do

**Environment and Version info (please complete the following information):**
 - OS: Ubuntu 23.04
 - System: Intel x86
 - OpenSCAD compiled from master at latests reference (time of writing it is: 660dd121c71b0e7b77a23fe3699eafa266769440)

**Library & Graphics card information**
Results of openscad --info when executed using the Nvidia GPU. 
QObject::startTimer: Timers can only be used with threads started with QThread
OpenSCAD Version: 2024.08.27
System information: Linux 6.2.0-39-generic #40-Ubuntu SMP PREEMPT_DYNAMIC Tue Nov 14 14:18:00 UTC 2023 x86_64 Ubuntu 23.04 16 CPUs 30.98 GB RAM
User Agent: OpenSCAD/2024.08.27 (Linux x86_64; Ubuntu 23.04)
Compiler: GCC "12.3.0" 64bit
MinGW build: No
Debug build: Yes
Boost version: 1_74
Eigen version: 3.4.0
CGAL version, kernels: 5.5.1, Cartesian<Gmpq>, Extended_cartesian<Gmpq>, Epeck
OpenCSG version: OpenCSG 1.5.0
Qt version: 5.15.8
QScintilla version: 2.13.3
InputDrivers: 
GLib version: 2.76.4
lodepng version: 20230410
libzip version: 1.7.3
fontconfig version: 2.14.1
freetype version: 2.12.1
harfbuzz version: 6.0.0
cairo version: 1.16.0
lib3mf version: 1.8.1
Features: fast-csg, fast-csg-safer, fast-csg-debug, manifold, roof, input-driver-dbus, lazy-union, vertex-object-renderers-indexing, textmetrics, import-function, predictible-output
GL context creator: GLX (new)
GLX version: 1.4
GLEW version: 2.2.0
OpenGL Version: 4.6.0 NVIDIA 525.147.05
GL Renderer: Quadro RTX 3000/PCIe/SSE2
GL Vendor: NVIDIA Corporation
RGBA(8888), depth(24), stencil(8)
GL_ARB_framebuffer_object: yes
GL_EXT_framebuffer_object: yes
GL_EXT_packed_depth_stencil: yes

damienmarchal, 2024-08-27:
I dumped in a png file the content of the framebuffer computed in the MouseSelector::select function. The results are quite surprising (and explains the difference in behavior). 

Here is the content with the Intel IGP
[image]

This is the content with the Nvidia GPU
[image]

... well it is not surprising that the selection does not work. 
The strange thing is that the content of the 3d view are the same, only selection buffer is different.

damienmarchal, 2024-09-02:
Hi all,

Is anyone can reproduce the problem. 
Otherwise it may be only on my system (thus, not worth investigating).

kintel, 2024-09-15:
Is this the same issue as #4595, or do you see any indication that it might be a different issue?
Each GPU will implement z buffers differently internally, so I believe all bets are off if z values are produced through different computational paths...

damienmarchal, 2024-09-15:
Hi @kintel,

I think they are different. 

In #4595 the problem is that the Z values computed by the fixed pipeline of OpenCSG and the ones computed using the programmable pipeline (the one used when a shader is on) can be slightly different. The 3D preview use OpenCSG to compute the depth, then in a second pass it renders the colors with a depth test set to GL_EQUAL. When a shader is activated (eg: for mouse selection or edge rendering) the small difference generates kind of z-fighting on the 3d view and the makes the mouse right clik fails where the glitches are. 

In #5274, there is a totally different problem which happens between NVIDIA and Intel GPU. As the picture I made previously shares the two problem at the same time here is a new one where I try to emphasize the two problems (this is the scene scene from same viewpoint so the rendering between nvidia and IGP should be the same). While, on Nvidia the object 1 is totally missing and object 2 has in the left one of its cylinder not subtracted. The problem only happen on the selection buffer... the rendering is fine. 

[image]

kintel, 2024-09-15:
The selection buffer just renders the CSG tree with "ID colors", right?

damienmarchal, 2024-09-16:
Yes. 
  
Here is (what I think) is the situation:

| Mode  | First pass | Second pass | Depth test on second pass | 
| ----------- | -------- | -------- | ----------| 
| Preview   | OpenCSG (fixed pipeline) | fixed pipeline | GL_EQUAL 
| Preview with edges  | OpenCSG (fixed pipeline) | programmable pipeline | GL_EQUAL
| Mouse selection   | OpenCSG (fixed pipeline) |  programmable pipeline | GL_EQUAL

In the previous screenshot, the different reds intensities are in fact the ID-colors (with a scaling factor to make that visible). 

On the left side of the screenshot, we can the  z-artifacts (the part of the objects which have no id because the z-test fails) because of the use GL_EQUAL when mixing fixed & programmable pipeline.  This is #4595 . 

On the right side, we can also the z-artifacts... but what I when to emphasize (issue # 5274)  is that, at least on my system, with NVIDIA there is no rendering Object1, and there is weird rendering of object 2.    

Currently I wonder if for issue #5274 can be reproduced so I would be happy to have feedback on anyone having a nvidia gpu (linux, windows? ) to help narrow the problem.

kintel, 2024-09-16:
I just noticed the GL error mentioned above. Does that error only surface on nVidia? Perhaps chasing that error could be a good investigation in itself. Is that trivially reproducible? I got an nVidia GPU on another system I can test.

damienmarchal, 2024-09-17:
Yes the error only appears on Nvidia and is trivially reproducible. I provided the step in the issue and a video of the scene to help. Feedback on a nvidia gpu would be great. 

I investigated the error, but it seems non trivial (for me to understand).

closed 2025-04-08 by damienmarchal