openscad/openscad #5275: CLI input on windows does not accept space charachters.

jjcohen, 2024-08-28:
**Describe the bug**
Space chrachters are not accepted in the cli arguments when launching from windows powershell.

**To Reproduce**
Steps to reproduce the behavior:
1. Enter a space in an input string on an openscad model input while executing from the windows cli.

**Expected behavior**
Space charachters should be accepted from the Windows cli.

**Code reproducing the issue**
This command works : `openscad.com -D 'input_message=\"Test\u0020test\"' model.scad`
This does not work : `openscad.com -D 'input_message=\"Test test\"' model.scad`

My model is also using `Cascadia Mono:style=SemiBold` as it's font choice.

**Screenshots**
Here is the error I get with my model (which works fine with the unicode space substitution):
[image]

**Environment and Version info (please complete the following information):**
 - OS: Windows 11
 - System: AMD 64 bit
 - OpenSCAD Version:  2021.01 from the openscad.org website

**Library & Graphics card information**
N/A

**Additional context**
N/A

Thanks to all the contributors, really loving openscad so far!

tylamb1, 2024-12-20:
I have the same issue. Same environment as jjcohen.

jordanbrown0, 2024-12-20:
Command line parsing on Windows is ... tricky.

For the Command Prompt processor, this incantation works:
```
"\Program Files\OpenSCAD"\openscad -D "x=\"hello world\"" -o echo.stl echo.scad
```

I haven't found an incantation that works for PowerShell.

t-paul, 2024-12-20:
Not an OpenSCAD issue.

See also https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/Using_OpenSCAD_in_a_command_line_environment

closed 2024-12-20 by t-paul

ostat, 2025-04-10:
> Command line parsing on Windows is ... tricky.
> 
> For the Command Prompt processor, this incantation works:
> 
> ```
> "\Program Files\OpenSCAD"\openscad -D "x=\"hello world\"" -o echo.stl echo.scad
> ```
> 
> I haven't found an incantation that works for PowerShell.

in powershell you can use
` "\Program Files\OpenSCAD"\openscad -D "x=""hello\u0020world""" -o echo.stl echo.scad`

to generate the args you can use this 
```
" -D `"x=`"`"$($thetext.Replace(' ','\u0020'))`"`"`""
```